fredlcore/BSB-LAN #593: [BUG] esp_littlefs: mount failed,  (-84)

inonoob, 2023-10-08:
**Before submitting a bug report, please make sure that you have checked [chapter 14]
Has been done. No chapter found for that error. Also Checked the FHEM forum and Github issues. 

Thanks in Advance for your support. I couldn't find any help by myself. So I hope to get support from here. 

**BSB-LAN Version**
#define MAJOR "3"
#define MINOR "2"
#define PATCH "2"
#define COMPILETIME "20230606224119"

**Architecture**
ESP32 Olimex EVB - NO SD CARD Installed
Logic Level Adapter 3.3V/12V V4.4

**Bus system**
LPS

**Describe the bug**
After flashing version 3.2.2 or version  3.1 of BSB-LAN, the board ESP32 Olimex EVB is not connecting to wifi as code seems to have issue with the esp_littlefs (see Serial Monitor logs). With LAN, system can be accessed but is not savings anything. 

System was running with an uptime of 2942113 with version 3.0.5-20230318092150. Why did I touch a running system. I couldn't get any MQTT messages with that version if set to plain text. JSON worked fine but not for everything in Home assistant. And everyone seems to us Plain text. I hoped with the new version I could fix that issue and have finally the MQTT plain text option working. But unfortunately, I have bigger issues now that the system is no longer running at all :(.  

**To Reproduce**
Steps to reproduce the behavior:
- Download the latest version from github 
- Remove all the default from the files 
- Add in the BSB_LAN_config.h the wifi SSID and WIFI password 
- Compile the sketch
- ensure to select proper board 
- in my case OLIMEX ESP32-EVB on  /dev/ttyUSB0
- Flash frequency 80 Mhz 
- Partition Scheme: Minimal SPIFFS (Large APP with OTA) 
- Flash to board 
- Flashing will work but will exit with the littleefs error. 

**Logs**

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0xb (HSAddress EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 819
Address EEPROM option 14: 823
Address EEPROM option 15: 1143
Address EEPROM option 16: 1149
Address EEPROM option 17: 1150
Address EEPROM option 18: 1154
Address EEPROM option 19: 1158
Address EEPROM option 20: 1162
Address EEPROM option 21: 1166
Address EEPROM option 22: 1168
Address EEPROM option 23: 1172
Address EEPROM option 24: 1176
Address EEPROM option 25: 1240
Address EEPROM option 26: 1304
Address EEPROM option 27: 1305
Address EEPROM option 28: 1306
Address EEPROM option 29: 1316
Address EEPROM option 30: 1317
Address EEPROM option 31: 1637
Address EEPROM option 32: 1638
Address EEPROM option 33: 1642
Address EEPROM option 34: 1643
Address EEPROM option 35: 1644
Address EEPROM option 36: 1645
Address EEPROM option 37: 1649
Address EEPROM option 38: 1714
Address EEPROM option 39: 1779
Address EEPROM option 40: 1844
Address EEPROM option 41: 1876
Address EEPROM option 42: 1878
Address EEPROM option 43: 1879
Address EEPROM option 44: 1880
Address EEPROM option 45: 1881
Address EEPROM option 46: 1913
Address EEPROM option 47: 1977
Address EEPROM option 48: 2017
Address EEPROM option 49: 2057
Address EEPROM option 50: 2097
Address EEPROM option 51: 2098
Address EEPROM option 52: 2099
Address EEPROM option 53: 2100
Address EEPROM option 54: 2101
Address EEPROM option 55: 2102
Address EEPROM option 56: 2103
Address EEPROM option 57: 2135
Address EEPROM option 58: 2136
Address EEPROM option 59: 2137
Address EEPROM option 60: 2138
Address EEPROM option 61: 2140
Address EEPROM option 62: 2142
Address EEPROM option 63: 2143
Address EEPROM option 64: 2144
Size of cmdtbl: 3668
free RAM: 286776
BSB-LAN version: 3.2.2-20230606224119
3.2.2-20230606224119
Microcontroller: ESP32/Olimex EVB
Using RX/TX pins 36, 17
PPS settings:
Starting SD..E (5504) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}

E (5505) esp_littlefs: mount failed,  (-84)
E (5507) esp_littlefs: Failed to initialize LittleFS
Waiting for DHCP address........................................................................................................................................................................................................
0.0.0.0
0.0.0.0
0.0.0.0
Waiting 3 seconds to give Ethernet shield time to get ready...
Device family: 0
Device variant: 0
Calculating free space on SD...122880 Bytes free
Start network services

Starting MDNS service with hostname BSB-LAN
Setup complete
60003 Ping!
Device family: 0
Device variant: 0

```

**Expected behavior**
Clean install of version 3.2.2 with parameter available for my heater

**Desktop (if applicable, please complete the following information):**
 - Linux Mint 20.3 Una
 - Arduino IDE 2.2.1 
 - esp32 by Expressif Systems 2.0.14

fredlcore, 2023-10-08:
Unfortunately, your bug report is not concise: 
1. If you leave the BSB_LAN_config.h as it is and just add the WiFi credentials, it will still use LAN because you have to activate `#define WIFI` which is by default deactivated. So I have to assume that the above log file comes from a LAN configuration
2. However, you write above that the error message only appears when using WiFi and LAN is not affected. So which is which now?
3. Again assuming that you made no other change in BSB_LAN_config.h, this means that `#define ESP32_USE_SD` is deactivated. If `define #LOGGER` is activated (as it is by default), logs will be written to the flash memory of the ESP32 using LittleFS.
4. Since you have upgraded from verson 3.0 to 3.1, the file system for the flash memory has changed from SPIFFS to LittleFS. BSB-LAN is programmed in such a way that if there is a problem mounting the LittleFS filesystem (for example because of a previous SPIFFS filesystem), the flash partition will be formatted (see the warning in the ChangeLog for version 3.1). If this does fail, there is a bigger problem at hand, such as a defective flash partition.

All these assumptions are just that - assumptions - because the relevant information was either not present in your bug report or the logs attached.
I would suggest two things: Erase the whole flash memory of the ESP32 (this is beyond the scope here, Google is your friend), or, as a workaround, enable `ESP32_USE_SD` and see if that already helps (and no, you do not need an SD card inserted to test this).

fredlcore, 2023-10-08:
Seriously, what is this? I write an extensive reply to your bug report regarding LittleFS and now you edit your report without even mentioning that/what you have edited? Some people...

fredlcore, 2023-10-08:
As for your WIFI problem, you probably just forgot to activate `#define WIFI`. Which is clearly described in the manual you said you have read.

closed 2023-10-08 by fredlcore

inonoob, 2023-10-08:
Dear Fredlcore, 

I'm deeply sorry. I did find out myself after checking the flash speed. Sorry for the issue. I haven't update the system in a long time. 

I missed the #define WIFI part !!

fredlcore, 2023-10-08:
Ok, never mind ;)...