fredlcore/BSB-LAN #596: [BUG]

proasnet, 2023-10-21:
**BSB-LAN Version**
BSB-LAN, Version 3.0.4-20221222224441

**Architecture**
Arduino Due

**Bus system**
BSB

Hello, please I need a little help. I need find parameter, that show me a position of heat/dhw valve.
Heater I have Baxi Luna Platinum. I am using BSB a few years, but this I need now.
Thank you very much.

fredlcore, 2023-10-21:
That's not a bug. And I don' even know what you mean with "position of heat valve". In short, if there is no parameter on your heater that provides the information you are looking for, there's nothing that BSB-LAN can do.

closed 2023-10-21 by fredlcore

proasnet, 2023-10-21:
I am sorry. Its not bug.
On a panel QAA75 is two icons that shows a position. I thing, that may be as some parameter, its from BSB bus.
[image]

fredlcore, 2023-10-21:
These are status information that do not belong to a specific parameter and are not part of the dump that the heater generates. However, you may find similar information in text form in the 8xxx range. Having said that, we did some level of telegram decoding prior to version 2.2, so you may want to install that version and see if there are parameters in the 8xxx range that provide the information you need and then port these to your current BSB_LAN_custom_defs.h as described in the manual. But because this is unverifyable data, we no longer include these in the supported parameters.