fredlcore/BSB-LAN #599: [BUG] compile error

Magnus8513, 2023-11-10:
**BSB-LAN Version**
As displayed in the web-configuration or copy from `bsb-version.h` file. Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly. We don't fix bugs for discontinued platforms (such as Arduino Mega).

**Architecture**
ESP32 Olimex EVB

**Bus system**
BSB

**Describe the bug**
compilation errorcode 1 and exit with ping timeout

**To Reproduce**
Steps to reproduce the behavior:
verify and upload sketch to Olimax using the steps described in the manual herer:
[https://1coderookie.github.io/BSB-LPB-LAN_EN/chap02.html#21-installation](url)

**Logs**
see file attached. 
[2023_11_10_Error_BSB_Lan_comp.txt](https://github.com/fredlcore/BSB-LAN/files/13318495/2023_11_10_Error_BSB_Lan_comp.txt)

**Expected behavior**
sucessful upload to olimex

**Desktop (if applicable, please complete the following information):**
 - OS: MacOS 12.7.1

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/fredlcore/BSB-LAN/files/13318486/BSB_LAN_config.txt)

Magnus8513, 2023-11-10:
here the "red" parts from the log:

`Compiling sketch...
/Users/magnusblatt/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.6-dirty\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/newlib/platform_include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/freertos/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/freertos/include/esp_additions/freertos -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/freertos/port/xtensa/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/freertos/include/esp_additions -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_hw_support/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_hw_support/include/soc -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32 -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_hw_support/port/esp32 -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/heap/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/log/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/lwip/include/apps -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/lwip/include/apps/sntp -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/lwip/lwip/src/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/lwip/port/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/lwip/port/esp32/include/arch -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/soc/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/soc/esp32 -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/soc/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/hal/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/hal/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/hal/platform_port/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_rom/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_rom/include/esp32 -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_rom/esp32 -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_common/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_system/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_system/port/soc -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_system/port/public_compat -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/xtensa/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/xtensa/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/driver/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/driver/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_pm/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_ringbuf/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/efuse/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/efuse/esp32/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/vfs/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_wifi/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32/include/esp_event/include -I/Users/magnusblatt/Library/Arduino15/packages/esp32/ [... cut ...]
`xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
python3 /Users/magnusblatt/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/gen_esp32part.py -q /private/var/folders/1q/wk9tf0610_zfkqh2dbs1_lw00000gn/T/arduino/sketches/AD65426CFB653523FEE91FD83B6B432D/partitions.csv /private/var/folders/1q/wk9tf0610_zfkqh2dbs1_lw00000gn/T/arduino/sketches/AD65426CFB653523FEE91FD83B6B432D/BSB_LAN.ino.partitions.bin
`

Magnus8513, 2023-11-10:
tried it with the older 2.0.8 framework ending with the same result:

[image]
[2023_11_10_Error_BSB_Lan_comp_2_0_8.txt](https://github.com/fredlcore/BSB-LAN/files/13318634/2023_11_10_Error_BSB_Lan_comp_2_0_8.txt)

Magnus8513, 2023-11-10:
solved it by researching the error messages above. Installing the developer Tools on my mac a

`xcode-select --install` 

as described here:
[https://apple.stackexchange.com/questions/254380/why-am-i-getting-an-invalid-active-developer-path-when-attempting-to-use-git-a](url)

fredlcore, 2024-02-26:
For the installation guide, we always have to decide between too much and too little information. This is an error that results from a broken build environment that is not specific to BSB-LAN, if you google the error message, there are many sites that come up with the solution you found, so that's why we're not going to add it in our manual.

closed 2024-02-26 by fredlcore