fredlcore/BSB-LAN #605: [BUG] "/JS" POST request unexpected behavior

timebis, 2024-01-02:
**BSB-LAN Version**

```
(base) thomas@thomas-MS-7848:~$ curl -X GET '192.168.1.11/JI' -H 'Content-Type: application/json'
{
  "name": "BSB-LAN",
  "version": "3.2.2-20230606224119",
  "hardware": "ESP32",
  "freeram": 208424,
  "uptime": 33967218,
  "MAC": "00:80:41:19:69:90",
  "freespace": 8192,
  "bus": "BSB",
  "buswritable": 0,
  "busaddr": 66,
  "busdest": 0,
  "monitor": 0,
  "verbose": 1,
  "onewirebus": 0,
  "onewiresensors": 0,
  "dhtbus": [

  ],
  "loggingmode": 0,
  "loginterval": 3600,
  "logged": [
    { "parameter": 8700, "destination": -1 },
    { "parameter": 8743, "destination": -1 },
    { "parameter": 8314, "destination": -1 }
  ]
}
```

**Architecture**
ESP32 Olimex EVB

**Bus system**
BSB

**Describe the bug**
I'm trying to change my comfort temperature (param 710 in my heat pump) with a POST request (/JS). The request seems to be accepted, but when I try to GET request the parameter again, the old value is still set. I do have the same problem with others parameters.
Setting the same param with the WEB page works well. Also, I noticed the web app used "/S710=<value>" request. I tried it with curl, it worked well.

Do you have an idea why do I have this issue with "/JS" query ?

**To Reproduce**
```
(base) thomas@thomas-MS-7848:~$ curl -X GET '192.168.1.11/JQ' -H 'Content-Type: application/json' -d '{"Parameter": "710"}'
{
  "710": {
    "name": "Comfort setpoint ",
    "dataType_name": "TEMP",
    "dataType_family": "VALS",
    "destination": "0",
    "error": 0,
    "value": "24.0",
    "desc": "",
    "precision": 0.1,
    "dataType": 0,
    "readonly": 0,
    "readwrite": 0,
    "unit": "°C"
  }
}
(base) thomas@thomas-MS-7848:~$ curl '192.168.1.11/JS' -H 'Content-Type: application/json' -d '{
           "Parameter": "710",
           "Value": "21.0"}'
{
  "710": {
    "status": 1
  }
}
(base) thomas@thomas-MS-7848:~$ curl -X GET '192.168.1.11/JQ' -H 'Content-Type: application/json' -d '{"Parameter": "710"}'
{
  "710": {
    "name": "Comfort setpoint ",
    "dataType_name": "TEMP",
    "dataType_family": "VALS",
    "destination": "0",
    "error": 0,
    "value": "24.0",
    "desc": "",
    "precision": 0.1,
    "dataType": 0,
    "readonly": 0,
    "readwrite": 0,
    "unit": "°C"
  }
}
(base) thomas@thomas-MS-7848:~$ curl -X GET "http://192.168.1.11/S710=20.0"
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN"><html><head><meta http-equiv="content-type" content="text/html;charset=UTF-8"><meta name="viewport" content="width=device-width,initial-scale=1.0,minimum-scale=1.0"><link rel="icon" type="image/svg+xml" href="/favicon.svg"><title>BSB-LAN</title><style>A{text-decoration:none}A:visited,active{color:blue}A:hover{color:red;background-color:yellow}A:focus{color:red}a.logo:hover{background-color:initial;text-decoration:initial}table{width:100%;max-width:1024px;margin:auto}td{vertical-align:top}td.header{vertical-align:middle}svg.logo{width:10%;height:10%}input{width:100%;box-sizing:border-box}select{width:100%}</style></head><body><script>function set(p){var v=document.getElementById('value'+p).value.replace(/\.$/,'').replaceAll(' ','_').replace('---','');window.open(document.getElementById('main_link').href+'S'+p+'='+v,'_self')}function setbit(p){var x=document.getElementById('value'+p),v,i;for(v=i=0;i<x.options.length;i++)if(x.options[i].selected)v=v+eval(x.options[i].value);window.open(document.getElementById('main_link').href+'S'+p+'='+v,'_self')}</script><font face='Arial'><center><a href='/' ID=main_link class='logo'><svg class='logo' width='400pt' height='400pt' version='1.0' viewBox='0 0 400 400' xmlns='http://www.w3.org/2000/svg'><g transform='translate(0 400) scale(.1 -.1)'><path id='B' d='m292 3510c-61-38-62-42-62-537 0-501 0-500 70-536 32-16 69-17 445-15 399 3 411 4 451 25 102 54 151 156 141 293-6 92-35 164-95 238-37 46-41 55-31 74 23 43 39 124 39 203 0 97-27 165-86 213-70 58-100 62-488 62-334 0-353-1-384-20zm708-275c0-60-21-98-64-119-28-13-70-16-235-16h-201v-125-125h246c203 0 252-3 279-16 40-19 65-66 65-123v-41h-305-305v305 305h260 260v-45z'/><path d='m1705 3516c-99-31-171-91-210-172-73-155-17-359 122-441 77-45 136-53 406-53h248l24-25c35-34 35-96 0-130l-24-25h-401-400v-126-125l418 3c407 3 418 4 473 26 77 31 142 90 176 160 24 50 27 69 28 152 0 84-3 102-28 152-33 67-90 121-166 157-55 26-59 26-338 31-325 6-318 4-318 91 0 41 4 54 24 70 22 18 44 19 377 19h354v125 125l-362-1c-264 0-374-4-403-13z'/><use href='#B' x='2500'/><path d='m1932 2152c-138-127-300-298-376-400-82-108-178-298-211-416-21-73-25-110-25-215 0-201 50-337 170-465 59-63 198-172 207-162 2 2-7 28-20 59-34 75-47 143-47 242v85h300 300v-82c0-99-21-206-53-270l-24-47 49 26c105 57 218 175 275 287 86 171 84 414-7 585-47 88-132 195-255 321-62 63-122 133-133 155-47 94-70 236-54 333 5 28 7 52 5 52s-47-40-101-88zm45-772c46-34 139-152 172-217l21-43h-240-240l11 27c30 80 185 263 223 263 7 0 31-14 53-30z'/><path d='m2735 1598c-55-43-55-44-55-593v-505h125 125v348 347l123-125c67-69 219-226 337-349l215-223 51-5c48-5 54-3 90 30l39 35 6 1052h-126-125l-2-345-3-345-237 247c-130 137-281 294-335 351l-99 102h-51c-36 0-58-6-78-22z'/><path d='m212 1098c3-459 5-515 20-536 41-59 22-57 531-60l467-3v126 125h-385-385v430 430h-125-126l3-512z'/></g></svg></A></center>
<table align=center><tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/K'>Heater functions</a></td><td class="header" width=20% align=center><a href='/K35'>Sensors</a></td><td class="header" width=20% align=center><font color=#000000>Display log file</font></td><td class="header" width=20% align=center><a href='/Q' TARGET='_new'>Device-specific parameter list</a></td></tr>
<tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/C'>Settings</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands' target='_new'>URL commands</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html' target='new'>Manual</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html' target='_new'>FAQ</a></td></tr></table><p></p><table align=center><tr><td class="header">
<tr><td>710 Heating circuit 1 - Comfort setpoint : 20.0 °C</td><td><input type=text id='value710' VALUE='20.0'></td><td><input type=button value='Set' onclick="set(710)"></td></tr>
</td></tr></table>
</body>
</html>

(base) thomas@thomas-MS-7848:~$ curl -X GET '192.168.1.11/JQ' -H 'Content-Type: application/json' -d '{"Parameter": "710"}'
{
  "710": {
    "name": "Comfort setpoint ",
    "dataType_name": "TEMP",
    "dataType_family": "VALS",
    "destination": "0",
    "error": 0,
    "value": "20.0",
    "desc": "",
    "precision": 0.1,
    "dataType": 0,
    "readonly": 0,
    "readwrite": 0,
    "unit": "°C"
  }
}

```

fredlcore, 2024-01-02:
I need the serial monitor log around the time you are issuing the /JS command.

fredlcore, 2024-01-02:
Also, you didn't use the syntax as described in the manual as you are missing the "Type" element.

timebis, 2024-01-02:
Thank you very much for the fast response !
Indeed I didn't understand well what was this "Type" element. I tried with "0" as a value and then forgot to try with "1" and remove it 😅.

I just retried with "1" and it works :)

What should the "Type" element be used for ?

closed 2024-01-02 by timebis

fredlcore, 2024-01-02:
All explained in the manual, chapter 5.3.

timebis, 2024-01-02:
>  Send: "Parameter", "Value", "Type" (0 = INF, 1 = SET)  

Yes I read it, but I still don't understand what does "INF" & "SET" stand for, sorry 😅

fredlcore, 2024-01-02:
Read on to chapter 5.4. Most commands require a SET, a few require an INF telegram.

timebis, 2024-01-02:
indeed, I missed it. Thanks !

fredlcore, 2024-01-02:
No worries, thanks for the support :)!