fredlcore/BSB-LAN #608: [BUG] Web Server Stopped Working

sebi5361, 2024-01-11:
_BSB-LAN version 3.3.2-20231115201023 running on ESP32 Dev Module connected to Atlantic Alfea Extensa Duo 6._

**The web server on my BSB-LAN stopped working.** This is the telnet debug log when I try to reach it using a browser:
```
Version: 3.3.2-20231115201023
GET / HTTP/1.1

Timeout
```
MQTT exchanges stopped as well, despite telnet debug connection being still alive.
Reset didn't do anything bringing back the http server to normal.
I had to reflash the whole firmware.
I didn't do anything special to corrupt the http server. Just adjusting some parameters.
Did other users experienced this too?
Is there a way to perform a full reset of the board (resetting config parameters to their original value) without having to reflash the entire firmware?

fredlcore, 2024-01-11:
If the debug log via telnet still shows the `GET / HTTP/1.1` then this means that the network connection is still active and the request for the website was received. There is no "webserver" per se (in the sense of a separate component) in BSB-LAN, so if it does not respond to the request then something else takes too long and leads to a timeout.
It's nearly impossible to diagnose this from afar, but you can reset all settings in the EEPROM by calling /NE which will clear the EEPROM including all settings and then restart BSB-LAN with the settings in BSB_LAN_config.h. If you have made changes in that file, go back as close as possible to the default settings, flash it and monitor the situation after each change you do.

sebi5361, 2024-01-11:
| URL Command | Effect  | 
|------------------|--------|
| /NE                   | Reset & reboot microcontroller (takes approx. 15 seconds) and erase EEPROM | 

Indeed I missed this command from the [manual](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands).
I will try when (if) it happens next time.

Thanks for your prompt answer 🤗

sebi5361, 2024-01-11:
I just realized that I set 27 MQTT logging parameters before freezing BSB-LAN:
|Settings|     |    |
|---------|---|---|
|Logging |Parameters|`700,710,712,720,721,1610,1612,1614,6800,6803,6805,6806,8000,8003,8304,8700,8703,8704,8740,8741,8742,8820,8830,8831,8950,8951,10019`|

20 parameters is the limit, right?
Could this be the reason for freezing HTTP and MQTT communications on BSB-LAN?

fredlcore, 2024-01-11:
Yes, 20 is the limit. And yes, of course, out-of-bound variable definitions can result in all sorts of problems.