fredlcore/BSB-LAN #609: [FEATURE] Heating-Engineer Additional Parameters Access

sebi5361, 2024-01-11:
_BSB-LAN version 3.3.2-20231115201023 running on ESP32 Dev Module connected to Atlantic Alfea Extensa Duo 6._

On my Atlantic Alfea Extensa Duo 6, there is a way to switch from **end-user** to **heating-engineer** mode, using the embedded  [AVS37](https://assets.new.siemens.com/siemens/assets/api/uuid:79e08242-2d5a-485d-834d-35811dbbb3f4/en-2014-07-30-user-manual-rvs46--.pdf) zone controller _(pressing OK in the default screen to access the settings menu, then pressing the information button for 3 sec, then selecting engineer_). 

This gives access to more settings. As such, an end-user can only access parameters **0 to 3** to set time, whereas a heating-engineer can access additional parameters **5 and 6** as well (beginning and end of summer time).

I didn't find a way (yet?) to switch to heating-engineer **permanently** on my heat exchanger.
Additionally, while in heating-engineer mode **temporarily**, I was able to access parameters 5 and 6 from my zone controller, but not from BSB-LAN (using URL commands /5 and /6).

**Is there a way to access the heating-engineer additional parameters from BSB-LAN?**

fredlcore, 2024-01-11:
The user-mode setting on the heater is always just temporary. 
With BSB-LAN, you can access all parameters on all levels that your heater advertises. This exclude the few parameters that are stored in the room unit directly, but that should not be more than a handful.

closed 2024-01-11 by fredlcore

sebi5361, 2024-01-11:
> With BSB-LAN, you can access all parameters on all levels that your heater advertises.

Ohh I thought all parameters were stored on the heater itself, and that room units were only dumb terminals... 
Thus, I thought that all parameters would be advertised on the bus, which is wrong. Thank you for the clarification 🙏.

Could you point me in the right direction to find some documentation regarding heater control and their settings?
Indeed on my heater, I am surprised that the flow temperature actual value of the heat circuit 1 is missing.
```
8743 Diagnostics consumer - Flow temp actual value heat circuit 1: --- °C	
```
It is like a temperature sensor is missing and I wonder how my heater works without this input (8743 is mentioned several times in the BSB-LAN documentation and seems to be important for most heater control).

fredlcore, 2024-01-11:
You should have received a manual from the company that set up your heating system, otherwise you'll have to contact the manufacturer. Each manufacturer can configure the controller how they like it (OEM).
If 8743 only gives you `---` then there is no sensor installed. Likewise, you'd have to contact your local heating service technician or check other parameters whether they carry the information you want.