fredlcore/BSB-LAN #610: [BUG] Constat reconnection to mqtt broker

iamer, 2024-01-11:
**BSB-LAN Version**
3.2.2-20230606224119

**Architecture**
Olimex ESP32-POE-ISO

**Bus system**
BSB-LAN is running on BSP 

I am not sure if this is a bug:
BSB-LAN keeps reconnecting to mqtt broker instead of reusing connection.
The topic status flip-flops between offline and online

This is the log from Mosquitto broker:
2024-01-11 22:10:48: New connection from 192.168.1.6:58879 on port 1883.
2024-01-11 22:10:48: Client BSB-LAN already connected, closing old connection.
2024-01-11 22:10:48: New client connected from 192.168.1.6:58879 as BSB-LAN (p2, c1, k15, u'mqtt').
2024-01-11 22:11:37: Client BSB-LAN has exceeded timeout, disconnecting.
2024-01-11 22:11:37: New connection from 192.168.1.6:58880 on port 1883.
2024-01-11 22:11:37: New client connected from 192.168.1.6:58880 as BSB-LAN (p2, c1, k120, u'mqtt').
2024-01-11 22:11:53: New connection from 192.168.1.6:52441 on port 1883.
2024-01-11 22:11:53: Client BSB-LAN already connected, closing old connection.
2024-01-11 22:11:53: New client connected from 192.168.1.6:52441 as BSB-LAN (p2, c1, k15, u'mqtt').
2024-01-11 22:12:18: New connection from 172.30.32.2:49868 on port 1883.
2024-01-11 22:12:18: Client <unknown> closed its connection.
2024-01-11 22:12:26: New connection from 192.168.1.6:62358 on port 1883.
2024-01-11 22:12:26: Client BSB-LAN already connected, closing old connection.
2024-01-11 22:12:26: New client connected from 192.168.1.6:62358 as BSB-LAN (p2, c1, k15, u'mqtt').

[BSB_LAN_config.h.gz](https://github.com/fredlcore/BSB-LAN/files/13908180/BSB_LAN_config.h.gz)

fredlcore, 2024-01-11:
Yes, there is a new call to mqtt_connect every log interval (i.e. 60 seconds), but inside the function is are actually two tests prior to actually establishing a new connection, one is 
`if(MQTTPubSubClient == NULL) {`
and the other one is
`if (!MQTTPubSubClient->connected()) {`
One of them seems to report that there currently is no connection, so you might want to add debug messages to both of them to figure out which one matches and results in establishing a new connection.

An actual disconnect from BSB-LAN is only done when the configuration is written, a reset is executed or MQTT is disabled in the webinterface, so none of that should be the case here.

iamer, 2024-01-12:
The  log interval in my settings was 30 seconds. Based on your message I increased it to 60, and the disconnects have not happened for 30 minutes now. I will keep an eye on it for a while before considering the issue fixed. Thanks!

iamer, 2024-01-12:
BSB-LAN has not disconnected from mqtt for 9 hours as far as I can see. I will close this bug. Thanks again.

closed 2024-01-12 by iamer