fredlcore/BSB-LAN #615: [BUG] PPS Errors with BSB-LAN Versions > 2.0.16

malansk, 2024-01-20:
**BSB-LAN Version**
2.0.16 (ok)
2.0.130 (PPS "Betriebsart" not switchable, PPS "Zieltemperatur" always 0° -> Heater not running)
2.2.1 (PPS-Error Displayed on Heater Control Module)
3.x (PPS-Error Displayed on Heater Control Module) 

**Architecture**
Arduino Due w. LAN shield

**Bus system**
PPS

**Describe the bug**
I have Version 2.0.16 running since 12/2020 but after approx. 3 months uptime I always need to restart the Arduino Due because BSB-LAN is no longer running (Web-Interface not reachable, WGET from Smarthome system gives no values)

Today I tried newer Versions of BSB-LAN but they are missing some PPS values or the block the PPS bus on the heater control module

Version 2.0.130
PPS "Betriebsart" can be changed but not change on heater control module 
PPS "Zieltemperatur" always shows 0°C (with Version 2.0.16 it shows 15°C or 24°C depending on Presence Switch)
Heater is not running in Automatic mode (only OFF mode w. Freeze-Protection)

Version 2.2.1 & 3.x
As soon as the Arduino Due is up and running, I get "Err" Message on the Heater Control Module. It shows Error 146 - PPS-Bus-Error
If I change the configuration from "Raumregelgerät" to "Passive" (no writing on PPS bus) the Error dissappears and I get some (but not all) PPS values

I am uploading each version from different folder on hard-drive, with the config files renamed (from .default) and with the necessary configuration settings (e.g. Bus Type "PPS" etc.).
I have even cleared the EPROM content by command line switch /NE

After Downgrading to 2.0.16, everything works ok (but after 3 months uptime needs restart).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs**
If possible, attach or copy/paste a log of the Arduino IDE's serial monitor when performing the above actions. If the error is reproducible, do so after pressing the reset button on the board.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**
 - OS: Windows 10
 - Browser Chrome 
 - Browser version 116.0.5845.141 (Offizieller Build) (32-Bit)

**Additional context**
Add any other context about the problem here.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Bug reports without this file attached will be closed immediately.**

fredlcore, 2024-01-21:
Since I no longer have access to a PPS controller, I would need serial monitor log files from all three versions, both when running in passive mode as well as in room unit mode. The logs need to begin when booting the Arduino and need to run for 1-2 minutes and during that time you should change the problematic parameters.

malansk, 2024-01-21:
Yes, I was trying to get serial monitor log files from the Arduino Due but was not successfull. Later I found some reference posts that one the Due it‘s not possible to get the SerialMonitor output from the same port like the programming was done, so I will try to open the case and switch over the USB cable from native port to programming port to get some logfile. Would logfile from SD-Card also work ? I saw some options to enable logging of telegrams.

fredlcore, 2024-01-21:
No, that is wrong information. Yes, the Due has two USB ports, but for BSB-LAN, both programming as well as serial output is via the programming port. And no, logging to SD card is not enough in case of troubleshooting as the logs do not cover both sending and receiving data.

fredlcore, 2024-02-26:
I have had reports from PPS users using the most recent BSB-LAN version on a rather old (even for PPS) RVA36 controller. So I assume your situation is a bit more specific. With the said log files, I can nevertheless have a look, but without these, there's nothing I can do. So for the moment, I'm going to close this issue until there is further background information.

closed 2024-02-26 by fredlcore

malansk, 2024-05-06:
Today I had finally time to collect some log files with Serial Monitor running on the Arduino Due Programming port.

1. "PPS Bus Errors when write on bus is enabled" where probably caused by bad EEPROM content e.g. Heating start / end-times of >24h.
EEPROM was probably not cleared by /NE switch due to missing definition of #define RESET in config.h.
After properly clearing EEPROM, no more PPS BUs Errors shown in the heating controller display

2. Version 2.0.16 shows real data on almost all channels and can write to all channels which have SET button in WEB-GUI 
(Betriebsart, Präsenztaste, Außentemperatur, Vorlauftemperatur, Mischervorlauftemperatur, Trinkwassertemperatur, TWW Reduziertsollwert, TWW Nennsollwert, Komfortsollwert, Reduziertsollwert,  Frostschutzsolltwert, Sollwert Maximum, Raumtemperatur (set by BSB-LAN), Heizperioden, Uhrzeit, Zieltemperatur.

 Version 2.016 is missing data on these channel (not shure if this is bug or if the heating controller doesn't send them over PPS)
Brennerstatus, Außentemperatur gemischt, Vorlaufsollwert

3. Version 2.1 is showing the most Temperature data in the Serial Monitor with 0°C  and showing wrong values for Außentemperatur in the Serial Monitor. Many telegramms are missing at all (Brennerstatus, Heating Times).
   
4. Version 3.3.5 does not recognize any PPS telegrams at all in PPS Bus read mode, even though the raw messages look the same (et least the unrecognized ones). With write mode enabled, the same telegrams like in 2.1 are shown (also with 0°C for the temperatures and missing the Brennerstatus and Heating times).

6. Version 3.3.5 is not able to connect to MQTT server with PPS write mode enabled.

Please let me know if you need some more logs / information.

malansk, 2024-05-06:
I have copied some interesting differences in the SerialMonitor_Vergleich.txt.
Is looks like some parts of the telegrams are missing the second half of the values (only 00) - the first half seems to be ok.
The values which I am able to set via WEB-GUI (e.g. Komfortsollwert or Reduziertsollwert) seems to be complete.
Maybe the PPS Bus read function truncates the second half of the telegrams ?

malansk, 2024-05-06:
@fredlcore 
please have a look at this, I'm not shure if you see this update since the issue has already been closed.

fredlcore, 2024-05-06:
Thanks for the intensive research, however, 3.3.5 is already several weeks/months old, current version is 3.4.6, and I also fixed a non-trivial bug for PPS in the meantime. So could you please check again with the most recent version and let me know if the problem persists? Thanks!

malansk, 2024-05-06:
Where do I get it ?
On Github it shows 3.3 as latest...
https://github.com/fredlcore/BSB-LAN/releases

fredlcore, 2024-05-06:
From the master repository, as described in the manual...

malansk, 2024-05-06:
Version 3.4.6 gives me PPS Bus Error on the heating system if I set PPS bus to write - even with EEPROM cleared.
Depending on QAA Type, Serial Monitor gives "Collision on Bus detected".

Attached are the latest logfiles for read and write.

malansk, 2024-05-06:
[SerialMonitor_3.4.6.txt](https://github.com/fredlcore/BSB-LAN/files/15221754/SerialMonitor_3.4.6.txt)
[SerialMonitor_3.4.6_write_PPSBUSERROR.txt](https://github.com/fredlcore/BSB-LAN/files/15221755/SerialMonitor_3.4.6_write_PPSBUSERROR.txt)
[SerialMonitor_Vergleich.txt](https://github.com/fredlcore/BSB-LAN/files/15221756/SerialMonitor_Vergleich.txt)

fredlcore, 2024-05-06:
You seem to have the "monitor" function active. Don't do that, that's basically only necessary for specific debugging purposes which is why it is turned off by default. In this case, it might cause problems because PPS is a very timing sensitive protocol.
Furthermore, I  don't see bus errors in your file S.erialMonitor_3.4.6_write_PPSBUSERROR.txt.

malansk, 2024-05-06:
I have changed #define DEFAULT_FLAG from 0 (which I was using in 2.0.16) to FL_SW_CTL_RONLY and now I am getting live PPS values from the heating controller :-)

But there are some issues:

*Mischervorlauftemperatur is shown as -512°C (this was ok in 2.0.16 but changed already in Version 2.1 to -512°C)
 (2.0.16 PPS-Bus - Mischervorlauftemperatur: 30.6 &deg;C00 2c 00 00 00 00 07 a5 00
#15004: 30.6 &deg;C
30.600000
30.600000)
(3.4.6 PPS-Bus - Mischervorlauftemperatur: -512.0 °C
00 2C 00 00 00 00 80 01 00
#15034: -512.0 °C)

*QAA Type (was "15062 PPS-Bus - QAA Modell: 83 - QAA 70fd 38 ff ff ff ff ff 53 7d" in Ver 2.0.16) is now "15046.0 PPS-Bus - QAA Modell: 65363FD 38 FF FF FF FF FF 53 7D" (Modell 83 changed to 65363 or so some other number starting with 65...).
*Betriebsart (was "15010 PPS-Bus - Betriebsart: 0 - Automatischfd 49 ff ff ff ff ff 00 bf" in Version 2.0.16 is now "PPS ANS 15000.0 PPS-Bus - Betriebsart: 65281" or some other number starting with 65...

Please check the logs for the values of QAA Type, Betriebsartumschaltung and Mischervorlauftemperatur.
By the way, "Präsenztaste" is working and I can switch it on and off.

Heating controller still shows "PPS Bus error" since Version 3.4.6 (it is not shown in the log of the serial monitor) and I can't change Betriebsart on the controller.
Maybe it is blocked by wrong QA type and wrong Betriebsart from BSB-Lan ?

I have turned off the "monitor" function off now.

malansk, 2024-05-06:
[SerialMonitor_3.4.6_write_PPSBUSERROR_PPS_Values_read_ok_FL_SW_CTL_RONLY_MonitorOFF_Betriebsartumschaltungen.txt](https://github.com/fredlcore/BSB-LAN/files/15222997/SerialMonitor_3.4.6_write_PPSBUSERROR_PPS_Values_read_ok_FL_SW_CTL_RONLY_MonitorOFF_Betriebsartumschaltungen.txt)

malansk, 2024-05-06:
There is no broken sensor, with 2.1.16 is shows the correct temp, with each version higher it shows -512°c.

The fix looks completely broken, for most telegrams, showing decoding error in GUI and other error in logfile (refer to log file)

malansk, 2024-05-06:
[SerialMonitor_3.4.6_newBuilt_Broken.txt](https://github.com/fredlcore/BSB-LAN/files/15224543/SerialMonitor_3.4.6_newBuilt_Broken.txt)

fredlcore, 2024-05-06:
Well, the imortant info (also) is that the telegrams in question before are now displayed correctly. With the bugfix I've just uploaded, it should work fine.
As for the sensor, I cannot change the telegrams coming from the heater. This is what your heater sends to BSB-LAN:
`1D 2C FF FF FF FF 80 01 3A`
The first byte (1D) indicates a broadcast information telegram from the heater to the QAA.
The second byte is indicating the type of parameter (2C = Mischervorlauftemperatur) and the 3rd- and 2nd-last bytes are the temperature (0x8001 divided by 64 (temperature operand) is -512).
In version 2.0.16 you quoted a different telegram, so that's the reason for the different temperatures displayed. As to _why_ your heater sends different telegrams with the different versions, I have no idea. Maybe you use a different QAA mode, but whatever it is, I have no PPS heating system to figure this out. If you figure out what causes the different telegrams and if this can be influenced by BSB-LAN, I can certainly try to adjust that, but otherwise, there is no chance for me to do so. 
Could it be that this specific temperature now shows up on a different parameter?

malansk, 2024-05-06:
I have found detailed manual, it shows that the "PPS Error" I was refering to on the Heater controller is in fact Error Code "146" which is "Ungültige Anlagenkonfiguration". The Anlagenkonfiguration specifies which type of heater (1-stufig / 2-stufig / modulierend) and which type of peripherals (Pumpen, Mischer). That's why in Version newer then 2.1.16 it shows -512°C for Mischervorlauftemperatur and "Ungültige Anlagenkonfiguration" because BSB-Lan somehow changes my original "Anlagenkonfiguration" Type 15 (1-stufig, Mischerkreis) to something else if write to pps bus is enabled !
Was there an update from 2.1.16 which somehow sets heater type / family type / device type ?
In the config.h I have always used this:
===============================
/* Set the device family and device variant of your heating system. Only change this if you _really_ know what you are doing!
 * Set fixed_device_family and fixed_device_variant to your device family and variant (parameters 6225 and 6226) here
 * if heating system is not running when BSB-LAN is powered on.
*/
uint16_t fixed_device_family = 0;
uint16_t fixed_device_variant = 0;
======================
Maybe this setting was not working for PPS in 2.1.16 but was updated to actually work from 2.1.xx ?
Is there a way to disable this in the code for testing ?

fredlcore, 2024-05-06:
If you want to try some more, you can _either_ uncomment the two `tx_msg` in lines 24 and 25 in include/pps_handling.h (case 3) _or_ in lines 28 an 29, so we can figure out which causes the issue. But still, a comparison with what your QAA sends in these cases would be helpful in order to hopefully decode what the actually mean...

malansk, 2024-05-06:
Uncommenting case 3 or case 4 doesn't make a difference.
But uncommenting case 0 (QAA Type setting) solves the issue - no more "Ungültige Anlagenkonfiguration" Error and everything is working like in 2.1.16 !
The weird thing is that looking at the Serial monitor I get the same result (FD 38 FF FF FF FF FF 53 7D) if case 0 is active in 3.4.6 like I get from 2.1.16 so I doubt that the actual code from this line in pps_handling.h is the culprit.
But since in case 0 "pps_values[PPS_QTP] = QAA_TYPE" is set, maybe it triggers a bug later somewhere in the code.
e.g. 
"case 0x38: pps_values[PPS_QTP] = msg[7+pps_offset]; break; // QAA type"
or
"case 0x38: msg_cycle = 0; break;"

If I leave case 0 enabled and uncomment the "case 0x38: msg_cycle = 0; break;" line the heater starts and runs for approx. 5 minutes before throwing the "ungültiger Anlagentyp" error.

So it looks like there is definitly something buggy with this QQA Type / address 0x38 handling at least for my system.

malansk, 2024-05-06:
I will let it run overnight with case 0 uncommented to see if this is truly stable,

malansk, 2024-05-06:
[SerialMonitor_3.4.6_latestBugfix_read.txt](https://github.com/fredlcore/BSB-LAN/files/15227235/SerialMonitor_3.4.6_latestBugfix_read.txt)
[SerialMonitor_3.4.6_latestBugfix_write.txt](https://github.com/fredlcore/BSB-LAN/files/15227236/SerialMonitor_3.4.6_latestBugfix_write.txt)
[SerialMonitor_3.4.6_latestBugfix_write_case0-uncommented.txt](https://github.com/fredlcore/BSB-LAN/files/15227237/SerialMonitor_3.4.6_latestBugfix_write_case0-uncommented.txt)

malansk, 2024-05-06:
One more thing, after restart of heating controller and BSB-LAN, I sometimes get "Verbindung Unterbrochen 1 - JA".
Maybe this is related to my uncommenting of case0 ? To get rid of that, I undid my change, restartet (which gave the error "ungültiger Anlagentyp"), did my change to uncomment case0 and everything was workinG again.

malansk, 2024-05-06:
One more thing, "15023 PPS-Bus - TWW Nennsollwert:" is not persistend. After restart it's at 0°C and I don't get warm water. After setting it manually to 55°C in WEB-GUI, the value gets transfered to the heating controller. But after next restart of BSB-LAN, it's again at 0°C. The value should be written to EEPROM after setting and read back from EEPROM after restart.
If I cange this value in the heating controller, BSB-LAN will not register this change. Looks like writing this value to PPS is ok but reading from PPS fails.

fredlcore, 2024-05-07:
Well, the reason the error is going away is that the heating system thinks that there is no QAA (i.e. BSB-LAN) connected and thus ignores any other kind of data from the QAA. That's why you also get the "connection interrupted" telegram with which the heating system informs any QAA to start sending telegram type 38 again. That's why it's case 0 because this has to be sent most often so that the heating system doesn't think the connection is interrupted.
So there is nothing buggy with that, but some other kind of telegram case in the message queue causes the problem. You could try to go through it one by one, but as you said that the "Anlagentyp" is not correct, it would be helpful to let me know which value it is at now that the error occurs?

For TWW Nennsollwert, I'll have to have a look, that should already be written to EEPROM, as it has FL_EEPROM flag set in BSB_LAN_defs.h. Room temperature is not written to EEPROM for safety reasons. Imagine your home automation which delivers the room temperature to BSB-LAN is no longer updating. Then the last known room temperature (say, 21.4 degrees) would continue to be written to the heater even if it is below zero already. This would at least become obvious when BSB-LAN restarts and does not get a fresh room temperature.
The other reason is that the room temperature usually changes most often and would thus lead to continous writes to the EEPROM, reducing its life-span. Since you have no other way to inform the heater about the room temperature anyway (because in write mode you have no QAA anymore that could do it for you), you have to make sure that there is a continous and reliable source of room temerature.

fredlcore, 2024-05-07:
By the way, a room temperature of 0 sent to the heater is most likely the reason for the error, because your configuration requires a valid room temperature. 
In the unlikely case that it's not, check telegram types 4E, 55 and 56. These are the only ones that do not have a known function yet.

malansk, 2024-05-07:
I have set a room temperatur wie WEB-GUI but it does not change the Error of "Ungültiger Anlagentyp". I have disabled Room-Temperatur correction function in the heater controller since longe time because I don't have a good room for reference anyhow.

The controller is expecting "Anlagentyp [1...150]" but shows "0" when I have write enabled. 15 would be the correct type which is shown with write disabled.

I have uncommented all 4E / 55 / 56 telegrams without change. 

In the Serial Monitor it shows "Collision on the bus, retrying".
I did not find this in any of the include files or in the .io sketch - where is this "Collision" detected & what's the retry code doing ?

malansk, 2024-05-07:
[image]

malansk, 2024-05-07:
In pps_handlinh.h there is this line:
if ((msg[0] & 0x0F) == 0x0E) {   // Heater requests information from the QAA (i.e. BSB-LAN) with telegram type 0x1E

The value 0x0E does not seem to match the value in the comment 0x1E.
Which one is the right one ?

In V2.1.16 log I see several times:
"Unknown request: 1e 00 00 00 00 00 00 00 00"
Maybe V2.1.16 is not answering the 1E requests but 3.4.6 does and this could be the problem ?
How can I disable answering 1E requests ?

fredlcore, 2024-05-07:
Please don't try to look for solutions in the code if you don't really understand what's going on, for example that 0x1E & 0x0F is actually 0x0E.
The only time BSB-LAN sends a zero (matching your wrong Anlagetyp) is with telegram type 0x4E. So you might want to change line 34 to 0x0F to see if that makes a difference. 
Also check what is necessary for the heater to go back to normal. Do you have to turn it off and on again or is it enough to set BSB-LAN to read-only? Maybe it takes a bit of time for the uncommented/changed telegram types to take effect?
Otherwise the only solution would be to disable each sent telegram one by one, starting from the bottom (case 26).

malansk, 2024-05-07:
I did change line 34 to 0x0F but no difference.
The heater goes out of error mode after 5..10s after switching from write to read. I also did try several cold-start of heater and BSB-Lan.
I did uncomment each case, but the only thing that makes a difference is when I uncomment case 0 (QAA Type).
This makes me think that the case 0 is the actual problem with my system.

I made this test:
*case 0 activ (original condition)
*compile sketch, upload -> heater in Error mode, "Verbindung unterbrochen 0 - Nein"
*case 0 uncommented tx_msg's
*compile sketch, upload -> heater out of Error mode, "Verbindung unterbrochen 0 - Nein"
-> I can now use WEB-GUI to change heater mode, Presence, WW Temperatursollwert, ... so everything is working like in 2.1.6., only problem is that after next restart of BSB-LAN or heater controller is is shown "Verbindung unterbrochen 1 - JA"
Obviously my heater needs case 0 for first time initialization apower up, (but there is somewhere an error in the data leading to error message), but after initialization it is happy to accept any command from BSB-lan (except for the 38 - QAA type which throws immediate error).

malansk, 2024-05-07:
Could you please provide a pps_handling.h stripped down to the bare minimum needed for initialization (set QAA type to QAA50, set Betriebsmode to Manual, read one known working value from heater e.g. Außentemperatur, set one known working value to heater e.g. Warmwassersollwert) without any variables & if..then..else ... switch ... case ... logic ?
Just to confirm that the low-level initialization done by QAA type announcment is working ?

fredlcore, 2024-05-07:
Since you choose to ignore my knowledge about the PPS protocol, why should I invest hours to strip down the pps_handling.h so that it works the way you think it should work? I told you what you can do, and that is to eliminate the problematic telegram. The only telegrams that BSB-LAN sends are the `FD` telegrams, and I told you how you can disable them step by step. Maybe it's a combination of several telegrams, maybe its specific values that your heater doesn't like, who knows. If you have a QAA70, you can comare the telegrams that BSB-LAN sends with the ones the QAA sends and then figure out which telegrams need to be changed. 

Your heater will work without a `38` telegram for a few minutes, but none of the known heaters do longer than that. It is basically the heartbeat of the QAA, and since you yourself admitted, it's the same telegram that gets sent in 2.0.16.

To repeat it again: The problem stops when you disable case 0 because none of the other telegrams will be accepted by the heater, even though you keep sending them. And don't get confused by the fact that the values you change are visible in the webinterface. They will always do, even if you disconnect the heater. But check a few minutes after you disable case 0 whether these values can still be checked on the heater itself. As I said, after a few minutes (once "Verbindung unterbrochen 1 - JA" comes up), none of the values will be accepted anymore.

fredlcore, 2024-05-07:
In your last log (SerialMonitor_3.4.6_latestBugfix_write.txt) there are a number of parameter values that can cause the error on the heater:
```
PPS ANS 15023.0 PPS-Bus - TWW Nennsollwert: 0.0 Â°C
FD 0B FF FF FF FF 00 00 FC
PPS ANS 15022.0 PPS-Bus - TWW Reduziertsollwert: 0.2 Â°C
FD 1E FF FF FF FF 00 0A DF
PPS ANS 15006.0 PPS-Bus - Frostschutzsollwert: 0.1 Â°C
FD 1B FF FF 00 00 00 06 E4
PPS ANS 15008.0 PPS-Bus - Raumtemperatur: 0.0 Â°C
FD 28 FF FF FF FF 00 00 DF
```
So if you want to work on a fix, fix these values first by setting them to valid parameters. The 0.2 and 0.1 values are a bit odd and may result from an issue in the EEPROM, but you can try setting them to sane values first and see what happens. Take note that in PPS mode, _all_ parameters that can be set are sent, continuously. The Set-button only stores them in BSB-LAN's memory/EEPROM. So just because you don't set a specific parameter doesn't mean that the parameter is not sent to the heater.

fredlcore, 2024-05-07:
I have now fixed the TWW-Reduziertsollwert issue and also added debug warnings if some of the main temperature values are not above 0.

malansk, 2024-05-07:
Thanks, I have tried it out but was not able to get it to send values because its complaining of invalid temperatur values.
I tried to set everything, but TWW Reduziertsollwert can't be set (SET Button missing).

malansk, 2024-05-07:
[image]

malansk, 2024-05-07:
pps_values[PPS_RTS] != 0 might not work since this value is calculated by the heater and send to the QAA for info only, it can't be changed by QAA. As far as I know QAA tells the heater Anwesend Ein/Aus and from this the heater is giving RTS as output (depending on Komfortsollwert and Reduziertsollwert).

fredlcore, 2024-05-07:
Good call, removed TWW-Reduziertsollwert in the check, so for now only room temperature comfort/reduced and TWW Nennsollwert are being tested against a value of 0.

malansk, 2024-05-07:
It's still complaining. Maybe these values are after all 0 ?
How do they get populated - from EEPROM ? In the WEB-GUI the values are shown & they are persistend after restart.
Will test some more after dinner.

fredlcore, 2024-05-07:
Sorry, I made a mistake there, of course the test must check for equal to zero, not the other way round. No need to download all again, just replace the line in 245 with this one:
`      if (pps_values[PPS_RTI] == 0 || pps_values[PPS_RTS] == 0 || pps_values[PPS_RTA] == 0 || pps_values[PPS_TWS] == 0) {`

malansk, 2024-05-07:
The check is now working, no error after start (write mode disabled) but error as soon as I enter 21 degree as room temperature.

I did now uncomment all tx_msg from case1 until case26 but I still get the error.
However, now the heater controller switches from error to no error several times every few seconds.
Maybe not the content is the problem but the way how the pps_handling.h is called from the main program ?
Since I use the Arduino Due which is quite powerfull maybe the bus messages are transmitted to fast to too many at the same time overflowing some receiver buffer in the heater controller ?

malansk, 2024-05-07:
[pps_handling.zip](https://github.com/fredlcore/BSB-LAN/files/15238546/pps_handling.zip)
[SerialMonitor_3.4.6_write_case0only_PPSBUSERROR.txt](https://github.com/fredlcore/BSB-LAN/files/15238551/SerialMonitor_3.4.6_write_case0only_PPSBUSERROR.txt)

fredlcore, 2024-05-07:
You comment out the right lines as far as I can see quickly, but I see now that commenting out will still send the telegram, just one with a wrong checksum (which should just be ignored, not raise an error).
Try this:
Change this line:
` if (msg_cycle > 24 || (pps_values[PPS_QTP] == 0x52 && msg_cycle > 5)) {      // QAA50 sends fewer parameters`
to this
`    if (msg_cycle > 0 || (pps_values[PPS_QTP] == 0x52 && msg_cycle > 5)) {      // QAA50 sends fewer parameters`
That should only send FD 38 messages after a 17 request from the heater. The result should be that the connection is not interrupted, and the error might go away. It could still result in an error because now too few messages are coming, so you can increase the > 0 to > 1, > 2 etc. This will then send the telegrams from case 0, case 1 and case 2 (uncomment them before, of course). You can then change the contents of case 1 and 2 with other cases so that other telegrams are sent and see what happens.

If you really think that the code is the problem, just do a comarison between pps_handling.h between version 2.0.16 and the current version and adjust as you like and see if it changes anything (I'd bet it doesn't). 

Have you tried out using a different room unit model in the settings? Such as QAA50? That only sends 5 parameters, maybe this way it's easier to check? You still have to have valid values in the relevant parameters, though.

malansk, 2024-05-07:
Changing the msg_cycle from 0 to 2 gets rid of the FF FF FF ... telegram as well as the "Verbindung unterbrochen 1 - ja" message due to too few messages.
But the heater still goes into error mode.
I have tried every possible option as QAA type but it does not change the behaviour.
I have searched the sourcecode of 2.0.16 for the include files but there are none, the program structure is completely different, all the functions are directly coded in the BSB-LAN.ino sketch.

fredlcore, 2024-05-08:
Ok, the last thing I can think of then is to replace bsb.h and bsb.cpp from version 2.0.16 into the current version and see if that changes anything in case there are some weird timing issues involved. Otherwise, the only other way would be to install every version in-between to figure out which commit actually caused the error. By version I do not mean release versions, but when you go to "history" and see all the commits, you can click on the icon that gives you the state of the repository at the time of this commit. There you can download the repository as it was at that time, install it and see what happens. Take note that the BSB_LAN_config.h has changed sometimes over time, so you won't be able to use one for all the tests, but make sure that you set UseEEPROM to zero, so that you don't get a conflict with old EEPROM data.
Once you have figured out which commit actually caused your heater to go into error, I can have a closer look, but since I do not have a PPS heater myself and I just recently got confirmation from other PPS users that things are working fine with them, there is nothing else I can do unless you figure out which commit exactly causes the problem.

fredlcore, 2024-05-09:
I thought again about whether it could be a timing issue, but I'm pretty sure that this can't be it, because the previous timing issues led to the heating system not accepting any commands that BSB-LAN sent because the answers from BSB-LAN didn't come quickly enough after a 0x17 (request to send) telegram from the heater.
In your case, the telegrams are accepted by the heater. The 0x38 telegram is accepted because the heater reacts with an answer that the connection is no longer interrupted. And you also mentioned that you can set other parameters. So the general communication on a hardware level is seemingly working fine. So I'm still pretty much convinced that it is one of the added telegrams that is causing the error. After reducing the telegrams sent to just the 0x38 telegram, have you checked the value in "Anlagentyp" in your heater again? It could be that it now stays at "15" (or whatever your correct value is) but that the error on the heater now comes from the fact that it is not getting enough parameter data that it would expect (e.g. it would assume the QAA to be faulty if it doesn't send the room temperature).

These are just some more thoughts that I had. If you think that all this is not the case and it's rather rooted in the code, you have no other way but to update your version step-by-step to figure out which version broke it. As some kind of "milestones" you could check the versions from the following dates first:
2 February 2021 (2.0.81)
24 September 2021
26 October 2021
24 March 2022
4 November 2022
12 December 2022
Figuring out where the error occurs in between these versions would be a first step, then you can move back/forth to identify which commit caused the issues...

malansk, 2024-05-10:
Thank you very much for your support, in the meanwhile I have tried the following things.

1. Initialization of heater controller to actually start sending telegram on PPS bus
   I have realized that with PPS Read only mode, my heater is not sending out any data on the bus (not even 2.0.16 version) since 
  there is no "heartbeat" from QAA (I don't have a physical real QAA).
  So I always had to compile 2.0.16, run it in PPS write mode shortly to initialze the heater controller, compile new version (3.x).
  and do any trials.
 To reduce this pain, I have added hardcoded initilization to pps_handling.h:
========== 
if (bus->getBusType() == BUS_PPS) {
          byte tx_msg[] = {0xFD, 0x38, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x53, 0x7d}; //hardcode SEND(FD) QAAtype(38) QAA70(53) w. Checksum(7d)
          byte rx_msg[10] = { 0 };
          bus->Send(0, 0, rx_msg, tx_msg);
======
This successfully initailzes the heater controller even if PPS write is disabled and I get correct time and day shown in WEB-GUI als well as the temperature telegrams.

However, it also triggers the "Ungültiger Anlagentyp" error (Anlagentyp shown as "0" in heater controller).
Since the only thing send is the single "0xFD, 0x38, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x53, 0x7d" command, I think that the problem is comming from the low-level routines in bbs.cpp.
To verify this, I have exchanged the bbs.cpp and bbs.h with the old version from 2.0.16 but it's not working (compiler error) since there way a change from bool to int8 type for the Send () function.
I made a DIFF of the files bbs.h and bbs.cpp for your convenience and I tried to undo some of the changes in the new files but without getting rid of the error.
The differences as far as I understand:
*ESP32 support (should not be relevant since I use DUE) 
*new function BSB Disable Interface [took that out but no change of error]
*Support for additinal PPS telegram (F8, FB, FE) [took that out but no change of error]
*timing was changed to comply with BSB bus requiremens (min. delay  between 26ms .. 85ms changed to 63 .. 82ms [changed that back but no change of error]
*changed collission check with rx_pin while waiting  [changed that back but no change of error]
*omission of readbyte() after serial flush (read and discard the byte that was just sent to prevent it showing up in incoming buffer) [changed that back but no change of error]
*DC225 emulation (after sending 0x17, abort loop) [took that ou but no change of error]
*additional collision check on the bus (that's why I get the "collission on the bus, retrying"warning with the new version !) [took that out but no change of error]
*changed return value generation of the function [this part I was not shure how to undo - maybe you can help here ?]

malansk, 2024-05-10:
[Windif_BSB.cpp_V2.0.16_vs_V3.4.6.pdf](https://github.com/fredlcore/BSB-LAN/files/15277021/Windif_BSB.cpp_V2.0.16_vs_V3.4.6.pdf)
[Windif_BSB.h_V2.0.16_vs_V3.4.6.pdf](https://github.com/fredlcore/BSB-LAN/files/15277022/Windif_BSB.h_V2.0.16_vs_V3.4.6.pdf)
[BSB.zip](https://github.com/fredlcore/BSB-LAN/files/15277052/BSB.zip)

malansk, 2024-05-10:
Where is the link to that "History" page ?
On Github (https://github.com/fredlcore/BSB-LAN/releases) I only find 0.44 (working), 2.0 (which is in fact 2.1.8 from 31.07.2022 and already not working).

fredlcore, 2024-05-10:
Here: https://github.com/fredlcore/BSB-LAN/commits/master/

malansk, 2024-05-10:
[Windif_BSB.cpp_V2.0.16_vs_V2.1.8.pdf](https://github.com/fredlcore/BSB-LAN/files/15277132/Windif_BSB.cpp_V2.0.16_vs_V2.1.8.pdf)
[Windif_BSB.h_V2.0.16_vs_V2.1.8.pdf](https://github.com/fredlcore/BSB-LAN/files/15277133/Windif_BSB.h_V2.0.16_vs_V2.1.8.pdf)

fredlcore, 2024-05-10:
You might again be too quick in jumping to conclusions, as I explained above that the error could result from too few telegrams coming from BSB-LAN which it would expect to be functioning properly. I don't have time into looking all the stuff you send me here, I told you what you have to do, even though it means that you cannot offload the work to me but have to do it yourself. Figure out when exactly the problem occurs and I'll have a look at it, but without having your exact controller, and knowing that your problem is only your problem and not any other PPS user's so far, you're the one who will have to do the work here...

fredlcore, 2024-05-10:
Ok, since you get the "Collision detected" error sometimes, this was the only major change that also affected PPS, the rest of the changes didn't really affect PPS.
So, take the most recent version (that's what I ran a comparison against with the last version I have from April 2021), and look for:
```
  if (HwSerial == true) {
    serial->flush();
    unsigned long timeout = millis();
    while ((millis()-timeout < 50) && serial->available() == 0) {
      delay(1);
    }
    if (serial->available()) {      
      for (uint8_t i=0; i<=loop_len; i++) {
        char readdata = readByte();
        if (msg[i] != readdata) {
#if DEBUG_LL
          Serial.println(readdata, HEX);
#endif
          Serial.println("Collision on the bus, retrying...");
          delay(146);   // Wait the duration of 11 bits at 4800 bps times 32 (maximum telegram size) in ms (*1000) times 2 (because we can just keep waiting for an answer telegram) 
          while (serial->available()) {
            char c = readByte();
#if DEBUG_LL
            if (c < 16) Serial.print("0");
            Serial.print(c, HEX);
#endif
          }
#if DEBUG_LL
          Serial.println();
#endif
          goto retry;
        }
      }
    }
  } else {
```
and change it to
```
  if (HwSerial == true) {
    serial->flush();
#if defined(ESP32)
    unsigned long timeout = millis();
    while ((millis()-timeout < 50) && serial->available() == 0) {
      delay(1);
    }
    if (serial->available()) {      
      for (uint8_t i=0; i<=loop_len; i++) {
        char readdata = readByte();
        if (msg[i] != readdata) {
#if DEBUG_LL
          Serial.println(readdata, HEX);
#endif
          Serial.println("Collision on the bus, retrying...");
          delay(146);   // Wait the duration of 11 bits at 4800 bps times 32 (maximum telegram size) in ms (*1000) times 2 (because we can just keep waiting for an answer telegram) 
          while (serial->available()) {
            char c = readByte();
#if DEBUG_LL
            if (c < 16) Serial.print("0");
            Serial.print(c, HEX);
#endif
          }
#if DEBUG_LL
          Serial.println();
#endif
          goto retry;
        }
      }
    }
  } else {
#endif
```
and let me know if that changes anything.

fredlcore, 2024-05-10:
It might be that the above solution still does not work or produce different problems. In that case, please send a log worth of 1-2 minutes of data. 
The problem is that everything BSB-LAN sends out on the bus comes back in again at the same time, but it first ends up in the UART buffer. Only when that one is full, we get notified. The problem is that our telegrams are smaller than the buffer (at least IIRC), so we might get notified "too late". That's why we have the 50ms delay at the beginning of the code in the  previous post which then just reads and discards the data we just sent to the bus, so that the buffer is clear afterwards.  It could be that the 50ms delay after sending a telegram is already "too long" for your PPS system. Meaning, after we have sent our data, the heater has already started sending its next telegram within the 50ms pause for BSB-LAN. So when we then read and comare the data to what we have just sent, it doesn't match and therefore you get the "Collision" warning, which leads to BSB-LAN sending the previous telegram again - which then does not match the expectation of the heater.
So this would be a somewhat possible explanation of why there is a problem now compared to the past where we didn't wait ater a send. But if the problem persists after removing this clearing of the UART input buffer, then it will become difficult: Because we don't know how long exactly we would have to wait: It has to be long enough to receive the just sent telegram, but short enough not to miss the next telegram from the heater.
The only hope could come from the fact that we know the length of a PPS telegram sent by BSB-LAN/QAA, it's always 9 bytes. 9 bytes with 10 bits each (8O1) at 4800bps take 2.08 ms per Byte, so that would be 18.72ms. So you could change the 50 to 20 and see if that makes a difference. 
But now that I think about it, we may not even have to wait that long: After sending out the first byte, it will arrive in the receiving buffer of the UART while we send out the second byte, and the second byte will reach us while we send the third byte etc. So it would be possible that we only have to wait for 2,08 (let's make it 3) milliseconds. So as a next step (if 20ms doesn't do the trick), you could set it from 50 to 3.
In parallel, I'll try this with my LPB system at home and see if that does the trick as well.

malansk, 2024-05-10:
I was working in parallel with the history of the commit in 2022 and found that this commit from 23.03.2022 breaks it for my system:
https://github.com/fredlcore/BSB-LAN/commit/03b8aff9aab004cacf1c7548841405f01464426f

I have done the following changes to the most recent master version 3.4.6:
===============
pps_handling.h
/* //Martin Start Uncomment
	if ((msg[0] & 0x0F) == 0x0E) {   // Heater requests information from the QAA (i.e. BSB-LAN) with telegram type 0x1E [original version from 3.4.6]
	//*/ //Martin End Uncomment
	if (msg[0] == 0x1E) {   // Heater requests information from the QAA (i.e. BSB-LAN) with telegram type 0x1E [old version before commit from 23.03.2022]  

==============
bsb.cpp
// ... until SOF detected (= 0xDC, 0xDE bei BSB bzw. 0x78 bei LPB)
	//Martin Debug Start 
	/*
	if ((bus_type == BUS_BSB && (read == 0xDC || read == 0xDE)) || (bus_type == BUS_LPB && read == 0x78) || (bus_type == BUS_PPS && ((read & 0x0F) == 0x07 || (read & 0x0F) == 0x0D || (read & 0x0F) == 0x0E || read == 0xF8  || read == 0xFB || read == 0xFD || read == 0xFE))) {    // PPS telegram types 0x17, 0x1D, 0x1E, 0xF8, 0xFB, 0x$FD and 0xFE
    */ //Martin Debug End
	if ((bus_type == BUS_BSB && (read == 0xDC || read == 0xDE)) || (bus_type == BUS_LPB && read == 0x78) || (bus_type == BUS_PPS && (read == 0x17 || read == 0x1D || read == 0x1E || read == 0xF8  || read == 0xFB || read == 0xFD || read == 0xFE))) {    // PPS telegram types 0x17, 0x1D, 0x1E, 0xF8, 0xFB, 0x$FD and 0xFE
      // Restore otherwise dropped SOF indicator
      msg[i++] = read;
      //Martin Debug Start
	  /*
	  if (bus_type == BUS_PPS && (read & 0x0F) == 0x07) {   // PPS RTS telegram (0x17)? [original version from 3.4.6]
	  */ //Martin Debug End
	  if (bus_type == BUS_PPS && read == 0x17) {   // PPS RTS telegram (0x17)? [old version before commit from 23.03.2022]

With these changes I no longer get the Error Message and I am able to write setting from the Web-GUI to the controller :-)

malansk, 2024-05-10:
I have checked now how the heater controller reacts to temperature value settings of 0°C or negative °C for Raumtemperatur Istwert, Raumtemperatur Nennsolweert, Raumtemperatur Reduziertsollwert, Trinkwasser Sollwert, Trinkwasser Reduziertsollwert. It's not problem to send 0 or even negative values, this does not generate an error. The heater defaults in such cases to its own internal limit (0° / 4° / 8° / ...)
So it would be probably safe to take out the check for illegal temperature values since it might confuse some people why no write enabled (if they don't check the SerialMonitor).

malansk, 2024-05-10:
I have checked the different timings (50 / 20 / 10 /3ms) on the original (without my patches to undo commit from 23.03.20222) 3.4.6 version but it does not make the heater error (or the SerialMonitor message) go away.

I have then also check on my patched version. 50 / 20 / 10 does not make a difference - heater shows no error, i can set values, SerMon Message (Collission...) is still coming one time after startup.
3ms is giving me the heater error (ungültiger Anlagentyp) - probably its flooding the bus due to too small delay and this errormessage from the heater is just a general "PPS-BUS" error.

malansk, 2024-05-10:
Some more observations with my patched version:
*Setting of DAY of the week from GUI to heater is working
*Setting of TIME from GUI to heater results in change of DAY and time (get set to 00:00)
 Since this was read-only in 2.0.16 I can't tell if this behaviour is due to my patch or if it is a separate issue of my heater or of BSB-Lan. But I don't need this feature so it's no problem for me if this is not working.
*Außentemperatur gemischt and Vorlauftemperatur Sollwert have previously always shown 0°C, now the show 292°C and 220°C. Maybe the commit from 23.03.2022 should have only been done on PPS_MCBA device family and not on the standard PPS ? Vorlauftemperatur Sollwert is only available on PPS_MCBA as far as I know. Außentemperatur gemischt is available on my heater controller but not shure if this gets transmitted over the bus (probably not).

[50 more replies]