fredlcore/BSB-LAN #622: [BUG] Set date and time resetted after few times

barbuslex, 2024-02-28:
**BSB-LAN Version**
```h
#define MAJOR "3"
#define MINOR "3"
#define PATCH "2"
#define COMPILETIME "20231115201023"
```

**Architecture**
ESP32

**Bus system**
BSB

**Describe the bug**
The entry of the date and time is reset to zero after a few times.

**To Reproduce**
I do not observe this behavior if the heat pump control panel remains on the first date and time setting display at startup.
Steps to reproduce the behavior:
Power Off Atlantic Extensa Duo. Power On. The date/time is resetted to 01.01.2015 00:00:00 by default.
At startup, the panel request to set date and time. Cancel this display with return button.
1. Go to http://192.168.1.188/0
2. Default value is '01.01.2015 01:50:57'
3. Go to http://bsblanip/S0=28.02.2024_13:12:23
4. Go to http://192.168.1.188/0
6. The time is correctly set to '28.02.2024 13:12:23'
7. Wait 1-2 minutes
8. Go to http://192.168.1.188/0
9. The time is resseted to '01.01.2015 01:52:57'

**Log files - Bug reports without log files will be closed**
Don't have any logs

**Expected behavior**
When the date and time is set by http, i need it will be persisted.

**Screenshots**
[image]
[image]

**Desktop (if applicable, please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox
 - Browser version: 123

**Additional context**
Maybe my BSB_LAN_custom_defs.h (attached) file is bad because i use same that my neighborood who have same heat pump than me.
But it seems its seems in integrated control panel which send every 1-2 minutes the date/time : 
[image]

[BSB_LAN_custom_defs.h.txt](https://gith
[BSB_LAN_config.h.txt](https://github.com/fredlcore/BSB-LAN/files/14440554/BSB_LAN_config.h.txt)
ub.com/fredlcore/BSB-LAN/files/14440550/BSB_LAN_custom_defs.h.txt)
[BSB-LAN-211-127-137987.txt](https://github.com/fredlcore/BSB-LAN/files/14440571/BSB-LAN-211-127-137987.txt)

fredlcore, 2024-02-29:
You need to provide a serial monitor log as described in the manual, from the time the heater starts until the time is reset to the wrong date, after you changed it via BSB-LAN. Otherwise there is nothing I can do.

barbuslex, 2024-02-29:
Is it possible to monitor log from wifi ? Is not possible for me to connect ESP32 on heater pump and on my pc in same time :S

barbuslex, 2024-02-29:
OK, i have try but it seems that i have only log from ESP32 but no log from heat pump. I see in log the date/time set by bsb-lan but not the reset.
Here my settings : 
[image]
[image]
Should i activate this setting ? 
[image]

fredlcore, 2024-02-29:
Please read up in the manual what a serial monitor log is, it is not the same as logging parameters. There is a dedicated chapter in the manual: https://1coderookie.github.io/BSB-LPB-LAN_EN/chap12.html#122-serial-monitor
All you need to activate is Debugging / Usage / Telnet

barbuslex, 2024-02-29:
I have enabled this settings : 
[image]

I confirm, if i let the heat pump on startup display, the date/time is not resetted : 
[image]

I a push 3 time to the return button (3) for cancel and return on home display the date is resetted after 1-2 minutes. But i not have the log of the reset. In log file it's just refresh (F5) of the BSB-LAN /K0 page.

In the log file:
Set date/time: line 166
Show reset by F5 on /K0: line 3414
[monitor.txt](https://github.com/fredlcore/BSB-LAN/files/14447794/monitor.txt)

Tell me if it's good or not...

fredlcore, 2024-02-29:
Yes, that log file is the one I'm looking for, but please disable all kind of logging, MQTT etc., otherwise the file is way too cluttered to find anything.

barbuslex, 2024-02-29:
Here the new clean monitor log : 
[monitor2.txt](https://github.com/fredlcore/BSB-LAN/files/14451284/monitor2.txt)

fredlcore, 2024-02-29:
Hm, that's strange, also because there is no other communication on the bus. Normally, the display (which identifies as "DISP" in addition to "HEIZ" and "LAN" which is the heater and BSB-LAN respectively) should send a telegram every 10 seconds to the heater to either query the displayed value or to make changes for example regarding the time. What makes me wonder is that the date falls back to 2015. Normally, the time is set via the display unit on the heater. It seems to me that the display "forgets" the correct time, as if it had lost its backup battery or something like that, and then just uses the first date it know (i.e. 2015). Since there is "silence" in the log unless you are doing something with BSB-LAN, I can't really help because I can't see what/if the display unit is interfering in any way. From the BSB-LAN part, all seems to be fine...

barbuslex, 2024-02-29:
Yes, in fact I'd like to be able to send the date and time back to the heat pump after a power cut. I don't have this problem if I set the date on the display and use it. The problem occurs after a power cut. So I'd have to force this date and time onto the heat pump display instead. Is there any way of doing this?

fredlcore, 2024-03-01:
It's the same with me, and both are part of the BSB bus. Can you try and run the serial monitor log while you change one parameter to a different value and then back again? Let's say comfort temperature setpoint (710) to 24 degrees and then back to the original. It should be visible on the serial monitor. If not, then your system may be different from mine and then I don't know if you can "reach" the display unit.

In order to send directly to the display unit, you could try and add a !10 after the /S0, so that it's /S0!10=... and also run a serial monitor log when doing so.
The second option would be to send it as an INF, i.e.
/I0=....
Also run a serial monitor log here.
If none of this works, then I'm at my wit's end...

barbuslex, 2024-03-01:
Ok i need to buy a long usb cable to do it. I will buy it and i will try. I will reply on next message in few days when i have the monitor log.

barbuslex, 2024-03-01:
Here the log : [monitor.txt](https://github.com/fredlcore/BSB-LAN/files/14466455/monitor.txt)

I have try to set year to 2025.
Set comfort temperature setpoint (710) to 24 degrees : OK
Set comfort temperature setpoint (710) to 21.5 degrees : OK
/S0!10=01.03.2025_23:02:32
set failed
/I0=01.03.2025_23:02:32
return nothing

fredlcore, 2024-03-02:
Sorry, I meant to change these parameters on the display and have the serial monitor running while you change it there...

barbuslex, 2024-03-02:
Ok i change just comfort temperature or date/time too ?
I tried the date and time in the previous monitor it seems and it displayed nothing. But i can retry.
I will test in 1 week because i'm not in my house actually. I will post the monitor in a next comment.

fredlcore, 2024-03-02:
Just the temperature is enough. I know you changed the time in the previous log, but since nothing of that showed up in the monitor log, I wanted you to try it with a different parameter. If that one also does not show up in the monitor log, then the display is somehow detached from the rest of the BSB bus - which is strange, but could be possible. But it would mean that you have no way of accessing the display unit and change its time. The last resort would be if you have a room unit. What happens if you change the time from there? If you have one, then changing the time via the room unit while you are capturing the serial monitor output would be the last thing I could look at.

barbuslex, 2024-03-02:
Oh.. ok i have understand... i will test when i return to my home.
I have 2 room display like this : 
[image]
But i can't change date/time with them just temperature and climate hot/cool.

fredlcore, 2024-03-02:
Yes, these are the "simpler" ones, unfortunately...

barbuslex, 2024-03-02:
What's strange is that when I change the temperature or turn the room thermostat on/off from the BSBLAN, it works. The heat pump display updates with the correct values after a while.

fredlcore, 2024-03-03:
No, unfortunately, that is not strange at all. The controller (RVS21) is the authoritative device in all cases _except_ for the time and date settings. Since it works that you change parameters via BSB-LAN on the RVS21, these will be reflected on the display as well because the display gets these values from the RVS21. The time is the only thing that the display "knows" and which it sends to the heater. But since we do not see any activity of the display on the bus, it must somehow be isolated from the rest of the bus. We'll know this for sure once you have made the last tests that I suggested.

barbuslex, 2024-03-13:
I can test at the end of the afternoon.
I thought about this scenario.

- On the heat pump display, set:
  ```
  Zone 1 temperature: 19 °C
  Zone 2 temperature: 19 °C
  ```
- On the zone 1 room display, set the temperature: 24°C
- On the zone 2 room display, set the temperature: 24°C
- From the BSB-LAN:
  ```
  Zone 1 temperature: 21 °C
  Zone 2 temperature: 21 °C
  ```

Is it a good test for you ?

fredlcore, 2024-03-13:
Yes, the important thing is that you have the serial monitor running when making all of these changes. I need to be able to see the telegrams that appear in the log when you make changes both on the heat pump display as well as on the room display(s).

barbuslex, 2024-03-13:
[monitor.txt](https://github.com/fredlcore/BSB-LAN/files/14591063/monitor.txt)

It seems the serial monitor not log the bus :S Just the BSBLAN actions are logged.

Maybe its because i capture log from telnet ?

fredlcore, 2024-03-13:
Yes, it seems that the connector to which you have connected BSB-LAN is isolated from both the heating unit's display and the other room units. This is really strange and new to me, but it has nothing to do with the telnet connection. Both serial and telnet just listen to what is on the bus, and they only see the messages going to and from the heater (which is most important of course), but nothing from the other devices. Can you post a photo so I can see where BSB-LAN is connected to and where the zone room display units are connected to?

barbuslex, 2024-03-13:
Global installation : 
[image]

Focus on X86:
[image]

barbuslex, 2024-03-13:
I just remembered that I changed a part behind the display following a call to the manufacturer Atlantic so that they could keep the date and time in the event of a power outage.

They are differentiated by the gray cylinder (capacitor or battery) which allows the date and time to be retained for around ten minutes in the event of a power outage (see photo).

At the top the new part and at the bottom the old one : 
[image]

Maybe I should put the old one back?

fredlcore, 2024-03-13:
Ah, ok, the zone units are connected to the AVS55. It may make sense that these do not show up on the bus because the AVS55 is connected directly to the underlying RVS21, so the messages may not go on the rest of the bus. The same could be true for the heater display, but that is still kind of new to me. I don't think that the device on your photo has anything to do with it, but on the other hand, I've never seen that either.
Bottom line is that as long as you don't see any data telegrams on the serial monitor when changing parameters on the main display, you can't change it, unfortunately...

barbuslex, 2024-03-13:
It's not possible to connect BSBLAN on AVS55 next to the other zone display (in purple on the global photo) ?

fredlcore, 2024-03-13:
I haven't found any documentation on this connector. If it measures around 12-15V you can safely try without breaking the adapter. Then you could try again and see if there is any activity on the bus once you are changing parameters on the zone units...

barbuslex, 2024-03-13:
The other zone display is connected to it I think it is the same voltage.

barbuslex, 2024-03-13:
I have found this documentation about AVS55 (in french). The Atlantic reference of this part is 075311.
http://www.iso-chauffe.fr/Files/27354/alfea-extensa-duo-alfea-excellia-duo-kit-carte-extension-notice-atlantic.pdf

fredlcore, 2024-03-13:
Looks good, the manual mentions "Sonde d'ambiance circuit mélangé (option kit 2 circuits)", and if my rusty French of over 25 years is still any good, then this means something like mixed circuit room temperature sensor, and if the voltage is the same, then I'd try and see what you can get...

barbuslex, 2024-03-13:
[monitor2.txt](https://github.com/fredlcore/BSB-LAN/files/14592040/monitor2.txt)

Same here. I can chage temperature from BSBLAN but i can't trace any bus telegram from heater or zone.

fredlcore, 2024-03-14:
Did you connect it to the exactly the same pins as one of the zone room units? If so, it must display something when you change a parameter on that unit. You may not see anything the main heater display does, but at least activity from the zone unit must be visible...

barbuslex, 2024-03-14:
Ok i will try this weekend. I will try with USB too.

barbuslex, 2024-03-28:
I plugged the ESP32 into the same port as a room screen. I tried capturing from USB but I don't seem to have any more information other than this:
```
15:35:00.126 -> 60002 Ping!
15:35:03.219 -> DC C2 00 0B 06 3D 05 00 64 5E E8 
15:35:03.219 -> Device family: 0
15:35:03.219 -> Device variant: 0
15:35:03.219 -> GET /0 HTTP/1.1
15:35:06.316 -> DC C2 00 0B 06 3D 05 00 0B C3 A1 
15:35:06.316 ->  query failed
15:35:09.388 -> DC C2 00 0B 06 3D 05 00 0B C3 A1 
15:35:09.388 ->  query failed
15:35:12.444 -> DC C2 00 0B 06 3D 05 00 0B C3 A1 
15:35:12.444 ->  query failed
15:35:12.444 -> 0
15:36:00.141 -> 120003 Ping!
15:36:03.172 -> DC C2 00 0B 06 3D 05 00 64 5E E8 
15:36:03.219 -> Device family: 0
15:36:03.219 -> Device variant: 0
15:37:00.103 -> 180004 Ping!
15:37:03.178 -> DC C2 00 0B 06 3D 05 00 64 5E E8 
15:37:03.178 -> Device family: 0
```

fredlcore, 2024-03-29:
How are your settings for verbose and monitor in the configuration?
What surprises me is that BSB-LAN recognizes DHW and burner status messages, but does not display the telegrams. Also, I don't understand why you cannot obtain the device family and variant. These queries should not fail, unless there is no "path" to the central heater. But there should be, because the room unit needs to "talk" to the central heater.

barbuslex, 2024-03-29:
Here my settings : 
[image]

The last log was captured (when see date) with this option : 
[image]

fredlcore, 2024-03-29:
Strange. The settings are correct, and the PPS mode does not apply to your system. 
I don't know what's going on there, but if you can't even see anything when connected to the room unit connector and and change or query parameters via the room unit, then I'm at my wit's end :(...

barbuslex, 2024-03-30:
Could this be due to a faulty ESP32?

I have buy this one : https://www.amazon.fr/dp/B07XH45MWW?psc=1&ref=ppx_yo2ov_dt_b_product_details

fredlcore, 2024-03-31:
No, that one looks ok - if you can set other parameters, everything works fine on the BSB-LAN side because that means you can send and receive data. That works as seen in your monitor2.txt further up. A QUR telegram is a query that you send from BSB-LAN to the heater, and an ANS telegram is the answer that you get from the heater. So on that level, technically is fine.
But usually, the BSB bus is the same across all connectors, so no matter which "FB", "CL+/-" etc. labelled connector you are using, you should always see all messages sent and received by all connected devices such as room units, heater display etc. For some reason that is not the case here with you, and there are no QUR/ANS telegrams for example between DISP (display) and HEIZ (central heater), you only see the ones between your own (LAN) and the heater (HEIZ).
I would have thought that maybe your heater is different in that way that the various BSB connectors are not connected to each other. But at least if you plug yourself into the very same connector to which the room unit is attached to, you must be able to see what the room unit sends/receives.

One last thing you can try is to connect BSB-LAN without being powered by USB. If you connect it to the BSB connector, the red LED should turn on even if the ESP32 is not powered on. If it is connected to the same connector as the room unit, the LED should flicker slightly each time you read another parameter on the room unit, in the same way it flickers when you queried parameters from BSB-LAN before. If that is the case, then we know that the problem must be somewhere with your setup. If the LED is just on all the time without any flicker, then the hardware of your heater must be designed in such a way that room unit and BSB-LAN are somehow magically separated in a way that I can't explain to myself...

fredlcore, 2024-04-23:
Are you still going to try the suggestions I made? Otherwise, I'll close this issue.

barbuslex, 2024-04-23:
Yes i will try this weekend.

fredlcore, 2024-06-04:
Any news from your side? Otherwise, I'll close this issue.

barbuslex, 2024-06-05:
I'm sorry, I haven't had time lately.
I'll try it tomorrow evening or this weekend.

barbuslex, 2024-06-13:
I was finally able to take the test! Here is the video: https://www.youtube.com/watch?v=WWFnyWNKBJQ

fredlcore, 2024-06-13:
Ok, this clearly shows that something is going on on the bus, but without a serial monitor log at the same time, I don't know if that's because of your activity or because there are other bus telegrams going on. Which connector did you connect BSB-LAN to?
So what you need to do is to look at the serial monitor and at the same time at the flickering of the LED. Then mark which telegrams match your activities (such as changing temperature). If you change something and the LED flickers, but you don't see anything on the serial monitor, then something is wrong beyond my knowledge.

barbuslex, 2024-06-14:
The BSB-LAN is connected like this pictures : 
[image]

Could the blinking be related to sending messages via MQTT? Because I did not disable MQTT sending during my tests.

fredlcore, 2024-06-14:
The LED means bus activitiy, irrespective of the source. So yes, it could be something that BSB-LAN is logging or the home automation system is querying from BSB-LAN, but it could also be bus messages independent of BSB-LAN. In any case there should be serial monitor activity whenever the LED is flashing.

barbuslex, 2024-06-14:
Ok I will try to do both simultaneously and I will send you everything.

barbuslex, 2024-06-23:
I have tested Yesterday but its same i don't have any logs... Thé led
blinking but no logs 🥴

fredlcore, 2024-06-23:
Ok, this is impossible. If the LED is flickering, then _some_ kind of data must be visible on the bus. BSB-LAN would even unknown telegrams if that was the case. But without having my hands on the device, there's nothing more I can do from here, so I guess, we'll have to close this. But feel free to come back here if you have any further information.

closed 2024-06-23 by fredlcore

barbuslex, 2024-06-24:
I have see it : 
https://community.home-assistant.io/t/read-modbus-with-esp32-and-esphome/151638

Maybe there are another way to read bus (directly from windows) ?

barbuslex, 2024-06-24:
@fredlcore i have see this device in Aliexpress : https://fr.aliexpress.com/item/1005005321189496.html
Have you an idea which chip is better between CH340, PL2303 and FT232 ?

fredlcore, 2024-06-24:
I don't know what you want with that. Why should the bus you are connecting to be ModBus or RS485?
We have established above that you can set parameters and that the heater responds to your set commands, right? The only two problems are that you cannot set the time and that for some unexplicable reason the messages from the heater cannot be seen on the bus. But that doesn't mean it's a different type of bus because otherwise you would not be able to set/read any parameter.

One last thing you can try is to set BSB-LAN into monitor mode. It will not function properly then, but you will be able to see any data that comes in. You can activate this in the settings and then set some parameters on the heater, and if the BSB-LAN LED flickers, you should be able to read some data on the serial monitor. If that isn't working, then I really have no idea what could be wrong.

barbuslex, 2024-06-24:
[image]
It's already in monitor mode.

The aim with the device is to try to do a serial read from the USB directly on windows to see if it's the esp32 that might have a problem or not. For the price of the component, it might be worth a try.

fredlcore, 2024-06-24:
Has monitor mode been on all the time? Then that might be a reason and there is a bug somewhere. Then you could turn it off and try again (usually, only verbosity mode is on, monitor mode is off).
I don't think the ESP32 has a problem, because in the log above, you could see the telegrams that you sent from BSB-LAN to the heater and the answer coming from the heater to BSB-LAN:
```
LAN->HEIZ QUR  0.0 Heure - Date actuelle / Heure actuelle:
DC C2 00 0B 06 3D 05 00 0B C3 A1
HEIZ->LAN ANS  0.0 Heure - Date actuelle / Heure actuelle: 28.03.2024 23:44:46
DC 80 42 14 07 05 3D 00 0B 00 7C 03 1C 04 17 2C 2E 00 FC EF
```
The first telegram came from BSB-LAN (`QUR`), and the heater has received (`ANS`) and the answer was displayed. So that should be fine.

If you really want to try this, you would need a TTL-to-Serial-USB dongle (like in your second post, I think) and then connect it with the RX/TX pins marked on the back of the BSB-LAN adapter. You would also still need to provice 3.3V and GND to the adapter to make it work. The transmission speed is 4800 bps, 8 data bits, odd parity, 1 stop bit.
But this is not something I can help you with any further. If you really find out something new, then let me know, but this is taking a lot of my time already...

barbuslex, 2024-06-24:
I managed to capture this by trying to switch the cold mode on and off on the room unit. (attached)
[log.txt](https://github.com/user-attachments/files/15960164/log.txt)
[log2.txt](https://github.com/user-attachments/files/15960165/log2.txt)

fredlcore, 2024-06-24:
Yes, that makes sense. I'd expect something similar when you set the time or change any other parameters. That's what confuses me.