fredlcore/BSB-LAN #632: [BUG] SD card vs. flash memory

DE-cr, 2024-03-15:
current version from master repo
ESP32 w/o SD card, BSB

On the config page, the text always reads "Freier Speicher auf SD Karte", even when "Speicherziel" for logging is set to internal flash memory and the remaining space in the flash memory is displayed here.

Maybe it would make sense to have a "General: Use for files" setting instead of "Logging: Speicherziel", and to display "Free files space" instead of "free space on SD card"?

Also, the system may not always distinguish between SD/flash where it should/could:
It seems I had accidentally switched logging to SD card on my system (which doesn't have one) yesterday. Today I realized that logging had stopped and that free space on SD card is displayed as 0. However, /DG still showed me the data logged to flash memory (which was far from full, btw) and for that also used the javascript libraries received from the flash memory via the "external" webserver.

fredlcore, 2024-03-15:
As for renaming the terms, makes sense, feel free to go ahead.
The system does not do sanity checks and/or switch to flash memory option if there is no SD card attached. In any case, if SD card is selected, it should not access the flash memory. If it does, it is an error, but I don't have time to investigate this one as I'm working on more serious errors right now...

fredlcore, 2024-04-02:
I've had a look at it, and it seems that the switch between the storage devices might require a reboot. Or rather, that one switch in one direction works, but not a subsequent one. Which is strange because the SDCard object is just switched between SD_MMC and LittleFS in `startLoggingDevice()`. @dukess, do you maybe have an idea why this switch does not work?
If we can't solve this, then I can just flag this setting to require a reboot, but if it's generally possible to switch without a reboot, I'd like to keep that.

fredlcore, 2024-04-06:
Ok, I think I found closing/opening the handle before assigning the reference solves the problem. At least I can now switch seamlessly and without rebooting between the two storage devices.

closed 2024-04-06 by fredlcore

dukess, 2024-04-06:
Frederik, sorry for late answer: i have lot of work, so have time on weekend only :(
But i glad to see you resolving the issue :)

fredlcore, 2024-04-07:
@dukess: No worries, thanks for getting back :)!