fredlcore/BSB-LAN #640: [BUG] Heater function list disappeared in version 3.4.3

guillaum111, 2024-03-29:
Hi, 

I have an issue witrh the last version of BSB Lan (3.4.3). All heater function list are empty and i receive binary code on my MQTT server. Switching back to version 3.3 work. 

I use the same custom_def for both and use the config file according to the version. 

Bus: BSB
HW: Nodemcu ESP32 

I try to update because set a parameter is not working. When i try to send a temperature on the S10000, i get the message : "ERROR: set failed!". If it's not solved with the update, i will open another issue. 

Thanks for your great job.

fredlcore, 2024-03-30:
There is a template for a bug report. Why did you choose to remove all the requested information? It also says to install the most recent version. If you had done so, you would have seen that the bug has already been fixed.
Parameter 10000 can not be written with /S but only with /I, see chapter 5, URL commands.

closed 2024-03-30 by fredlcore

guillaum111, 2024-03-30:
Hi, 

Sorry for the template, I will respect it now. I made a copy of the git on the 28/03 and didn't see that an update appear in the meantime. 

Ok, when you use the set button from the heater function list, it redirect to the link with /S.

fredlcore, 2024-03-30:
Yes, that's why it says "Set" - BSB-LAN does not know which parameter requires which kind of message type, and sometimes both can be used. Most of the time, it's the SET command (/S) that's why we added this one in the web interface. But if you know that your parameter requires an INF command, then you have to do this manually via /I

fredlcore, 2024-03-30:
BTW, I have now added an internal switch between SET and INF if the parameter has the SPECIAL_INF flag set, as this indicates that an INF telegram is required anyway. So now it should be possible to transmit a room temperature also via the webinterface's "Set" button.

guillaum111, 2024-03-31:
Hi, 

Thanks for your fast adaptation i made the update to 3.4.4-20240331020948 but when i push the set button on the parameter 10000, it's still the url with the /S, is there an option to enable it somewhere that i missed ?

fredlcore, 2024-03-31:
It still uses /S, but internally it sends an INF command which you should be able to veryify in that there is no response coming back after you hit  "Set".

guillaum111, 2024-03-31:
I'm still receiving the "ERROR : set failed !" as before. I double check the 8740 parameter to ensure that the value is not registered and it's not registered.

guillaum111, 2024-03-31:
Will do this evening when i come back home.