fredlcore/BSB-LAN #642: [BUG] Set button is not working on parameter that require /I in the we interface

guillaum111, 2024-04-01:
Hi,

BSB-LAN Version : 3.4.4-20240331020948

Architecture : ESP32 NodeMCU

Bus system : BSB

Description of the bug : 

Set a parameter that requires an INF message is not working from the web interface

To Reproduce : 
Write access need to be granted
1. Open the web interface 
2. Click on « Heater Functions »
3. Click on « User defined »
4. Wait for the list to be loaded
5. Enter a value on in the field of the parameter 10000 (20.0 in my example) 
6. Click on set to apply the value
 The url http://<ip_address>/S10000=20.0 is called and the message « Error : set failed ! » appear. The paramètre is not applyied 

Expected behavior : Parameter is applyed

**Screenshots**
If applicable, add screenshots to help explain your problem.

Desktop (if applicable, please complete the following information):
 - OS: Windows 11
 - Browser :Chrome
 - Browser version : 123.0.6312.86
 
[log.txt](https://github.com/fredlcore/BSB-LAN/files/14821986/log.txt)
[BSB_LAN_config.h.txt](https://github.com/fredlcore/BSB-LAN/files/14822003/BSB_LAN_config.h.txt)

fredlcore, 2024-04-01:
Ok, thanks, that should not happen, I'll take a look at it...

fredlcore, 2024-04-01:
Fixed it. Please download most recent version and try again and change parameter 10000 in BSB_LAN_custom_defs.h to this:
```
{0x2D3D0215,  VT_TEMP,            10000, S10000,   0,                  NULL,         DEFAULT_FLAG+FL_WONLY+FL_FORCE_INF, 255, 255},
```

guillaum111, 2024-04-01:
Everything is working fine, does it mean that we need to add the "FL_FORCE_INF" declaration to  all the parameter that require an INF ?

fredlcore, 2024-04-02:
Well, so far we only know of the room and outside temperature parameters that require an INF when you set them. All other parameters usually require a SET, so there should be no need for you to change anything.

closed 2024-04-02 by fredlcore