fredlcore/BSB-LAN #653: [BUG] v3.4.7 crashes ESP32/Wifi/BSB when submitting room temperature

schlaubstar, 2024-05-29:
**BSB-LAN Version**
crashing: 3.4.7 20240514013417
working: 3.4.6 20240504101907

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
Initially I assumed that the device runs in reboot loops but after observing the behavior in the debugger I found out, that the reboot-loops are caused by the room-device emulation where I do submit the actual room temperature in 3 min intervals. So the actual bug is, that submitting the room temperature to parameter 10000 causes a reboot of the device.

**To Reproduce**
Flash 3.4.7 20240514013417, and power up and submit a temperature value to parameter 10000

**Log files - Bug reports without log files will be closed**
see attached log file

**Expected behavior**
device should continue to accept the submission of a room temperature

**Screenshots**
not applicable: before device locks up/reboots, everything is normal ;-)

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**

[DebugLog_crashing.txt](https://github.com/fredlcore/BSB-LAN/files/15482212/DebugLog_crashing.txt)
[BSB_LAN_config-anon.h.txt](https://github.com/fredlcore/BSB-LAN/files/15482293/BSB_LAN_config-anon.h.txt)

fredlcore, 2024-05-29:
Thank you, even though it is not the most recent version of BSB-LAN, I can nevertheless reproduce the problem. Execution is stuck in a loop and after the watchdog timeout expires, the device reboots. I'll look into it.

schlaubstar, 2024-05-29:
terribly sorry for not using the most recent version but I am glad, that you can also reproduce it ! ;-) Thanks a lot !

fredlcore, 2024-06-04:
Sorry, fixed this with https://github.com/fredlcore/BSB-LAN/commit/8c2446ad606a35cf15500a2d0862d59cc297e628 and https://github.com/fredlcore/BSB-LAN/commit/5caa0d4899c554f4046f41fa5f7db70a2d237f86 already a couple of days ago without getting back to you. You'll have to change the SPECIAL_INF flags in your BSB_LAN_custom_defs.h as described in the changelog at the beginning of BSB_LAN.ino, i.e. add FL_SPECIAL_INF for parameters 10000, 10001 and 10002 and remove it everywhere else it might have been before (such als 10017).

closed 2024-06-04 by fredlcore

schlaubstar, 2024-06-11:
Thanks for the fast response: Adding the FL_SPECIAL_INF-Flag to 10000-10002 solved the problem.
My custom_defs.h does neither show 10017 nor any other occurance of FL_SPECIAL_INF so in my case it was really only adding it