fredlcore/BSB-LAN #654: Migrating BSB-LAN to ESP32 framework 3.0.0

fredlcore, 2024-05-30:
Since a few days, the Arduino IDE now installs version 3.0.0 as the default ESP32 framework. This comes with a long, long list of breaking changes, the one most noticable is that the OneWireNg library that comes with BSB-LAN no longer compiles.

For the moment, this means that BSB-LAN can currently only compile under version 2.0.17 of the ESP32 framework. A compiler warning/error has been added, so that users will hopefully be able to fix this themselves without opening dozens of bug reports.

However, while we can make this a requirement, it would probably be better to go with the flow and adjust BSB-LAN if possible.

Apart from the libraries, I assume that a number of other functions will also generate problems, especially when it comes to networking. Currently, the seamless change from Ethernet to WiFi and back again (which for example allow the setting up of the BSB-LAN accesspoint if the ethernet function is not working) relies on the fact that the EthernetClient object is derived from the WiFiClient object. However, this seems to have changed with 3.0.0.

There is a migration guide here:
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

Currently, I haven't found a way to install 3.0.0 under PlatformIO/VSCode, so I can only reproduce the errors in the Arduino IDE, and thus not really start on working on the code changes.
Maybe someone else has time capacities and knowledge how and what to change here? Ideally, BSB-LAN should still work under 2.0.17, at least as long as 3.0.0 is also available under PlatformIO. But if there has to be a breaking change, there has to be a breaking change, of course...

Looking forward to any kind of help or suggestions!

fredlcore, 2024-06-03:
I have now done (hopefully) all adjustments myself which mainly affected these three areas:
- ESP32 with LAN and fixed IP
- ESP32 with LAN and DHCP
- ESP32 with DS18B20 temperature sensors

Could anyone who meets any of the above criteria report back to me if things continue to work as expected with the most recent version from the master repository? Thank you!

DE-cr, 2024-06-08:
Fyi: On my naked esp32, build 4.0.11 using board 3.0.0 hang a while in the version check on bsb-lan/, then always said that there's no newer version. Maybe httpclient.available() never became true? With 4.0.13 and 3.0.1, this seems to work again.

fredlcore, 2024-06-08:
Oh, good to know! The networking component of framework 3.x seems to have been completely rewritten, and I haven't even begun to understand how that affects all the component BSB-LAN is using. For the moment, I'm glad that it compiles without any errors ;)...

DE-cr, 2024-06-09:
I experienced this delay again. Seems to be the case only when there *is* a newer version available.

fredlcore, 2024-06-09:
Hm, but it should be the same either way because all it does is retreiving one very small web page and parse it. Maybe you could let it output the version it received from the server and the current version so that it's also more obvious what kind of change it is (patch level vs. minor/major)? Then it would be obvious if there is an issue in transmission.

DE-cr, 2024-06-15:
Fyi: The issue is somewhere in this call to a library function in `include/print_webpage.h:139`:
`httpclient.connect("bsb-lan.de", 80);`
This line frequently hangs and then fails for me since upgrading to esp32 board library 3.0.x
(...and the following code then reports that there's no newer version available.)

fredlcore, 2024-06-15:
That's true, but that would mean your site has problems connecting my server. It is more under load at night when it's performing backup tasks, but I've just called my BSB-LAN's index page a dozen times and it's always reported the correct answer for me (which is that I need to update). You can test it by adding
```
Serial.print(c);
```
after 
```
    while (httpclient.available()) {
      char c = httpclient.read();
```
and you should see the contents of each line of bsb-version.h up to the quotation mark (after which the version number is parsed). I don't know if you can extract some kind of error number/mesage from the `httpclient` object, but that would be necessary to further troubleshoot the issue.

DE-cr, 2024-08-11:
Fyi:
Recently I've updated my https://github.com/DE-cr/BSBmonCR, and used esp32 framework 3.0.3 for it.
Some 8-9 days after installing the update, calls to external web sites from BSBmonCR stopped working.
My "fix" to this problem for now:
- go back to esp32 framework 2.0.17 for BSBmonCR, to get back its full functionality
- disable version check in BSB-LAN

DE-cr, 2024-08-16:
Fyi: Seems the 3.x framework was not the problem with BSBmonCR, but rather the target web server in question suddenly taking much longer to reply, i.e. I now have to wait for its reply becoming available. This change in behavior just happened to more or less coincide with my move to the new esp32 framework.

fredlcore, 2024-09-15:
Ok, then we can close this, I guess...

fredlcore, 2024-09-15:
Ok, good to know.
Still waiting for someone to test the DS18B20 sensors with the current version...

fredlcore, 2024-11-01:
As keeping this issue open seems to confuse some users regarding the support of the 3.0.x framework, I'm going to close it for now, as everything except the DS18B20 sensors has been tested and confirmed running.
If someone finds the time to test some DS18B20 sensors with BSB-LAN version 4.x, then I'd be happy to hear about it!

closed 2024-11-01 by fredlcore