fredlcore/BSB-LAN #655: [BUG] Sketch too big

stonie81, 2024-06-04:
**BSB-LAN Version**
#define MAJOR "4"
#define MINOR "0"
#define PATCH "11"
#define COMPILETIME "20240604202858"

**Architecture**
ESP32 Olimex EVB

**Bus system**
BSB

**Describe the bug**
compiles, but sketch is too big

**To Reproduce**
Steps to reproduce the behavior:
1. compile

**Log files - Bug reports without log files will be closed**
Der Sketch verwendet 1359989 Bytes (103%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 52316 Bytes (15%) des dynamischen Speichers, 275364 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
Textbereich überschreitet den verfügbaren Platz auf der Platine

Compilation error: Textbereich überschreitet den verfügbaren Platz auf der Platine

**Expected behavior**
sketch is not too big

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/user-attachments/files/15570751/BSB_LAN_config.txt)

fredlcore, 2024-06-05:
Read the manual and set the partition scheme to minimal SPIFFS.

closed 2024-06-05 by fredlcore

Svetko, 2024-10-05:
Updating the ESP32 framework within Arduino IDE seems to reset the partition scheme to “default”.

fredlcore, 2024-10-06:
Ok, thank you, will add this to the manual!