fredlcore/BSB-LAN #656: [BUG] Don't forget to adjust links to moved documentation ;)

DE-cr, 2024-06-09:
```
BSB_LAN.ino: *         German: https://1coderookie.github.io/BSB-LPB-LAN/
BSB_LAN.ino: *         English: https://1coderookie.github.io/BSB-LPB-LAN_EN/
json/lang_de.json:  "MENU_LINK_TOC": "https://1coderookie.github.io/BSB-LPB-LAN/inhaltsverzeichnis.html",
json/lang_de.json:  "MENU_LINK_FAQ": "https://1coderookie.github.io/BSB-LPB-LAN/kap15.html",
json/lang_de.json:  "MENU_LINK_URL": "https://1coderookie.github.io/BSB-LPB-LAN/kap08.html#81-auflistung-und-beschreibung-der-url-befehle",
json/lang_ru.json:  "MENU_LINK_TOC": "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html",
json/lang_ru.json:  "MENU_LINK_FAQ": "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html",
json/lang_ru.json:  "MENU_LINK_URL": "https://1coderookie.github.io/BSB-LPB-LAN_EN/appendix_b.html",
json/lang_en.json:  "MENU_LINK_TOC": "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html",
json/lang_en.json:  "MENU_LINK_FAQ": "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html",
json/lang_en.json:  "MENU_LINK_URL": "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#81-listing-and-description-of-the-url-commands",
localization/LANG_EN.h:#define MENU_LINK_TOC "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html"
localization/LANG_EN.h:#define MENU_LINK_FAQ "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html"
localization/LANG_EN.h:#define MENU_LINK_URL "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands"
localization/LANG_DE.h:#define MENU_LINK_TOC "https://1coderookie.github.io/BSB-LPB-LAN/inhaltsverzeichnis.html"
localization/LANG_DE.h:#define MENU_LINK_FAQ "https://1coderookie.github.io/BSB-LPB-LAN/kap15.html"
localization/LANG_DE.h:#define MENU_LINK_URL "https://1coderookie.github.io/BSB-LPB-LAN/kap05.html#51-url-befehle"
localization/LANG_RU.h:#define MENU_LINK_TOC "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html"
localization/LANG_RU.h:#define MENU_LINK_FAQ "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html"
localization/LANG_RU.h:#define MENU_LINK_URL "https://1coderookie.github.io/BSB-LPB-LAN_EN/appendix_b.html"
localization/LANG_FR.h:#define MENU_LINK_TOC "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html"
localization/LANG_FR.h:#define MENU_LINK_FAQ "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html"
localization/LANG_FR.h:#define MENU_LINK_URL "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands"
localization/www/lang_fr.js:UIStrings["MENU_LINK_TOC"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html";
localization/www/lang_fr.js:UIStrings["MENU_LINK_FAQ"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html";
localization/www/lang_fr.js:UIStrings["MENU_LINK_URL"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands";
localization/www/lang_en.js:UIStrings["MENU_LINK_TOC"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html";
localization/www/lang_en.js:UIStrings["MENU_LINK_FAQ"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html";
localization/www/lang_en.js:UIStrings["MENU_LINK_URL"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands";
localization/www/lang_de.js:UIStrings["MENU_LINK_TOC"] = "https://1coderookie.github.io/BSB-LPB-LAN/inhaltsverzeichnis.html";
localization/www/lang_de.js:UIStrings["MENU_LINK_FAQ"] = "https://1coderookie.github.io/BSB-LPB-LAN/kap15.html";
localization/www/lang_de.js:UIStrings["MENU_LINK_URL"] = "https://1coderookie.github.io/BSB-LPB-LAN/kap05.html#51-url-befehle";
localization/www/lang_ru.js:UIStrings["MENU_LINK_TOC"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html";
localization/www/lang_ru.js:UIStrings["MENU_LINK_FAQ"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html";
localization/www/lang_ru.js:UIStrings["MENU_LINK_URL"] = "https://1coderookie.github.io/BSB-LPB-LAN_EN/appendix_b.html";
```

DE-cr, 2024-06-09:
...if necessary?
Some minutes ago, my system led to a 404 message when clicking on the "Handbuch" link.
Seems like you can just close this issue here, though. Sorry for bothering you!

fredlcore, 2024-06-09:
No, thanks for the info, I haven't finally decided on the location of the new manual, but will finalize this sometime today.

closed 2024-06-09 by fredlcore

DE-cr, 2024-06-09:
Just as a reminder:
[image]

fredlcore, 2024-06-09:
Thanks, fixed.