fredlcore/BSB-LAN #659: [BUG] Zusätzliches Leerzeichen im MQTT-Topic

EB3R, 2024-06-14:
**Describe the bug**
Alle Parameter werden über MQTT nach Node-Red geleitet. Aus dem Topic "BSB-LAN/#" bekomme ich alle Nachrichten unterhalb übermittelt. Die Nachrichten bestehen aus topic:daten. Dabei bin ich auf folgende Besonderheit aufmerksam geworden. Einstellige Topics unterhalb von BSB-LAN erhalten nach dem "/" eine zusätzliche Leerstelle.
Beispiel:
Aufforderung: BSB-LAN/# erwartet u.a. BSB-LAN/0:<datum uhrzeit> es wird aber übermitelt: BSB-LAN/ 0:<datum uhrzeit>. Das ist bei allen einstelligen Parametern der Fall.

**To Reproduce**
1. MQTT Explorer öffnen
2. publish Message to "BSB-LAN/0"
3. Subscribe Topic "BSB-LAN/0"
4. Unter Topic wird angezeigt: "BSB-LAN/ 0"

Gruß Eberhard

fredlcore, 2024-06-14:
Bitte den Bug-Report vollständig ausfüllen und nicht eigenmächtig entscheiden, welche Informationen ich brauche und welche nicht. Nur vollständige Bug-Reports werden bearbeitet.

closed 2024-06-14 by fredlcore

fredlcore, 2024-06-15:
Da ich den Fehler bei mir glücklicherweise auch so direkt reproduzieren konnte, ging es diesmal auch ohne, beim nächsten Mal bitte an die Regeln halten. Fehler gefixt mit https://github.com/fredlcore/BSB-LAN/commit/0d5f1176e19f0cef2931faddb33924d89815ec57