fredlcore/BSB-LAN #661: [BUG] MAC address corruption(?)

DavidS, 2024-08-20:
**Version of the Arduino IDE: **

> Version: 2.3.2
> Date: 2024-02-20T09:54:08.669Z
> CLI Version: 0.35.3

**Version of the ESP32 or Arduino framework**  

> Arduino SAM Boards 1.6.12

**BSB-LAN Version**  

> #define MAJOR "4"
> #define MINOR "0"
> #define PATCH "57"
> #define COMPILETIME "20240721015110"

(current HEAD off github, default BSB_LAN_config.h contents)

```
david in debian in BSB-LAN-git on  master [?] 
❯ git describe --always --long
v0.41-4108-gfc8041e6

david in debian in BSB-LAN-git on  master [?] 
❯ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	BSB_LAN/BSB_LAN_config.h
	BSB_LAN/BSB_LAN_custom_defs.h

nothing added to commit but untracked files present (use "git add" to track)

david in debian in BSB-LAN-git on  master [?] 
❯ diff BSB_LAN/BSB_LAN_config.h*

david in debian in BSB-LAN-git on  master [?] 
❯ 
```

**Architecture**  

> Arduino Due with ethernet shield

**Bus system**  

>BSB

**Describe the bug**  

BSB-LAN uses a corrupted version of the configured MAC address to DHCP.

Configured:
```
/*
 *  Enter a MAC address, found either on the EthernetShield or use the one below.
 *  Change this if you have more than one BSB-LAN adapter in your LAN, so that there aren't any address conflicts.
 *  MAC address here only affects the LAN shield, it doesn't apply to the WiFi-ESP-solution.
*/
byte mac[6] = { 0x00, 0x80, 0x41, 0x19, 0x69, 0x90 };
```

used:

```
00:00:00:00:80:41
```

**To Reproduce**  
Steps to reproduce the behavior:
1. Switch on BSB-LAN device

**Log files - Bug reports without log files will be closed**  

```
READY
Reading EEPROM
0
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
43
44
45
46
47
48
49
50
51
52
53
54
55
56
Reading done.
EEPROM schema v.13 Program schema v.13
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 3
Address EEPROM option 3: 7
Address EEPROM option 4: 227
Address EEPROM option 5: 307
Address EEPROM option 6: 491
Address EEPROM option 7: 0
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 496
Address EEPROM option 13: 816
Address EEPROM option 14: 820
Address EEPROM option 15: 1140
Address EEPROM option 16: 1146
Address EEPROM option 17: 1147
Address EEPROM option 18: 1151
Address EEPROM option 19: 1155
Address EEPROM option 20: 1159
Address EEPROM option 21: 1163
Address EEPROM option 22: 1165
Address EEPROM option 23: 1169
Address EEPROM option 24: 1173
Address EEPROM option 25: 1237
Address EEPROM option 26: 1301
Address EEPROM option 27: 1302
Address EEPROM option 28: 1303
Address EEPROM option 29: 1313
Address EEPROM option 30: 1314
Address EEPROM option 31: 1634
Address EEPROM option 32: 1635
Address EEPROM option 33: 1639
Address EEPROM option 34: 1640
Address EEPROM option 35: 1641
Address EEPROM option 36: 1642
Address EEPROM option 37: 1675
Address EEPROM option 38: 1740
Address EEPROM option 39: 1805
Address EEPROM option 40: 1870
Address EEPROM option 41: 1902
Address EEPROM option 42: 1904
Address EEPROM option 43: 1905
Address EEPROM option 44: 1906
Address EEPROM option 45: 1907
Address EEPROM option 46: 1939
Address EEPROM option 47: 2003
Address EEPROM option 48: 0
Address EEPROM option 49: 2004
Address EEPROM option 50: 2036
Address EEPROM option 51: 2037
Address EEPROM option 52: 2038
Address EEPROM option 53: 2039
Address EEPROM option 54: 2041
Address EEPROM option 55: 2043
Address EEPROM option 56: 2044
Address EEPROM option 57: 0
Address EEPROM option 58: 0
Address EEPROM option 59: 2045
Address EEPROM option 60: 2046
Size of cmdtbl: 23520
free RAM: 81687
BSB-LAN version: 4.0.57-20240721015110
Microcontroller: Arduino Due
Using RX/TX pins 19, 18
Bus type defined in config: BSB
Init One Wire bus...
numSensors: 0
PPS settings:
Starting network connection via Ethernet/LAN...
IP: 10.0.0.18 netmask: 255.255.255.0 gateway: 10.0.0.138
Waiting 3 seconds to give Ethernet shield time to get ready...
```

**Expected behavior**  
Uses the configured MAC address without modification.

**Screenshots**  

Screenshots from my router, one is the original entry, the other is what is happening now.
[image]
[image]

**Desktop (if applicable, please complete the following information):**  
n/a

**Smartphone (if applicable, please complete the following information):**  
n/a

**Additional context**  
n/a

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**

default contents, proof see in the version section.

closed 2024-08-21 by fredlcore

DavidS, 2024-08-21:
This version: https://github.com/fredlcore/BSB-LAN/blob/fc8041e65436e399de7de1438675df222205293a/BSB_LAN/BSB_LAN_config.h.default , the version on current `master`.

fredlcore, 2024-08-21:
So you want my help and support and still refuse to follow the requirements stated above? 
I wlll not entertain such behaviour for the simple fact that it sets a precedent for others who then think they can also decide for themselves what kind of information I need and what not (as has already happened in the past).

DavidS, 2024-08-21:
straight from my arduino folder: [BSB_LAN_config.h.zip](https://github.com/user-attachments/files/16689431/BSB_LAN_config.h.zip)

fredlcore, 2024-08-21:
Well, it says "Do not ZIP or otherwise compress it" above, but never mind...

Please do the following and report back if there are any changes:
1. Run `/NE` in order to erase the flash memory of the board.
2. Set the webinterface settings to "advanced" and see if the MAC address is displayed correctly there or if the error occurs there as well
3. If everything shows up normal and the problem still persists, add some debug output before line 7092 (`if (Ethernet.begin(mac)) {  // DHCP`). `mac` is a byte array, so outputting `mac[0]` etc. should do it. As you are a developer, I assume I don't have to explain how to do that, otherwise let me know.