fredlcore/BSB-LAN #663: [BUG]Login to web failed

proasnet, 2024-08-24:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
1.8.19

**Version of the ESP32 or Arduino framework**  
1.6.12

**BSB-LAN Version**  
Latest (4.0)

**Architecture**  
Arduino Due

**Bus system**  
BSB

**Describe the bug**  
In Settings - filled a username and password for web interface. After store it, with this credentials is not possible login to web

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on username and 'URL password' and fill it
3. Click on Store button
4. Open web browser and try to login with previous filled credentials
5. Can not login into web interface
6. 401 Unauthorized.

fredlcore, 2024-08-24:
Hm, how do you access the web interface if password is active (and not working)?
Can you try and erase the EEPROM by calling `/NE` and see if that changes anything?
Also, version 4.0 is rather unspecific. Please make sure you try the very latest version from the master repository.

proasnet, 2024-08-24:
Until today I was using the older version 3.3. It worked fine in the old version. Today I downloaded the latest version, and after uploading I got to the web interface. I set everything up and it works. When I finally entered my credentials, and confirmed "Store parameter". And after refreshing the page, I couldn't get to the website anymore.
My next step was press reset button.
This reset the settings back to their original values ​​and the password disappeared. It was default reset.

proasnet, 2024-08-24:
I just did the eeprom cleanup but it behaves the same.

fredlcore, 2024-08-24:
The Changelog states that with any (major) version, the configuration has to be recreated (usually). 
But now I see you error: Passkey is the component of the URL (such as `http://bsb-lan.local/PASSKEY/`). And the HTTP authentication has to be entered in the format `username:password`. Both is clearly stated in the manual, by the way:
```
HTTP authentification 	USER_PASS 	If not empty, this setting contains the username and password for HTTP authentification, separated by a colon. 	Username:Password (Username:Password)
URL Passkey 	PASSKEY 	Passkey adds a user-defined sequence that needs to be added to the URL after the hostname.
For example, a passkey of 1234 requires every URL to be written as http://bsb-lan.local/1234/ instead of just http://bsb-lan.local/1234/. Leave empty for no passkey.
```

closed 2024-08-24 by fredlcore

proasnet, 2024-08-24:
I trying IPadress/username:password
But, browser still requied enter username and password.
And I don't know why the hardware reset button resets all the settings in the eeprom from the fixed definitions.

fredlcore, 2024-08-24:
> IPadress/username:password
This is not what I wrote above, and since you are not reading the manual, I can't help you.

proasnet, 2024-08-24:
I'm very sorry. You are right. I haven't accessed the BSB LAN for a long time and I forgot the username:password method and today I missed it and made a mistake. It already works. I realized that.
Thank you