fredlcore/BSB-LAN #672: License?

beppemarazzi, 2024-11-16:
Cannot find an explicit license file for the whole repo.. only the final line of readme.md (is it the license for that file or the whole project?)

Please consider to add a LICENSE,md file to the root of the repo (cfr. https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository)

Thanks

fredlcore, 2024-11-16:
It says "This work...", not "This page...", so with a little bit of common sense, the answer should be clear. Moving the same statement into a different file wouldn't make a difference, legally. Your link even states that the README is also used for declaring the license, and based on my experience with users who barely even read the README, I'll stick with what I know works best.

closed 2024-11-16 by fredlcore

beppemarazzi, 2024-11-16:
Not fully agree: "this work" while reading a "README.md" IMHO is not fully clear... also because it is citing a "Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License" and AFAIK it's a kind of license mainly used for "documentation" not for source code...

Another poin t is the visibility: following the github guideline (i.e. the ilnk i posted states "As a best practice, we encourage you to include the license file with your project.") the license is reported by github in the project headers and sidebars ....