fredlcore/BSB-LAN #681: [FEATURE REQUEST] Add device ID to MQTT poll endpoint

uschindler, 2024-12-05:
While implementing the cool new MQTT `/poll` endpoint, I had that idea which IMHO would better fit the new topic structure:

Currently the `/poll` MQTT endpoint is on top level. You can send parameters to update, but if you don't want the default device, you have to use some crazy syntax: 
[image]

I'd propose to use (conforming to the new topic structure) `/<devid>/poll` instead and simplyfiy the payload to only accept parameter numbers or category/parameter numbers pairs.

This is just a suggestion for further improving the topic structure and get rid of crazy payload syntaxes. No need to implement that ASAP or at all. See it as discussion point.

The only downside of that syntax is that you need "one poll per device" which is IMHO acceptable.

fredlcore, 2024-12-05:
I'm not sure what the "crazy syntax" is supposed to be. It's either a list of simple numbers in the same way you'd use them when using the web-interface or in the web-configuration, or a list of parameters in topic-notation. For most users (i.e. all users who use the BSB bus), they won't have to worry about the device ID anyway because in BSB, you will always use device ID 0. So in that case, those users would use exactly the same payload, just not send it to the topic `BSB-LAN/poll` but `BSB-LAN/0/poll`. I don't see the improvement here...

uschindler, 2024-12-05:
The improvement is only to make the topic structure more clear. For all other get/set it requires device ID, but for polling you need to send device id inside the payload. This is inconsistent.

If you want to poll a single parameter, you also need dev id in topic.

So basically this is a suggestion to make it more clear. It's a syntactic improvement because you are already improvig the topic structure, which is very nice now, although I would have used "state" or "get" instead of "status" for the getter. "status" is commonly used in MQTT only for online/offline LASTWILL messages.

fredlcore, 2024-12-05:
`BSB-LAN/poll` also makes migrating from the BSB-LAN-internal logging to MQTT logging easier, because you just need to copy whatever you have in the webinterface. For some people who poll 20 parameters, that's easier than splitting that up into polls for two or more devices etc.

fredlcore, 2024-12-05:
Bottom line, if I were to implement `/dev_id/poll`, then `BSB-LAN/poll` would have to go in order to not introduce any confusion due to the different syntax. And the consistency between the web-interface and the notation for the `BSB-LAN/poll` syntax in my opinion trumps any potentially more logical syntax.

uschindler, 2024-12-05:
Let's keep this issue open for a while, this is mostly cosmetic, but let's others comment on that.

> The most logical syntax would be to use `BSB-LAN/0/16/700/poll`, exactly mirroring `/status`, `/set`etc.

That's already implemented, but has the problem of not being bulky. Of course this perfectly fits the topic structure.

> `BSB-LAN/poll` also makes migrating from the BSB-LAN-internal logging to MQTT logging easier, because you just need to copy whatever you have in the webinterface. For some people who poll 20 parameters, that's easier than splitting that up into polls for two or more devices etc.

Of course. This is what I did before splitting the lists into multiple update frequencies. But for those who copypaste from web interface, it should poll on all devices, if no dev id is given in payload.

Anyways, I am fine. It was just a discussion point. No need to implement anything. It keeps the discussion of your project alive on Github.

uschindler, 2024-12-05:
OK, I thought the log settings poll on all devices. Sorry. I don't have multiple devices.

uschindler, 2024-12-05:
If I enable the option to show unknown parameters, it shows them. But without that option it takes very long time and sometimes times out. Now it worked, it just showed nothing. The device only crashes when it times out and you press reload in browser.

The issue was obviously caused by the setting "don't show unsupported parameters" when deploying the new config (as EPROM was erased).

closed 2025-11-29 by fredlcore