fredlcore/BSB-LAN #686: [FEATURE REQUEST] Add MAC address in properties of the zeroconf discovery protocol

liudger, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
When setting up an integration with zeroconf in Home-Assistant we need to check if it's already setup.
For this we use the MAC address as ip address can change. The problem is that this information is not send in the properties of the zeroconf message.
So we can't check if BSBLan integration is already setup for the device.

**Describe the solution you'd like**
add MAC address to service text (txtContent?) as a property

Hi Fred,
I don't have the latest firmware installed yet. Will look into this shortly. I suspect this MAC address is not in there? If so discard this request.

liudger, 2024-12-11:
https://github.com/fredlcore/BSB-LAN/blob/71805809edc40267a0ab62ec1fcaac2e714ea7fa/BSB_LAN/BSB_LAN.ino#L7860
We should add here something like:
`MDNS.addServiceTxt("bsblan", "tcp", "mac", escapedMac.c_str());`

liudger, 2024-12-11:
Something like this?

```
#if defined(ESP32)
    MDNS.begin(mDNS_hostname);
    MDNS.addService("http", "tcp", 80);
    MDNS.addService("BSB-LAN web service._http", "tcp", 80);

    // Add MAC address for ESP32 (WiFi)
    String macAddress = WiFi.macAddress();
    MDNS.addServiceTxt("BSB-LAN web service._http", "tcp", "mac", macAddress);
#else
    if (network_type == WLAN) {
#if defined(WIFISPI)
        mdns.begin(WiFi.localIP(), mDNS_hostname);
#endif
        // Add MAC address for WiFi
        String macAddress = WiFi.macAddress();
        mdns.addServiceTxt("BSB-LAN web service._http", "tcp", "mac", macAddress);
    } else {
        mdns.begin(Ethernet.localIP(), mDNS_hostname);

        // Add MAC address for Ethernet
        byte mac[6];
        Ethernet.MACAddress(mac);
        char macStr[18];
        snprintf(macStr, sizeof(macStr), "%02X:%02X:%02X:%02X:%02X:%02X",
                 mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
        mdns.addServiceTxt("BSB-LAN web service._http", "tcp", "mac", macStr);
    }
    mdns.addServiceRecord("BSB-LAN web service._http", HTTPPort, MDNSServiceTCP);
#endif
    printFmtToDebug("Starting MDNS service with hostname %s\r\n", mDNS_hostname);
}
```

fredlcore, 2024-12-11:
Sure, can you come up with a (tested) pull request? Happy to add that, but don't have the means to test it at the moment...

liudger, 2024-12-13:
I will try.

fredlcore, 2025-01-07:
The above wouldn't work on the ESP32 because after the first `#else` come the command for the Arduino Due. Maybe that's the one you have, but it should work at least also for the ESP32.

fredlcore, 2025-02-07:
I found this one here which you probably also found before because it sounds quite similar ;)...
https://github.com/Aircoookie/WLED/issues/305
I have added the MAC address now for the ESP32, but wasn't really sure about the proper syntax for the Arduino Due, so maybe someone who is still running BSB-LAN on the Due can help here?

fredlcore, 2025-02-08:
The most recent update now also contains the MAC address for Arduino Due. Could someone check that the MDNS entries now look fine on the Due as well? Thanks!

fredlcore, 2025-02-08:
@tiger42 and @proasnet, would you be able to test this?

tiger42, 2025-02-08:
I've switched back and forth between BSB-LAN 4.2.74 and 4.2.75 multiple times (to be sure) and found out that the changes seem to break the mdns discovery.
On Linux the `avahi-browse` command does not find the BSB-LAN service any more.
On Windows, the ["Bonjour Browser"](https://hobbyistsoftware.com/bonjourbrowser) (the only tool I could find that can print out TXT records), still finds the service (although with a different name without "web service") but cannot display the additional service information (see screen shots).

The output of `avahi-browser -art` for BSB-LAN 4.2.74, where it still works, looks like this:
```
 = enp2s0 IPv4 BSB-LAN web service                           Web Site             local
   hostname = [BSB-LAN.local]
   address = [192.168.0.95]
   port = [80]
   txt =
```

**BSB-LAN 4.2.74**

[image]

**BSB-LAN 4.2.75**

[image]

fredlcore, 2025-02-08:
Great, thanks a lot for such fast and thorough checking!
The problem is that I myself don't really understand the differences between the two MDNS implementations for ESP32 and Arduino respectively:
The ESP32 MDNS implementation provides two commands, `addService` and `addServiceTxt`.
`addService` expects a service, a protocol and a port; `addServiceTxt` expects a name, a protocol, as well as a key and a value.
The commands 
```
    MDNS.addService("http", "tcp", HTTPPort);
    MDNS.addServiceTxt("http", "tcp", "description", "BSB-LAN web service");
    MDNS.addServiceTxt("http", "tcp", "mac", (const char*)macStr);
```

result in the following output of `avahi-browser -a -r`:

```
= ovs_eth0 IPv4 BSB-LAN                                       _http._tcp           local
   hostname = [BSB-LAN.local]
   address = [192.168.1.50]
   port = [80]
   txt = ["description=BSB-LAN web service" "mac=ac:67:b2:46:57:f4"]
```

This looks fine for me.

The MDNS implementation for the Arduino only provides the command `addServiceRecord` which expects a name, a port, a protocol and optional text content.
Previously, the command was like this:
```
mdns.addServiceRecord("BSB-LAN web service._http", HTTPPort, MDNSServiceTCP);
```
However, since the `BSB-LAN web service._http` didn't work with the ESP32 (where I also used this until recently), I stumbled about some examples which said that (for the ESP32 implementation), any description should be put in the `txt` field.
Therefore, I tried to do the same for the Arduino - just leaving the device name and put the description in a description field.

Going by the ArduinoMDNS source code, a repeated call of `addServiceRecord` should add one more entry with each call, so that I thought that this is the way to add multiple txt records as can be seen above in the output for the ESP32.
What baffles me is that the entry shows up as `BSB-LAN`, but that the added text doesn't. On the other hand, in 4.1.74, your screenshot shows a table for TXT records, but these are empty (which makes sense because no txt record information was added back then). It's a bit difficult for me to do further debugging without really understanding MDNS enough and not having a Due to actually do some tests with it. Especially since my code doesn't look significantly different than the MDNS TXT examples here:
https://github.com/arduino-libraries/ArduinoMDNS/tree/master/examples/Ethernet/RegisteringServicesWithTxtRecord

As for `avahi-browse` not finding the (new) entry, I sometimes had to restart the ESP32 before any changes showed up in `avahi-browse`, maybe that helps on your side, too? A direct comparison between the entries between 4.1.74 and 4.1.75 might point us to the right direction...

Again, thanks for helping out and I hope we can somehow solve this without much more work...

fredlcore, 2025-02-08:
If you could try it again, could you comment out the second line, i.e. like this:
```
mdns.addServiceRecord("BSB-LAN._http", HTTPPort, MDNSServiceTCP, "description=BSB-LAN web service");
// mdns.addServiceRecord("BSB-LAN._http", HTTPPort, MDNSServiceTCP, service_txt);
```
and see if that brings up the entry again and what `avahi-browse` outputs then? Maybe the double call is not feasible...

tiger42, 2025-02-08:
I've found a working solution after hours of trying different things...

mdns.addServiceRecord may in fact be called multiple times, but only to add new services (like "BSB-LAN._info" or whatever you can imagine). That does not help with multiple TXT records.

All TXT records must go into one single string, which must have a very specific format. The format is documented here: http://www.zeroconf.org/Rendezvous/txtrecords.html. The format is: length of single record in bytes + record, where record is defined as key + "=" + value. Repeat this pattern for further TXT records. There must be no delimiters between the records!

I've tried lots of ways of building the TXT record string, using `snprintf`, `strncpy`, `strncat`, converting from char* to char[] and vice versa and so on. Nothing generated a string that would work with `mdns.addServiceRecord`. For example, a simple solution like this does **not** work:
```
snprintf(txt_records_buffer, sizeof(txt_records_buffer), "%c%s%c%s%s", 0x1F, "description=BSB-LAN web service", 0x16, "mac=", macStr)
```
So the solution I've come up with in the end admittedly does not look elegant at all, but it works. Change it and it will break, I promise. 😉 Maybe it would make sense to move the code to a separate function:
```C++
    mdns.begin(Ethernet.localIP(), mDNS_hostname);

    // temporary buffer for concatenated string
    char txt_records[50];

    // we need to manually build the string, as it must have a very specific format
    int pos = 0;

    // first segment: length + "description=BSB-LAN web service"
    txt_records[pos++] = (char)31;  // length of first string
    memcpy(&txt_records[pos], "description=BSB-LAN web service", 31);
    pos += 31; 

    // second segment: length + "mac=" + MAC address
    txt_records[pos++] = (char)(4 + strlen(macStr));  // length of second string
    memcpy(&txt_records[pos], "mac=", 4);
    pos += 4;
    memcpy(&txt_records[pos], macStr, strlen(macStr));
    pos += strlen(macStr);

    mdns.addServiceRecord("BSB-LAN._http", HTTPPort, MDNSServiceTCP, (const char*)txt_records);
```
Here's the output of `avahi-browse -art`:
```
= enp2s0 IPv4 BSB-LAN                                       Web Site             local
   hostname = [BSB-LAN.local]
   address = [192.168.0.95]
   port = [80]
   txt = ["mac=00:80:41:19:69:90" "description=BSB-LAN web service"]
```

And this is what "Bonjour Browser" displays:

[image]

fredlcore, 2025-02-08:
Awesome, thanks a lot! Now some other reference I found makes total sense, as I was wondering what all the hex values do in that string. I'll post the link once I find it.
But rather than adding a complex function, can you post the output of that string here? Because its length and therefore those bytes won't change, as the length of the c description and the length of the MAC address always stay the same. I would then rather hardcode that string and reference this issue in case there are any changes necessary in the future.

Again, thanks a lot!

fredlcore, 2025-02-09:
Here's the link I mentioned:
https://cdck-file-uploads-europe1.s3.dualstack.eu-west-1.amazonaws.com/arduino/original/3X/8/4/84cff7288b0885549f0c9cd51cea426f7a144075.txt
```
mdns.begin(WiFi.localIP(), "microwave");
String txtString("\x05" "c#=12" "\x04" "ff=1" "\x14" "id=" + id + "\x08" "md=Ver 1" "\x04" "s#=1" "\x04" "sf=1" "\x04" "ci=1");
mdns.addServiceRecord("Microwave Starter._hap", port, MDNSServiceTCP, txtString.c_str());
```

fredlcore, 2025-02-09:
I think I found some small issues in your initial approach and maybe that's why it failed:
Your suggested code
```
snprintf(txt_records_buffer, sizeof(txt_records_buffer), "%c%s%c%s%s", 0x1F, "description=BSB-LAN web service", 0x16, "mac=", macStr)
```
has a discrepancy because, you set the length of the MAC string to 0x16, but your code snippet calculates the correct length at 0x15.
Also, depending on how large you set txt_records_buffer, it might lead in a cut-off string. I tried the snprintf command and set txt_records_buffer to the same size as txt_records (i.e. 50), and it led to this unexpected result that may be the reason why it is not accepted.
Here's the test code:
```
#include <iostream>
#include <cstring>
#include <cstdint>

using namespace std;

int main() {
    char txt_records[50];
    char txt_records_buffer[50];
    char macStr[18];
    uint8_t mac[] = { 0x00, 0x80, 0x41, 0x19, 0x69, 0x90 };

    snprintf(macStr, sizeof(macStr), "%02x:%02x:%02x:%02x:%02x:%02x", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    snprintf(txt_records_buffer, sizeof(txt_records_buffer), "%c%s%c%s%s", 0x1F, "description=BSB-LAN web service", 0x16, "mac=", macStr);

    // we need to manually build the string, as it must have a very specific format
    int pos = 0;

    // first segment: length + "description=BSB-LAN web service"
    txt_records[pos++] = (char)31;  // length of first string
    memcpy(&txt_records[pos], "description=BSB-LAN web service", 31);
    pos += 31; 

    // second segment: length + "mac=" + MAC address
    txt_records[pos++] = (char)(4 + strlen(macStr));  // length of second string
    memcpy(&txt_records[pos], "mac=", 4);
    pos += 4;
    memcpy(&txt_records[pos], macStr, strlen(macStr));
    pos += strlen(macStr);
    for (int i = 0; i < pos; i++) {
        printf("char: %c, value: %02X, char: %c, value %02X\n", txt_records[i], txt_records[i], txt_records_buffer[i], txt_records_buffer[i]);
    }
}
```
And this is the result, notice both the difference in the length byte for the mac address information as well as the cut-off MAC address entry:
```
char: , value: 1F, char: , value 1F
char: d, value: 64, char: d, value 64
char: e, value: 65, char: e, value 65
char: s, value: 73, char: s, value 73
char: c, value: 63, char: c, value 63
char: r, value: 72, char: r, value 72
char: i, value: 69, char: i, value 69
char: p, value: 70, char: p, value 70
char: t, value: 74, char: t, value 74
char: i, value: 69, char: i, value 69
char: o, value: 6F, char: o, value 6F
char: n, value: 6E, char: n, value 6E
char: =, value: 3D, char: =, value 3D
char: B, value: 42, char: B, value 42
char: S, value: 53, char: S, value 53
char: B, value: 42, char: B, value 42
char: -, value: 2D, char: -, value 2D
char: L, value: 4C, char: L, value 4C
char: A, value: 41, char: A, value 41
char: N, value: 4E, char: N, value 4E
char:  , value: 20, char:  , value 20
char: w, value: 77, char: w, value 77
char: e, value: 65, char: e, value 65
char: b, value: 62, char: b, value 62
char:  , value: 20, char:  , value 20
char: s, value: 73, char: s, value 73
char: e, value: 65, char: e, value 65
char: r, value: 72, char: r, value 72
char: v, value: 76, char: v, value 76
char: i, value: 69, char: i, value 69
char: c, value: 63, char: c, value 63
char: e, value: 65, char: e, value 65
char: , value: 15, char: , value 16
char: m, value: 6D, char: m, value 6D
char: a, value: 61, char: a, value 61
char: c, value: 63, char: c, value 63
char: =, value: 3D, char: =, value 3D
char: 0, value: 30, char: 0, value 30
char: 0, value: 30, char: 0, value 30
char: :, value: 3A, char: :, value 3A
char: 8, value: 38, char: 8, value 38
char: 0, value: 30, char: 0, value 30
char: :, value: 3A, char: :, value 3A
char: 4, value: 34, char: 4, value 34
char: 1, value: 31, char: 1, value 31
char: :, value: 3A, char: :, value 3A
char: 1, value: 31, char: 1, value 31
char: 9, value: 39, char: 9, value 39
char: :, value: 3A, char: :, value 3A
char: 6, value: 36, char: 6, value 36
char: 9, value: 39, char: 9, value 39
char: :, value: 3A, char: :, value 3A
char: 9, value: 39, char: 9, value 39
char: 0, value: 30, char: 0, value 30
```

fredlcore, 2025-02-09:
I've modified the relevant code in BSB-LAN now, so could you (as well as anyone else with a Due) test if this works now?

uschindler, 2025-02-09:
As an alternative way: Why not add the Mac address to the service name? See the examples above of Shellys and the Denon audio device. They have serial numbers in the description. If the Mac address is part of description, then home Assistant Integration can still differentiate between devices.

fredlcore, 2025-02-09:
That's another possibility, but I guess to do it via the TXT records (which is made for such kind of information, see the "version" information in the linked documentation above) would be the cleaner solution. In terms of differentiation, it might be useful to add the serial number of the device and/or the device model, since there are a few users who have two BSB-LAN running and would need to differentiate between the two.

tiger42, 2025-02-10:
I was really too stupid to count bytes 😉 All my initial attempts failed just because of this...

The changes from your latest version _nearly_ work now.
You forgot an underscore though. Without it, the service can not be found.
```diff
diff --git a/BSB_LAN/BSB_LAN.ino b/BSB_LAN/BSB_LAN.ino
index 4e47b28b..af44d5a5 100644
--- a/BSB_LAN/BSB_LAN.ino
+++ b/BSB_LAN/BSB_LAN.ino
@@ -8032,7 +8032,7 @@ active_cmdtbl_size = sizeof(cmdtbl)/sizeof(cmdtbl[0]);
     MDNS.addServiceTxt("http", "tcp", "mac", (const char*)macStr);
 #else
     char instance_name[35];
-    snprintf(instance_name, sizeof(instance_name), "BSB-LAN %s/%lu.http", dev_lookup[0].name, dev_lookup[0].dev_serial);
+    snprintf(instance_name, sizeof(instance_name), "BSB-LAN %s/%lu._http", dev_lookup[0].name, dev_lookup[0].dev_serial);
     mdns.begin(Ethernet.localIP(), mDNS_hostname);
     char service_txt[60];
     snprintf(service_txt, sizeof(service_txt), "%c%s%c%s%s", 0x1F, "description=BSB-LAN web service", 0x15, "mac=", macStr)
```
After this litte change, the service can be discovered and looks correct:
```
= enp2s0 IPv4 BSB-LAN LMU7/807020064                        Web Site             local
   hostname = [BSB-LAN.local]
   address = [192.168.0.95]
   port = [80]
   txt = ["mac=00:80:41:19:69:90" "description=BSB-LAN web service"]
```

fredlcore, 2025-02-10:
Great you spotted that! Fixed the underscore in the most recent version, if you could try that one one last time, then we can finally close this :).

tiger42, 2025-02-10:
Device discovery tested under Windows and Linux, both look fine now 👍🏻

fredlcore, 2025-02-10:
Great, thanks a lot again, case close :)...

closed 2025-02-10 by fredlcore