fredlcore/BSB-LAN #687: [BUG] The fallback partitions.csv table's SPIFFS partition is not big enough for BSB-LAN

iUltimateLP, 2024-12-12:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
No Arduino IDE, I'm using PlatformIO

**Version of the ESP32 or Arduino framework**  
`espressif/arduino-esp32#3.0.1`

**BSB-LAN Version**  
4.2.41

**Architecture**  
ESP32 Olimex EVB

**Bus system**  
BSB

**Describe the bug**  
I'm using PlatformIO to compile the project. It's confirmed to be working with my own partition table, and I'm successfully running BSB-LAN on an ESP32-EVB board with the firmware compiled by PlatformIO. 

Since the "Minimal SPIFFS partition" table was for some reason smaller than the one used in the Arduino IDE, I was using `huge_app.csv` for previous builds, which worked fine. Now, I've seen that you're supplying your own `partitions.csv` table as a fallback. However, the defined SPIFFS partition size of `0x20000` (131072 bytes) is too small to fit the SPIFFS data of BSB-LAN. As a result, I get:

```
Building FS image from 'data' directory to .pio\build\esp32-evb\spiffs.bin
/c3.js.gz
/d3.js.gz
SPIFFS_write error(-10001): File system is full.
```

**To Reproduce**  
Steps to reproduce the behavior:
1. Clone [my fork](https://github.com/iUltimateLP/BSB-LAN) of this repo, which contains a valid platformio configuration
2. Open the project in PIO (or use CLI)
3. Try to build the SPIFFS image 

**Log files - Bug reports without log files will be closed**  
```
Processing esp32-evb (platform: platformio/espressif32; board: esp32-evb; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-evb.html
PLATFORM: Espressif 32 (6.9.0) > OLIMEX ESP32-EVB
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.1+sha.cc50d90 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.442a798083 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Converting BSB_LAN.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- SPI @ 2.0.0
|-- EEPROM @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Ethernet @ 2.0.0
|-- LittleFS @ 2.0.0
|-- Update @ 2.0.0
|-- WebServer @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
[... 290 lines cut ...]
// Contra: Someone who has access to your network and finds out about BSB-LAN's username and password can install their own software on the ESP32.
boolean enable_ota_update = false;

// Reduce clock speed of ESP32 from 240 MHz to 80MHz, saving ca. 20% energy.
// Works well when connecting via LAN, but since it reduces WiFi range and log file display times when using WiFi, it is disabled by default.
boolean esp32_save_energy = false;

// "External" web server. Read files from SD-card / flash memory. Only static content: html, js, css, jpg, etc.
boolean webserver = false;

// Debug options
byte debug_mode = 1;  // Debug mode: 0 - disabled, 1 - send debug messages to serial interface, 2 - send debug messages to telnet client
byte verbose = 1;     // If set to 1, all messages on the bus are printed to debug interface. Set to 2 only if you are asked by the developers as the extensive output will significantly slow down the microcontroller.
bool show_unknown = true; // true - show all parameters, false - hide unknown parameters from web display (parameters will still be queried and therefore take time!)

#define CONFIG_VERSION 39

/************************************************************************************/
/************************************************************************************/
/* Settings -   END                                                                 */
/************************************************************************************/
/************************************************************************************/
```

Cheers!

iUltimateLP, 2024-12-12:
Addendum: I believe you took the partition configuration from [here](https://github.com/espressif/arduino-esp32/commits/master/tools/partitions/min_spiffs.csv), in which for some reason the SPIFFS partition size got changed in commit https://github.com/espressif/arduino-esp32/commit/161b1673d9ad2804ff5a60a8db89464ad9bf69e7

fredlcore, 2024-12-13:
Yes, that's where I took it from. Since this is the official partition definition, I wonder why I should change it - and to what. The `huge_apps` variant that you are suggesting only has one app partition, meaning you can't do an OTA update, so that's not an option. Furthermore, the system compiles fine with the standard settings. If you want to add additional stuff to the SPIFFS partition, you have to use your own partition definition.

closed 2024-12-13 by fredlcore

iUltimateLP, 2024-12-13:
Okay, weird. Just looked up the SPIFFS partition that's used by the Arduino IDE, and that's correct. Something with block sizes or metadata sizes in PlatformIO's mkspiffs configuration seems to be different. Anyways, I've adjusted the partition table to save some space on the app partitions and increased the SPIFFS partition. That will probably be fine for now, thanks!