fredlcore/BSB-LAN #688: ESP32-H2 compatibility

rroblik, 2024-12-14:
Hi, 

I has a lot of freeze issue with my arduino board. I'm looking to upgrade it to a more home automation "compliant" protocol from WiFi to zigbee. I found [esp32-h2](https://www.espressif.com/en/products/socs/esp32-h2) board, so I ask to myself if this can work out of the box with BSB-LAN ? Or what changes will that require? Found nothing about it...

Thanks !

fredlcore, 2024-12-14:
I don't know - could work, but no guarantee, and no support from my side. A normal ESP32 is cheap, and the boards for right in, so I don't see the need for other models.

rroblik, 2024-12-14:
Because heating system are commonly far away from the living where the main of the home automation live. Also because I want to stop using WiFi for automation and because I have a lot of zigbee routers around my heating system. What is the main thing to check for compatibility ?

thanks

janstuyck, 2024-12-14:
Hi rroblik,

According to my experience there is no problem compiling (I found it very easy and little work) for other ESP32 board as long as they have >=4MB flash. At this time I am using a XIAO ESP32-C6 with 4MB flash (I would have preferred 8MB regarding the partition scheme).

The only things I had to do was:

- Change the TX/RX pins in de code to the ones I chose to use on my C6 board (Next to GND and 3.3V these are the four connections that need to be wired to the BSB-LAN adapter you have. My adapter is currently still just connected by four wires...

- I also had to make a custom partitions.csv because for the XIAO_ESP32C6 board theer is no option for a "Minimal SPIFFS" partition scheme as instructed in step 9 of the Quickstart instruction in the manual (https://docs.bsb-lan.de/quickstart.html).

Is it correct that the H2 doesn't come with WiFi? I would suggest you to go for a C6 version because then you can easily start without having to get all the changes for merging from WiFi to ZigBee right. You could leave all the WiFi related code where it is and use it as a reference for your new ZigBee code?

I'm sharing a link to my XIAO ESP32 C6 project. I didn't make myself familiar with working in a Github fork so it's in another cloud:
https://e.pcloud.link/publink/show?code=kZUMQwZXesjqnOqUNJlfRoQmH6rIBlFSxpy

I am also thinking about using its ZigBee / Thread capabilities in some future. Since this not incorporated (yet, will it ever be?) in ESPHome this could be a nice project to familiarize myself with it in the Arduino IDE. So I'm very interested in your work concerning implementation of ZigBee!!

Have fun!

Kind regards,

Jan

fredlcore, 2024-12-14:
@rroblik: So you mean you want to replace WiFi/Ethernet with ZigBee? That would probably include major changes in the way BSB-LAN works, including the web interface, MQTT and telnet access. I haven't really worked with ZigBee yet, and if the added footprint is tolerable, I would consider adding it sometime in the future, but not as a replacement for networking functionality. 

I'm converting this issue into a discussion, so you can continue discussing the matter without it being on my "to do list".