fredlcore/BSB-LAN #694: Complitation error

Danielnope, 2024-12-17:
#define MAJOR "4"
#define MINOR "2"
#define PATCH "44"
#define COMPILETIME "20241215161246"

OLIMEX ESP32-EVB

Beim Versuch eine Kompilierte Binärdatei zu erstellen wird mit folgendem Fehler abgebrochen.

In file included from D:\Daten\BSB-Lan\BSB-LAN-master\BSB_LAN\BSB_LAN_defs.h:16,
                 from D:\Daten\BSB-Lan\BSB-LAN-master\BSB_LAN\BSB_LAN.ino:160:
D:\Daten\BSB-Lan\BSB-LAN-master\BSB_LAN\localization\LANG_DE.h:382:10: fatal error: localization/UNDEF_LANG_DE_LEGACY.h: No such file or directory
 #include "localization/UNDEF_LANG_DE_LEGACY.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: localization/UNDEF_LANG_DE_LEGACY.h: No such file or directory

Die Dateien UNDEF_LANG_DE_LEGACY.h und LANG_DE_LEGACY.h befinden sich im Ordner localization.

Wenn ich eine Ältere Version versuche zu Kompilieren, zum Beispiel 
#define MAJOR "3"
#define MINOR "3"
#define PATCH "2"
#define COMPILETIME "20231115201023"

funktioniert dies ohne Probleme.

fredlcore, 2024-12-17:
Sorry, I cannot reproduce this. Just downloaded the same version from Master:
```
#define MAJOR "4"
#define MINOR "2"
#define PATCH "44"
#define COMPILETIME "20241215161246"
```
Just renamed BSB_LAN:config.h.default and BSB_LAN_custom_defs.h.default according to the manual and compiled the file without any problems.

fredlcore, 2025-01-06:
Ok, this seems to be related to Arduino Due and is fixed now.

closed 2025-01-06 by fredlcore