fredlcore/BSB-LAN #696: [BUG]

clausprivat, 2024-12-21:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
Please note: Versions 1.8.19 and below are not supported by this project anymore.

Version: 2.3.4
Date: 2024-12-03T10:03:03.490Z
CLI Version: 1.1.1
Copyright © 2024 Arduino SA

**Version of the ESP32 or Arduino framework**  
See ***Tools/Board/Board Manager*** for exact version you are using

esp32 by Espressif Systems 3.0.7 installed

**BSB-LAN Version**  
As displayed in the web-configuration or copy from `bsb-version.h` file. Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly. We don't fix bugs for discontinued platforms (such as Arduino Mega).

#define MAJOR "4"
#define MINOR "2"
#define PATCH "46"
#define COMPILETIME "20241220005136"

**Architecture**  
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.)

ESP32 NodeMCU

**Bus system**  
Which bus BSB-LAN is running on (BSB, LPB, PPS)?

LPB
 

**Describe the bug**  
A clear and concise description of what the bug is.

Frederik Holst <bsb@code-it.de>
 An: Claus Reitmeier <claus.reitmeier@gmail.com>
 Hallo Claus,
 15. Dezember 2024 um 02:56
 ja, Deine Heizung ist kompatibel (siehe die markierten Anschlüsse beim RVS21 Regler in unserem Handbuch), und
...
Daher Anschluss siehe Screenshot. , die rote Led des Adapters leuchtet, Anzeige Reglerspezifische-Parameterliste :

BSB-LAN kann keine Verbindung zum Heizungs-Regler aufbauen. Bitte sicherstellen, dass der Adapter mit dem Regler verbunden ist und die Heizung angeschaltet ist. Dann BSB-LAN neu starten, um die Verbindung herzustellen.

[2024.12.21_Reglerspezifische-Parameterliste-Anzeige.txt](https://github.com/user-attachments/files/18216840/2024.12.21_Reglerspezifische-Parameterliste-Anzeige.txt)

Die Heizung ist eingeschaltet, der Versuch wurde mehrmals gestartet.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Reglerspezifische-Parameterliste'
2. Click on '....'
3. Scroll down to '....'
4. See error
BSB-LAN kann keine Verbindung zum Heizungs-Regler aufbauen. Bitte sicherstellen, dass der Adapter mit dem Regler verbunden ist und die Heizung angeschaltet ist. Dann BSB-LAN neu starten, um die Verbindung herzustellen.

in Serial-Monitor-Log.txt
query failed
 query failed
 query failed
8700
 query failed
 query failed
 query failed
8700
78 0E 00 42 C0 02 00 14 06 3D 05 05 21 F5 18
...

**Log files - Bug reports without log files will be closed**  
If you cannot provide a log file, explain here why in order to prevent closing this bug report. Attach or copy/paste the log of the Arduino IDE's serial monitor when performing the actions that lead to the error. If possible start the log right before BSB-LAN boots (pressing the reset button on the board).
[2024.12.21_Serial-Monitor-Log.txt](https://github.com/user-attachments/files/18216841/2024.12.21_Serial-Monitor-Log.txt)

**Expected behavior**  
A clear and concise description of what you expected to happen.
Reglerspezifische-Parameterliste-Anzeige sollte etwas anzeigen...

**Screenshots**  
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (if applicable, please complete the following information):**  
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**  
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. Safari, FireFox]
 - Browser version [e.g. 22]

**Additional context**  
Add any other context about the problem here.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**

das .h File kann nicht hochgeladen werden, daher hab ich die Extension von .h auf .txt geändert.

[BSB_LAN_config.txt](https://github.com/user-attachments/files/18216872/BSB_LAN_config.txt)

Vielen Dank für den Support!

fredlcore, 2024-12-21:
Why did you select LPB if you have connected BSB-LAN to the BSB connector of your heating system? Change that and it'll work.

closed 2024-12-21 by fredlcore

clausprivat, 2024-12-21:
thank you, it works with BSB....