fredlcore/BSB-LAN #699: [FEATURE REQUEST] Add mqtt commands

apocaliss92, 2024-12-26:
**Is your feature request related to a problem? Please describe.**
Not really an issue, just a way to ease the mqtt communication

**Describe the solution you'd like**
With mqtt autodiscovery is possible to also provide selectors/inputs to interact with the bsb controller

**Describe alternatives you've considered**
Instead of discovery sensors, more structured data can be used instead, such as selector and input

**Describe your own contribution**
I'm not very familiar with this language; but I can help providing the mqtt configurations to be sent during the discovery functionality

**Additional context**
These are 2 examples of what I'm manually doing on my ha instance

  - state_topic: "BSB-LAN/0/7/700/status"
    name: BSB-LAN Modalità riscaldamento
    command_topic: "BSB-LAN/0/7/700/set"
    command_template: "{{ value.split(' -')[0]}}"
    options:
      - "0 - Protezione"
      - "1 - Automatico"
      - "2 - Ridotto"
      - "3 - Comfort"
    optimistic: false
    qos: 0
    retain: true
    
  - state_topic: "BSB-LAN/0/7/710/status"
    name: BSB-LAN Setpoint riscaldamento confort
    command_topic: "BSB-LAN/0/7/710/set"
    command_template: "{{ value }}"
    max: 50
    optimistic: false
    qos: 0
    retain: true

fredlcore, 2024-12-26:
Not sure what exactly you want to send additionally, but it has to be based upon what BSB-LAN "knows" about the parameters it gets from the heating system. So, for example, "max: 50" would require BSB-LAN to know that this is the maximum value for parameter 710. But we don't know that (except when experimenting or looking it up in the manual).
Or maybe I don't understand what you mean with providing selectors/inputs.
If you provide me with a clear example that shows what is sent now during auto-discovery, and what you want BSB-LAN to send instead, then I can look into it, but otherwise not.

apocaliss92, 2024-12-26:
Thanks for the quick reply!

The max value in this case is nothing strict to my heater, just ajnui utility to limit the input. 

This is an example for the value 1600, this is what bsb provides now as autodiscovery

Autodiscovery Topic: homeassistant/sensor/BSB-LAN/1600-97-100-610090368/config
unique_id: 1600-97-100-610090368
state_topic: BSB-LAN/0/9/1600/status
device:
  identifiers: BSB-LAN-94E68601F140
  manufacturer: bsb-lan.de
  model: 4.2.44
  name: BSB-LAN
name: 00-09 Acqua calda sanitaria (ACS) - 1600 - Reg di funzionamento

Something like this would make this an input entity instead of just a sensor

Autodiscovery Topic: homeassistant/number/BSB-LAN/1600-97-100-610090368/config
state_topic: "BSB-LAN/0/9/1600/status"
name: BSB-LAN Modalità sanitari
command_topic: "BSB-LAN/0/9/1600/set"
command_template: "{{ value.split(' -')[0]}}"
options:
  - "0 - Off"
  - "1 - On"
  - "2 - Eco"

The main difference should only be in the domain of the autodiscovery topic (select/number instead of sensor) and provide a command_topic (command_template if the value to send is different from what is shown)

Of course only the data bsb is aware of should be give, I think the options it's part of it since the web interface shows them

Please let me know if you need any other info, I will try to be as clear as possible and sorry for the text formatting, I'm from my smartphone

apocaliss92, 2024-12-26:
This is instead an input example for the value 710

Autodiscovery topic: homeassistant/sensor/BSB-LAN/710-97-100-610090368/config
unique_id: 710-97-100-610090368
state_topic: BSB-LAN/0/7/710/status
icon: mdi:thermometer
unit_of_measurement: °C
state_class: measurement
device:
  identifiers: BSB-LAN-94E68601F140
  manufacturer: bsb-lan.de
  model: 4.2.44
  name: BSB-LAN
name: 00-07 Circuito riscaldamento 1 - 710 - Setpoint comfort

What could be to make it a numeric input

Autodiscovery topic: homeassistant/number/BSB-LAN/710-97-100-610090368/config
state_topic: "BSB-LAN/0/7/710/status"
name: BSB-LAN Setpoint riscaldamento confort
command_topic: "BSB-LAN/0/7/710/set"
command_template: "{{ value }}"

These new autodiscovery topics could be just added on top of the existing ones without adding breaking changes, since their domains will be different and won't interfere with the current sensors provided

fredlcore, 2024-12-26:
Regarding 1600, I cannot reproduce this. On my system, it has the autodiscovery topic `homeassistant/select/BSB-LAN/1600-255-255-40528` and it comes with the settings that you can choose from.
For 710, I cannot send it as `number` because `number` requires additional information (IIRC min/max) that we don't know. If these are omitted, it will use default ranges which may or may not cover the full range.

apocaliss92, 2024-12-26:
Oh I see, do you have any any idea why I would not get it as selector? 

About the numbers, I think the min and max are not required values, at least they are marked as optional on documentation, I will give it a try without later today and let you know

fredlcore, 2024-12-26:
No, further down, like this:
`{0x313D0571,  VT_ENUM,            1600,  S_BA_TEXT,sizeof(E_BA_TW),    E_BA_TW,      DEFAULT_FLAG, 255, 255}, // Betriebsart`

apocaliss92, 2024-12-26:
{0x313D0571,  VT_ENUM,            1600,  S1600_0,  sizeof(E1600_0),    E1600_0,      DEFAULT_FLAG, 97, 100}, // Reg di funzionamento
{0x313D0571,  VT_ENUM,            1600,  S_BA_TEXT,sizeof(E_BA_TW),    E_BA_TW,      DEFAULT_FLAG, 255, 255}, // Reg di funzionamento

fredlcore, 2024-12-26:
Could it be that you have BSB-LAN in the settings running with write-enabling set to off? This turns all parameters to read-only and those are transmitted as sensors.

closed 2024-12-26 by fredlcore

apocaliss92, 2024-12-26:
Yes it was set to read-only when I requested the parameters list, I have enabled it only yesterday unfortunately

apocaliss92, 2024-12-26:
Should I just resend the email with the new list with the write mode active?

fredlcore, 2024-12-26:
No, it has nothing to do with the list. Just run /M1 again to send the updated configuration to MQTT/HA.

apocaliss92, 2024-12-26:
Thank you very much, will try right away