fredlcore/BSB-LAN #702: [FEATURE REQUEST] Log INF telegrams and "foreign" responses optionally via MQTT

uschindler, 2024-12-30:
**Is your feature request related to a problem? Please describe.**
When looking at the serial output, the code logs several additional messages:
- INF Telegrams like "Brennerstart". Interestingly the log output does not have a parameter number, so I am not sure if this is rellay related to a parameter. My knowledge of the protocol is not too good, I just noticed those log lines.
- You also log some parameters requested by the Display Unit (DSP), like Kesseltemperatur and a few others. There are parameter numbers. The responses (HEIZ->DSP) are also logged. So actually we just listen on the bus and receive a lot of communication that we can use for our purposes.

**Describe the solution you'd like**
It would be great to get the INF Telegrams sent without requests from The Heating reported also via MQTT. I did not see them anywhere (I tried several logging settings). To me it looks like they are never reported via MQTT, only logged to serial console. If they have no parameter number, maybe have an additional MQTT path that you can subscribe for informational messages. This would allow to trigger further updates more frequently. I'd like to report some stuff like Kesseltemperatur or FA Betriebsanzeige in 5 second intervals like the display uint, but only when the Brenner has started.
In addition, if you could optionally enable logging of telegrams which were coming on some other bus participant to MQTT, it would make a lot of sense. The Display Unit all the time requests in 10-15 seconds some general parameters. We don't need to poll them, if the respinse messages are just logged to the correct MQTT topic. So when a special logging option is enabled, all parameters will be logged to MQTT, although the teölegram was for somebody else.

**Describe alternatives you've considered**
Dramatically decrease the polling interval by sending more /poll MQTT resuqest. This puts a lot of additional traffic on the bus. So I'd like to avoid this. Especially the INF telegram "Brennerstart" is coming exactly when the "Flamme" goes on on the display. This can only be emulated with very low polls, although "Brennerstart" has no parameter number (see above).

**Describe your own contribution**
This would help people to get some parameters and informational messages without requesting them without any delay.

uschindler, 2024-12-30:
Of course, if for the "parameterless" INF telegrams a separate topic is added, don't forget to announce it for HASS autodiscovery.

fredlcore, 2024-12-30:
All parameters that BSB-LAN detects on the bus that match a parameter are automatically pushed to MQTT, if so configured.
"Brennerlaufzeit" and other status parameters are logged in parameters 20000+ and likewise pushed to MQTT, if so configured.
Raw data messages won't be pushed to MQTT, but can be logged to SD-card for analytics purposes. If you find a telegram that is of use for you, you can add a parameter in the 10200+ range and then use/push/poll it like any other parameter.
If you really want the full string, you could do it by adding your own code in `BSB_LAN_custom.h` and use this to push to whatever MQTT topic you like.

closed 2024-12-30 by fredlcore

uschindler, 2024-12-30:
> All parameters that BSB-LAN detects on the bus that match a parameter are automatically pushed to MQTT, if so configured.

Do you mean to add those parameters to the general list of parameters to log of the config? There is no special config for MQTT logging.

I tried it, the Display Unit sends a query for 8310 (Brennertemperatur) every 5 seconds. The response is an unicast message which is directly addressed to display unit. Although I configured it in the log settings, it is not published to MQTT.

Maybe you misunderstood my request: I'd like to log all parameters which were requested by other bus participants (not queries by myself), so their target address ist not 66 (BSB-LAN), but e.g. the display unit.

fredlcore, 2024-12-30:
No, I understood what you want. I have it running here myself. The heater (device ID 1) informs the controller (device ID 0) about the flow temperature (parameter 10044 in my setup). This is automatically pushed to MQTT.
Should be the same on your system.

uschindler, 2024-12-30:
Hmmm,
this is how it is logged in the console:
```
DSP1->HEIZ QUR 8310.0 Uhrzeit - Kesseltemperatur-Istwert:
DC 8A 00 0B 06 3D 0D 05 19 4F 8C
HEIZ->DSP1 ANS 8310.0 Uhrzeit - Kesseltemperatur-Istwert: 36.8 °C
DC 80 0A 0E 07 0D 3D 05 19 00 09 31 23 61
DSP1->HEIZ QUR 8310.0 Uhrzeit - Kesseltemperatur-Istwert:
DC 8A 00 0B 06 3D 0D 05 19 4F 8C
HEIZ->DSP1 ANS 8310.0 Uhrzeit - Kesseltemperatur-Istwert: 36.8 °C
DC 80 0A 0E 07 0D 3D 05 19 00 09 31 23 61
```

I have the folowing config enabled:
[image]

In MQTT nothing about 8310 is logged, unless I poll for it explicit:
```
LAN->HEIZ QUR 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert:
DC C2 00 0B 06 3D 0D 05 19 A7 86
HEIZ->LAN ANS 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 36.8 °C
DC 80 42 0E 07 0D 3D 05 19 00 09 31 1B 30
#8310: 36.8 °C
Publishing to topic: BSB-LAN/0/21/8310/status
Payload: 36.8
Successfully published...
```

fredlcore, 2025-01-14:
Sorry, I didn't get a notification for your last post for some reason. 
Only values are pushed that are actively changed in some part of the system (heater display, room unit, BSB-LAN). What you see here is the 10 second poll interval that the display uses to display the current temperature on the display. These are obviously not pushed to MQTT as it would fill your database in no time (and even if you are careful enough, most others are not and will complain that their harddisk is full). But it's the same as if you would poll this parameter from elsewhere every 10 seconds, it's not an indication that the value actually changed.

fredlcore, 2025-01-15:
This is how you could add a new parameter 10014 to display the burner status in `BSB_LAN_custom_defs.h`, which should be pushed (either as an unsigned byte or as a binary representation, not sure right now) automatically if such a telegram is discovered on the bus:

```
const char S_BURNER[] = "Status Brenner";
const char E_BURNER[] = {
"\x00"
"\x04\x04 " "Brennerstufe 1 Ein\0"
"\x10\x10 " "Brennerstufe 1+2 Ein"
};
```
and then in cmd_tbl:
```
{0x05000213,  VT_CUSTOM_BIT,      10014, S_BURNER, sizeof(E_BURNER),   E_BURNER,     FL_RONLY, 255, 255},
```

uschindler, 2025-01-16:
Hi,
Thanks. I will update to latest version and add this to the custom devs and provide debugging output. More or less my main interest is to get a active push notification from Brenner if something changes to query other related parameters.

Uwe

fredlcore, 2025-01-17:
That should work once you have added arameter 10014.

fredlcore, 2025-01-17:
> It would be good if we published somewhere how to adjust your config-defs to get this in. In new versions you can probably add this by default.

Well, it's here, and you got it working without any issues, so that's that.

> Keep in mind that Home Assistant only updates the database if the value actually changes, so when Kesseltemperatur does not change, nothing happens.

Good for Home Assistant, but that's by far not the only home automation system that people are using. Plus, the heater display is the only device that constantly polls (one and the same) parameter, usually the boiler temperature. As I said, you can easily do that with an automation as I have shown in the video. No justification at all to clutter the configuration with an almost redundant configuration option.

uschindler, 2025-01-17:
> As I said, you can easily do that with an automation as I have shown in the video.

That's working since long time. The issue was to just not overload the bus with several devices querying the same parameter over and over.

If you are interested, the automation for polling now works like that: Whenever a parameter like temperature, Betriebsanzeige FA,... of the burner changed on the last polling, it triggers another update after 8 seconds. Once the burner gets inactive there are less updates (e.g., temperature changes no longer within those 8 seconds), it silences itsself. The problem was now: How to retrigger those without any polling? And here comes the new 10014 into play. As soon as this triggers without any polling, the game with 8 second updated of important burner parameters starts again. Previously the change in temperatures triggering more frequent updates were often delayed because they were only triggered on the normal poll job (much too late). P.S.: I don't care about the value in 10014, it just needs to change.

fredlcore, 2025-01-17:
You can't really "overload" the bus, only your database. The bus is fine with polls every few seconds, as you can see when you query a category, for example. So if it's true what you say that e.g. HA only updates the database on changes, you could have just polled one of the 800X parameters which give you the burner status (8005 in my case) and then do the same as you do now with 10014.

My comment was directed at your request that BSB-LAN should "Log all status telegrams in response to queries by other bus participants". And that's nonsense, tbh. The only other bus participant that does queries regularly is the heater's display. And that is (depending on configuration) the boiler temperature. That one changes usually a lot during the heating season, so it will fill up your database due to changes in the 0.1 degree range. But as I said, if this is what you really want, you can just poll this parameter by yourself without having a special setting in BSB-LAN to do that for you. 

P.S.: If you only look at the change of 10014, you could still have a change from burner stage 1 to stage 1+2, so better test between a change from zero to non-zero and vice versa rather than any change. But that's just a hint, no need to discuss the details here.

fredlcore, 2025-01-17:
The problems with pushes is that you may miss one - if BSB-LAN is doing some kind of maintenance work and an INF telegram passes by undetected, you may act on false assumptions, especially if you just switch a status based on a changing state.
Polling therefore is always safer than waiting for pushed telegrams...

uschindler, 2025-01-17:
> The problems with pushes is that you may miss one

Thats no problem, but thanks for reminding me. The use case for the pushes is to get a notification on status changes earlier (more "in time") if you miss it, the next regular polling will get it anyways. The trigger for the frequent brenner status updates is using both "if 10014 changes OR any other (polled) temperature/status value changed recently".

In addition, pollung is also not 100% safe. If for some reason BSB-LAN is restarting and you poll a topic you might not get an respone. QoS=1 does not help in that case and QoS=2 is not supported by pubsubclient.

uschindler, 2025-01-17:
Last question, is it possible to extend this definition in custom_defs to decide 3 bytes instead of one to help you with issue #709?

```
{0x05000213,  VT_CUSTOM_BIT,      10014, S_BURNER, sizeof(E_BURNER),   E_BURNER,     FL_RONLY, 255, 255},
```

VT_CUSTOM_BIT only prints the first byte as bitmap into the MQTT message. Can I change the definition to print all three bytes?

At moment the gas burner only has 00000000 and 00000100 as distinct values in first byte. Schornsteinfeger Mode does not change anything.

fredlcore, 2025-01-17:
No, you have to add another parameter like that one:

```
const char S_SONDER[] = "Anzeige Sonderbetrieb";
const char E_SONDER[] = {
"\x01\x01 "    // position, length
};
```
```
{0x05000213,  VT_CUSTOM_BYTE,     10015, S_SONDER, sizeof(E_SONDER),   E_SONDER,     FL_RONLY, 255, 255},
```

uschindler, 2025-01-18:
After reading the code of `print_trelegram.h`, to print all 3 bytes use (unfortunately it will be printed as decimal not hex):

> ```
> const char S_SONDER[] = "Burner INF Telegram Raw";
> const char E_SONDER[] = {
> "\x00\x03 "    // position, length
> };
>
> {0x05000213,  VT_CUSTOM_BYTE,     10015, S_SONDER, sizeof(E_SONDER),   E_SONDER,     FL_RONLY, 255, 255},
> ```

This will then print a long created from all 3 bytes.

My problem is that the default implementation seems to only print the first parameter found in cmd_table that matches heater spec in last columns. So combining all 3 bytes into one `VT_CUSTOM_BYTE` looks best.

Many thanks anyways. Once I figured out how it works best I will post a note to the other issue how to do debugging via MQTT of the burner inf message.