fredlcore/BSB-LAN #703: WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/BSB-LAN/501.1-211-127-501298/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic

clausprivat, 2024-12-30:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
Please note: Versions 1.8.19 and below are not supported by this project anymore.

Version: 2.3.4
Date: 2024-12-03T10:03:03.490Z
CLI Version: 1.1.1
Copyright © 2024 Arduino SA

**Version of the ESP32 or Arduino framework**  
See ***Tools/Board/Board Manager*** for exact version you are using

esp32 by Espressif Systems 3.0.7 installed

**BSB-LAN Version**  
As displayed in the web-configuration or copy from `bsb-version.h` file. Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly. We don't fix bugs for discontinued platforms (such as Arduino Mega).

#define MAJOR "4"
#define MINOR "2"
#define PATCH "46"
#define COMPILETIME "20241220005136"

**Architecture**  
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.)

ESP32 NodeMCU

**Bus system**  
Which bus BSB-LAN is running on (BSB, LPB, PPS)?

BSB

**Describe the bug**  
A clear and concise description of what the bug is.

In HA, the MQTT auto discovery works fine, but some sonsors.*, text.* actors are not shown.
e.g. this topic works fine and gets received in MQTT and discovered and values are shown as sensor:  BSB-LAN/0/1/501
e.g. this topic  does not work and gets received in MQTT but NOT discovered and values are NOT shown as sensor: BSB-LAN/0/1/501.1 
in "HA > System > Protokolle" I see the following logging

e.g. for 501.1
2024-12-30 10:57:17.518 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/BSB-LAN/501.1-211-127-501298/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Log files - Bug reports without log files will be closed**  
If you cannot provide a log file, explain here why in order to prevent closing this bug report. Attach or copy/paste the log of the Arduino IDE's serial monitor when performing the actions that lead to the error. If possible start the log right before BSB-LAN boots (pressing the reset button on the board).

The Logfile from the Adaptor is fine, and everythings runs with the bsb-lan, but the issue is within HA mqtt, here this log:

[2024.12.30_HA_system_log_topic_contains_not_allowed_characters_01.txt](https://github.com/user-attachments/files/18274047/2024.12.30_HA_system_log_topic_contains_not_allowed_characters_01.txt)

**Expected behavior**  
A clear and concise description of what you expected to happen.

It would be great do see all sensors in HA for all weekdays.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**  
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**  
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. Safari, FireFox]
 - Browser version [e.g. 22]

**Additional context**  
Add any other context about the problem here.

I assume that the dot  within the names (e.g. 501.1) should be exchanged with any symbol that is allowed for mqtt names.

The only place where I found this string was Line 5201 in BSB_LAN_custom_defs.h 

const cmd_t cmdtbl[]={
...
{0x053D0A8D,  VT_TIMEPROG,        501.1, S_WD_2,   0,                  NULL,         DEFAULT_FLAG, 211, 127},

[BSB_LAN_custom_defs.txt](https://github.com/user-attachments/files/18274874/BSB_LAN_custom_defs.txt)

Please help me to fix this.

Thank you for your support!

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/user-attachments/files/18274839/BSB_LAN_config.txt)

fredlcore, 2024-12-30:
Thanks for highlighting this, I just had another user mentioning this a few days ago and provided a fix a few days ago. Can you please update to the most recent version, issue a /M1 again to push the new auto-discovery messages (ideally clear all retained messages from your broker) and see if that fixes the problem? Thanks!

fredlcore, 2024-12-31:
Great, thanks for testing this so quickly!

closed 2024-12-31 by fredlcore