fredlcore/BSB-LAN #711: [BUG]

matejek, 2025-01-16:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.
I tried to find resolution via FAQ and Community and github: no success

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
Please note: Versions 1.8.19 and below are not supported by this project anymore.
Arduino IDE: V2.3.4

**Version of the ESP32 or Arduino framework**  
See ***Tools/Board/Board Manager*** for exact version you are using
esp32 by Espressif Systems: V3.0.7

**BSB-LAN Version**  
As displayed in the web-configuration or copy from `bsb-version.h` file. Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly. We don't fix bugs for discontinued platforms (such as Arduino Mega).
#define MAJOR "4"
#define MINOR "2"
#define PATCH "60"
#define COMPILETIME "20250112021726"

**Architecture**  
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.)
ESP32 DEVKITV1 (ESP WROOM 32) - worked well with very old version BSB-LAN

**Bus system**  
Which bus BSB-LAN is running on (BSB, LPB, PPS)?
BSB

**Describe the bug**  
I cannot configure BSB-LAN using web-interface, as I am not able to enter the WIFI-AccessPoint "BSB-LAN"
I can access the WIFI but entering the password "BSB-LPB-PPS-LAN" leads to "wrong password"

**To Reproduce**  
Steps to reproduce the behavior:
1. Power on and boot ESP32
2. On Smartphone select "BSB-LAN" WIFI
3. Entering "BSB-LPB-PPS-LAN" as password
4. "wrong password"

**Log files - Bug reports without log files will be closed**  
If you cannot provide a log file, explain here why in order to prevent closing this bug report. Attach or copy/paste the log of the Arduino IDE's serial monitor when performing the actions that lead to the error. If possible start the log right before BSB-LAN boots (pressing the reset button on the board).
I cannot access the device via WIFI. No insight how I could retrieve access related logs from BSB-LAN on Esp32

**Expected behavior**  
A clear and concise description of what you expected to happen.
Access BSB_LAN configuration via ttp://192.168.4.1

**Screenshots**  
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (if applicable, please complete the following information):**  
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**  
 - Device: Samsung S24Ultra
 - OS:Android14, OneUI6.1
 - Browser n/a
 - Browser version n/a

**Additional context**  
Add any other context about the problem here.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.h.txt](https://github.com/user-attachments/files/18435843/BSB_LAN_config.h.txt)

matejek, 2025-01-16:
[image]

fredlcore, 2025-01-16:
You have to add a serial monitor log, see the YouTube video in the BSB-LAN channel for instructions.

matejek, 2025-01-16:
I started to edit the BSB_LAN_config.h file with my personal WLAN config - that seems to work and I have access now.

Still, I assume fallback-AP is not working (for me).

closed 2025-01-16 by matejek

fredlcore, 2025-01-16:
Well, if you want help and get this fixed, you have to submit a serial monitor log from when the ESP32 boots (and with your working WiFi credentials removed) - otherwise there is no chance for me to make any analysis. But since it works for others, I assume it's a local issue.