fredlcore/BSB-LAN #713: [BUG] Compilation warning with newest ESP32 v3.1.1

uschindler, 2025-01-18:
With latest ESP32 devkit release, the following compilation warning is presented. Some API usage in BSBLAN's code is deprecated.

```
xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Ethernet\src\ETH.cpp: In member function 'void ETHClass::end()':
xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Ethernet\src\ETH.cpp:886:22: warning: 'void NetworkEvents::removeEvent(NetworkEventFuncCb, arduino_event_id_t)' is deprecated: removing functional callbacks via pointer is deprecated, use removeEvent(network_event_handle_t) instead [-Wdeprecated-declarations]
  886 |   Network.removeEvent(onEthConnected, ARDUINO_EVENT_ETH_CONNECTED);
      |   ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Network\src/Network.h:9,
                 from xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Ethernet\src\ETH.h:70,
                 from xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Ethernet\src\ETH.cpp:24:
xxx\packages\esp32\hardware\esp32\3.1.1\libraries\Network\src/NetworkEvents.h:204:8: note: declared here
  204 |   void removeEvent(NetworkEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX)
      |        ^~~~~~~~~~~
```

This can maybe be fixed a bit later, but I want to open a bug report in case the deprecated usae gets unsupported. At moment, the resulting build output still works fine and no network problems occurred.

Version of "ESP32 by Expressif Systems" is 3.1.1.

fredlcore, 2025-01-18:
Why do you remove all that is necessary for a bug report? It is clearly stated that bug reports that do not provide the required information will be closed immediately.

closed 2025-01-18 by fredlcore

uschindler, 2025-01-18:
This was not necessary meant as bug report.

In short: the new version of ESP32 API 3.1.1 came out. This caused a compiler warning. Not urgent. This just to inform you.

If you are so ignorant I will not try to help anymore. You have closed it although all information was given it's not my problem. Sorry.

fredlcore, 2025-01-18:
You submitted it as a bug report (that's what the subject still says), and you deleted all the questions with the requested information.
If you had followed and provided all the information requested in the bug report, you would have probably realized yourself that this has nothing to do with BSB-LAN. The warnings come from the network library as part of the ESP32 framework, both the use of the function as well as the definition of the function. So there is nothing I can do in BSB-LAN's code to change that. It's probably just a warning from the coders to other coders of the framework to identify the locations where this is relevant and fix it at some point.

You don't know how often this happens that people think they can take the liberty of omitting information that I deem relevant because they think that I don't need it, and then something like this is the result. Deliberately ignoring a request that I make to people who want me to fix things is disrespectful, and I won't have that. To the least because others will see these kinds of bug reports and think that they can also just ignore the requested information.

But yeah, I'm the one who is ignorant.

uschindler, 2025-01-18:
I changed the title.

It might be correct that this comes from the devkit itself. I just wanted to share the suddenly appearing warning with the project owners. It also costs time to open issues, but if somebody finds a warning it's a good idea to help to fix it.

There is no need to fix it asap and it is not meant to urge you to do something, it was just an early reminder that there's a warning.

I am part of many (huge) projects on GitHub via the Apache Software Foundation and learned that closing issues without any positive sounding feedback is unrespectful. I never do this. Of course, I criticize people openly (which makes people call me "typical German"), but I give them generally a chance to edit their submission.

fredlcore, 2025-01-18:
It is clearly stated in the text that you removed from the bug report what I do in these cases, so I don't really understand where the expectation comes from that I should rather give some positive feedback for this. 

Yes, opening a bug report costs time. But not doing it properly costs me even more time, especially when like in this case, it turns out that the issue is unrelated to BSB-LAN. 
By filling out a proper bug report, people are at least forced to spend a few more minutes to think about the bug they are reporting. And I have often realized myself that a supposed bug I found actually wasn't one, or at least not one related to the project I was reporting it to. Just like in this case, someone with C++ knowledge would have probably realized after going through these lines one more time that these warning messages do not stem from BSB-LAN, and thus the bug report would have had to be made with Espressif.

And nobody would have stopped you to edit your bug report or create a new one. Closing such an issue simply removes it from the publicly visible "to-do-list" where an improper bug report would make people think that it's ok to just do it like that. 

You do the things the way you do (including calling me ignorant), and I do the things the way I do (including setting clear rules and act accordingly). Let's leave it at that.

uschindler, 2025-01-18:
> And nobody would have stopped you to edit your bug report or create a new one. 

In your project's settings there's no way to reopen the bug report for the reporter. This makes closing a bug report by the project owner look like a dead end.

Maybe a suggestion: you can tag issues with many custom or pre-provided labels. Stiff you want too be implemented can be tagged differently. Stuff like this could be marked as "Pending" or "Maintenance". Having an empty issue list in a projects looks strange for an outsider. This looks like a dead project where nobody is interested or like one where "issues are not wanted because my code is bug free and I refuse to fix stuff".

fredlcore, 2025-01-18:
Why should users who ignore the rules have the liberty to re-open their issue? I really don't get that. As you can see perfectly here, a closed issue does not end a discussion, including a subsequent re-opening of the issue (as has happened several times in the past here as well). But if people don't realize that they don't play by the rules, then yes, it should be a dead end.

And if you've made the experience that people think a project is "dead" if there are no open issues (which btw is not the case here), then rest assured, this is not the feedback that I'm getting in this project.