fredlcore/BSB-LAN #715: Is anyone else having issues if ESP32 is powered solely bei PoE?

fredlcore, 2025-01-19:
### Discussed in https://github.com/fredlcore/BSB-LAN/discussions/714

<div type='discussions-op-text'>

<sup>Originally posted by **parsley** January 19, 2025</sup>
Hi!

In November 2024 I updated my BSB-LAN setup from Arduino Due to Olimex ESP32-PoE-ISO. One of my goals was to power it with PoE only. 
But BSB-LAN on the ESP32 did never run as stable as on the Due and frequently rebooted especially if I browsed the Web-UI. I was unable to debug this issue but at some point I realized the issue did not occur as long as my PC was connected via USB for debugging.
After that I "fixed" the issue by powering the ESP32-PoE-ISO with a 5V power supply. Today I have an uptime of 42 days on version 4.2.37-20241207010916. But I still would like to get rid of the extra 5V power supply...

PoE is providet by an Unifi USW-24-PoE rated at 44-57V (Pins 1, 2+; 3, 6-).
I'm not 100% sure but remamber having instant issues wenn calling bsb-lan web pages that trigger BSB bus communication and slightly less instant issues when calling pages like /C or the index page that do not trigger BSB communication.

Does anyone have the same issue or any clue why PoE is not sufficient?

Regards!</div>

fredlcore, 2025-01-19:
@parsley: I have created an issue from this, so it's easier for others to find and refer to it in case we found a solution.

Yes, unfortunately, that could be a hardware issue from Olimex which we couldn't completely solve yet because all my testers in the end did not follow up on me :(.
See this issue over at Olimex for some background:
https://github.com/OLIMEX/ESP32-EVB/issues/57#issuecomment-2162946166
as well as here:
https://github.com/OLIMEX/ESP32-POE/issues/49
and Olimex kind of confirmed that it could be the same issue:
https://github.com/OLIMEX/ESP32-EVB/issues/57#issuecomment-2421587187

If you are willing to make some tests and are comfortable with editing BSB-LAN's sourcecode, I'd be willing to guide you through the process. Because of course, if there is anything that I can do on the source-code side, then I'm happy to do that. But since I don't have PoE myself, I cannot test it myself...

fredlcore, 2025-01-19:
In order to figure out whether something can be done on the software side, these are the steps that would need to be tested:

1. Could you replace bsb.cpp in the src/BSB folder with the bsb.cpp that Dan Koloff attached to [one of his posts](https://github.com/OLIMEX/ESP32-EVB/issues/57#issuecomment-2165214478)? Can you confirm that the problem of reboots goes away then? If yes, could you still also access the heater and get valid data? Or do his changes simply mean you can access the web-interface, but due to the invalid data coming in, it’s basically impossible to interact with the heater?

2. Could you do the following tests with the BSB-LAN adapter removed(!) from BSB-LAN and after each code change revert to the original state before running the next test (of course, removing the adapter means you cannot access the heater, only the web-interface):
2.1. Can you confirm that the problems of reboots are **still there** when the adapter is **removed**?
2.2. Can you change line 26 in bsb.cpp from "serial = &Serial1;" to "serial = &Serial;" and confirm that the problem goes away and accessing the web-interface is possible without any problems?
2.3. Can you change line 48 in bsb.cpp from "Serial1.begin(4800, SERIAL_8O1, rx_pin, tx_pin);" to "Serial1.begin(4800, SERIAL_8O1, 1, tx_pin);“ and confirm that the problem goes away?
2.4. If the problem goes away, could you test with pins 4, 36, 16, 13, 15, 2 and 14 and see which ones create a problem and which ones would work?

If all of this turns out to be true, the barrel connector / the PoE power-supply circuitry is somehow affecting pin 36 (and maybe other pins, depending on the outcome of 2.4.). If one of the pins from 2.4. work, I could change the RX pin in future hardware revisions of BSB-LAN if Olimex isn’t going to fix this somehow. And existing users could just use a small jumper wire, if they rely on powering the Olimex through the barrel connector.

parsley, 2025-01-19:
First of all: 
Wow! What an insanely fast response! 
I‘m sorry I won’t be that fast with further responses. Mostly due to lack of spare time. 

That said: 
Thank you for the additional information. 
I'm happy to support you and the project. Code editing is no issue and hardware tools are available as well. But my single ESP32-PoE-ISO is taking care of my family home heating right now. 🫣😉
I‘ll read through the links you posted and will see what I can do. 

By the way: One reason why this topic pops up right now is that yesterday Youtube suggested this video: https://www.youtube.com/watch?v=NICRlC5gzBo It's two years old but new for me and sparked my desire to solve this issue...

uschindler, 2025-01-19:
Hi,
I started to use a OLIMEX ESP32 PoE ISO in October. At beginning it worked well and it was really nicely integrated in the heating, as you would wish it:

[image]

This worked well for about two weeks. It then started to suddenly disconnect from ethernet, but it was not rebooting! In the serial log, you were able to see that it reported "ethernet disconnected" and reconnected withing milliseconds. This happeneded about once per day, mostly during morning time when the inner part of heating got hotter due to warmup after night. In addition to several possibly related disconnections due to a bad MQTT connections stability (which may has to do with some lost packets on the LAN wire and a wrong implementation of the pubsubclient, so it wasn't able to detect this status correctly (see issue #682 and PR #683), this looked like I was able to live with.

In December 15th then it started to get worse. The physical ethernet disconnects started to happen every half hour and later the device turned into a on/off flipper. The switch showed "connection on and connection off" all the time and DHCP server ackowledged the same IP over and over. The device was never rebooting.

The reason for this is well known and OLIMEX did not fix it and there is only one seolution to fix it (I did not try it, see below). The problem is that the OLIMEX needs to take at least 12 mA of current from the switch, otherwise it could be switched off by the switch (this is called MPS "Maintain power signature" in PoE). Therefor they added a larger 1W resistor with 4.7kOhm in newer versions of the board in parallel to the ~48 V (52 Volts on my switch). If you calculate how much power is translated to heat you will figure out that this exactly matches the 12 mA, but costs more than half a Watt of power. The resistor gets really really hot (its on the donw side of the board, a larger SMD one). The resistor is close to 2 Elkos and the large black voltage transformer. The heating of the resistor wanms both elkos and the transfomer and possibly (accoring to forums) the elkos fail due to 90 °C of heat, especially in small enclosures or inside a Brötje Heater.

OLIMEX Forum:
- Original complaint: https://www.olimex.com/forum/index.php?topic=7255.0
- Again: https://www.olimex.com/forum/index.php?topic=8639.0

I disabled PoE on my switch and connected an USB device, but the ethernet connection did not restore to normal operation. It looks like in my case the large heat from the resistor "melted" the transformer or one of the elkos around, possibly it may have also broke the ethernet plug. Not fully sure! In my case the Ethernet is now .... damanged! As I did not want to buy a new ESP32, I switched over to WiFi with an USB plug that I connected to with clamps to the 230 volts output top-right on the LMU in above picture. My new installation does not looks so nice anymore, but is completely stable. No reboots, no disconnects. The cable is still there if I am going to replace the broken platina later. I placed the antenna behind the display unit, so it can get a connection in the otherwise "Faraday Cage" treated Brötje box. I have no actual photo of that, will take one once it is open again.

If your ethernet part is still working without PoE (disabled in switch and connected USB), you may to take a solder and remove the resistor R41 from the lower side. It is completely useless, as the device together with the BSB-LAN Adaptor takes more than 12 mA. This spares between 0.5 and 1 Watt of wasted power. According to the switch, the device with the resistor takes 2.7 W, so you can save a lot of power (and heat).

fredlcore, 2025-01-19:
Thanks, this is also mentioned in the video @parsley linked above, but this is a different (albeit similarly disturbing) error on behalf of Olimex.
I'd like to focus here on the initial issue because even if the PoE implementation is broken also otherwise, solving it would also help those users who power their Olimex with a barrel connector.

uschindler, 2025-01-19:
The behaviour on my side was exatly like described above. Sometimes when connecting to the web page it returned nothing. At some point it was not even possible to upload a new firmware as conmnection broke.

So this looks like a hardware issue.

The original reporter said, that PoE was unstable and the USB barrel was stable. So this is the opposite of the issues mentioned above:

> But BSB-LAN on the ESP32 did never run as stable as on the Due and frequently rebooted especially if I browsed the Web-UI. I was unable to debug this issue _but at some point I realized the issue **did not** occur as long as my PC was connected via USB for debugging._

So in my opinion, he has the well known PoE issue, so we can't solve this in software.

Or did I misunderstand the original issue description?

fredlcore, 2025-01-19:
Please take the time and read through the linked issues before making any assumptions. USB power supply and barrel connector supply are two completely different things. 
It is a hardware issue, but it is not related to the "well known PoE issue". I have even linked to the comment where @DanKoloff explains what the issue is here, and yes, it would be solvable in software, the question being whether it can be done with the way the BSB-LAN adapter is designed. 
In any case, it is not related to anything you have described. So please let's focus here on the issue that was raised by @parsley.

parsley, 2025-01-19:
@uschindler thanks for the warning. Luckily I disabled PoE once I connected the 5V PSU. :-)

But on your second assumption you are wrong where as @fredlcore is right. This issue is not about the heat of that MPS-resistor. (However I must admit, that the heat caused by that resistor will be the next issue to be solved after my issue raised in this thread.)

@fredlcore I'm supprised I already found the time to read through all the links you posted but I actually did! :) The most interesting part for me was the hint towards serial rx. This rang a bell and I searched through my mails. I found a mail sent by me (24.11.2024 13:52) and an answer form you (24.11.2024 17:56) but I have to dig deeper and reinvestigate. So far I do not remember exactly what I found back them...

parsley, 2025-01-19:
I forgot to explain/respond to this:

> Well, it won't have to impact your heating in any way - ...

I'm afraid it's not that easy in my case: I gather room temperature of all rooms with my Timberwolf Server, calculate a mean value and send it via BSB-LAN to the boiler which heats accordingly. This works so well, that I do not use any per-room-valve-regulation anymore and by that could further increase the efficiency of my boiler.
That's why it's not so easy for me during winter to test this stuff. Sorry.

But nevertheless I'll keep investigating. If I should not come up with results during this heating season please feel free to ping me in spring! 😉

uschindler, 2025-01-19:
Hi,

> [@uschindler](https://github.com/uschindler) thanks for the warning. Luckily I disabled PoE once I connected the 5V PSU. :-)

This is not necessary, of you have the PoE ISO version. But when the PoE is active and gets 48 or 52 volts (with or without USB), the resistor heat burns the two Elkos and the voltage converter (big black block) on the other side of the platina.

> But on your second assumption you are wrong where as [@fredlcore](https://github.com/fredlcore) is right. This issue is not about the heat of that MPS-resistor. (However I must admit, that the heat caused by that resistor will be the next issue to be solved after my issue raised in this thread.)

Ok, all fine. To me it looks like there was addditional communication. I did not want to hijack this issue, but the story I posted above is EXACTLY fitting the issue's description: "Is anyone else having issues if ESP32 is powered solely bei PoE?". This is what I did and it worked well initially but then broke after a few weeks till the ethernet connection wasn't stable anymore. On the heat issue others also told, that some Olimex PoE-ISO burned in other ways like the voltage regulator gets unstable and you get sudden restarts. So the above story is very well describing an issue of this piece of hardware. The outcome of the overheating by the restsitor leads to different outcomes from failing ethernet to power issues on high load to failure of voltage regulator in combination with exploding elkos.

I am now connected to a stone-aged (black) 5V SAMSUNG Smartphone charger (not the newer white powerful ones) and gave up on using ethernet. Rock solid no crush, no reboots, no network stability issues.

You also said that the issues went away when you connected a 5V adapter. So just for my personal information: Could you describe in which combination of connections the issues started (solely PoE, with USB brick, some other external 5V input?) and if they also started over time? For me on the day of the setup all went well and my solely PoE powered Olimex PoE ISO had no restarts and the BSB communication through the UART was rock stable.

The only difference to my case is this last sentence and this is what @fredlcore is about: "I'm not 100% sure but remamber having instant issues wenn calling bsb-lan web pages that trigger BSB bus communication and slightly less instant issues when calling pages like /C or the index page that do not trigger BSB communication."

This issue did not happen for me neither with PoE only or with Ethernet-no-PoE or Wifi. So it seems to be only affecting some hardware variants and is caused by too low power on the serial UART. If this problem is focused here on, then please change the issue description. Thanks.

Good luck in solving the issue. I would be interested, because I'd like to buy a new Olimex ESP32 PoE ISO and desolder the resistor. If there are related problems with the UART and the main power it should be solved!

parsley, 2025-01-19:
Hi,

> This is not necessary, of you have the PoE ISO version. But when the PoE is ...

I sense you do differentiate between the ISO and the non-ISO version?

> Ok, all fine. To me it looks like there was addditional communication.

Kind of "yes and no" at the same time since originally the past communication was about som completely different topic: 
Last year I had a look at the schematic of the BSB-LAN adapter because I hand soldered my BSB-LAN adapter on some prototyping PCB. Back then I suspecteded, that the RX path might have an issue reaching the specified HIGH/LOW levels of the ESP32. Reading through https://github.com/OLIMEX/ESP32-EVB/issues/57 mentioned above by @fredlcore I was curious what I found and changed back then since DanKoloff said he found an issue with random data generated by a partially powered CH340T. Since besides mentioning the CH340T-RX-data in the other issue they also switch between serial and serial1 during debugging, I suspected a possible connection. Perhaps the BSB-LAN adapter sometimes also generates weird random RX "data"?
However: I slightly changed the schematic to actually be able to reach proper LOW level but still I do have the issue.

> ...the story I posted above is EXACTLY fitting the issue's description: "Is anyone else having issues if ESP32 is powered solely bei PoE?". 

Ok, now I understand what you meant. The difference is that my issue (as well as https://github.com/OLIMEX/ESP32-POE/issues/49) appear immediately and even on new boards once the PoE-ISO board is powered solely by PoE. What is even more interesting, that the same effect appears on the EVB boards (that do not even have PoE) when they get powered solely through their barrel connector (https://github.com/OLIMEX/ESP32-EVB/issues/57). I don't have an EVB so that info was new to me and it clearly points to an issue completely unrelated to PoE. (I think that is why @fredlcore asked whether you have read the thread about the EVB issue ;) )

Nevertheless it's good to know that (by time) the heat of the PoE MPS resistor may cause an issue with a very similar behavior. In the other thread the first assumption was some power issue and later on the suspect was the software acting weird on gibberish serial RX data. While my finding on the schematic might support the serial-RX-therory, your heat dried elkos might support the power-regulator-theory.

> ... Could you describe in which combination of connections the issues started (solely PoE, with USB brick, some other external 5V input?) and if they also started over time?

On the day I switched from Arduino Due based BSB-LAN setup to the ESP32-PoE-ISO based setup I always had my MacBook connected via USB to the new board (with Ethernet and PoE already connected at the same time). I did that because I wanted to check the serial monitor output to make sure everything works like previously with my old setup. And it did! No issue what so ever as long as my Mac was connected. The trouble started immediately once I disconnected my Mac. But since the issue disappeared as soon as I reconnected my MacBook via USB to the board I had no chance to investigate the serial monitor output. Soon I found out, that the issue also disappeared when I just connected an old iPhone USB charger with an USB-A-to-micro cable. Hence I suspected a power issue with PoE and checked CAT-cable and PoE voltage etc with no finding on that.

Since I already mounted my BSB-LAN on DIN rail in a cabinet I wanted to get rid of the phone charger. I took a KeyStone USB module, cut the 5V path on the tiny PCB inside the KeyStone module in between both USB connectors and soldered a pair of wires to GND and 5V on the USB connector facing the ESP32. These wires are connected to a MeanWell HDR-15-5 that now powers the ESP32 via the USB connector. On the other side of the KeyStone USB module I can connect my Mac. Since 5V is cut only GND, D+ and D- get connected to the Mac which is enough for communication. 
At that time I also switched off the PoE supply on the port of my network switch to spare the extra heat. 
Switching on or off PoE had no effect on my setup. As long as 5V are supplied on the USB micro connector of the ESP32 board ther is no instability whatsoever.
This is the setup that now has a uptime of 42 days with no issue.
(I tried to add some pictures here but failed...)

> For me on the day of the setup all went well and my solely PoE powered Olimex PoE ISO had no restarts and the BSB communication through the UART was rock stable.

By "communication through the UART" you mean the BSB bus UART with no PC connected via USB to the Olimex, right? That is interesting since I never had that. Perhaps @fredlcore might be interested in the exact BSB software version you used at that point since it perhaps might give a clue if or how the software might be involved in this issue? Do still know that version?

On a side note: 
Today I originally started this topic as a discussion because I wanted to know if others had similar problems and hopefully a solution that could help me without spending too much time for an avoidable investigation. But since this was a familiar issue to @fredlcore he immediately converted the discussion into an issue. Which is great! (Albeit a solution would have been even better. 🤪🤣) However this means that I have to review and verify the observations I remember and assumptions I made. That said I hope everything I wrote today is correct and not altered by wrong memories of mine.

fredlcore, 2025-01-20:
> I'm afraid it's not that easy in my case: I gather room temperature of all rooms with my Timberwolf Server, calculate a mean value and send it via BSB-LAN to the boiler which heats accordingly. This [...]

This is exactly the same situation as with me, and I often run these kinds of tests and checks in a similar way. Rest assured, if you remove BSB-LAN and set a "safe" comfortable setpoint temperature (such as 20 or 21 degrees or whatever), no one will notice if you do the abovementioned checks which won't take longer than 30-60 minutes. This is because even if you don't have a room temperature sensor, your heater will heat based on the outside temperature and the measured flow temperature. Plus, even in a badly insulated home, the temperature shouldn't drop more than 1 degree within one hour even if the heating is completely off.
Just make sure that you have your current configuration etc. ready to re-flash everything after you are done doing the tests, then I'm certain that the whole thing will be seamless.

fredlcore, 2025-01-20:
> To me it looks like there was addditional communication. I did not want to hijack this issue, but the story I posted above is EXACTLY fitting the issue's description: "Is anyone else having issues if ESP32 is powered solely bei PoE?".

It may fit the headline, but not the description of the problem in the threads linked in my first post. But never mind.
And no, the only additional communication is the discussion in the threads that I linked to. The mails that @parsley is referring to were about observations of resistors on the Olimex in combination with the ones on my adapter, but these are irrelevant to this case because - again, as written above - it might well be that the problems are still there even if the adapter is removed because the irregular data is generated due to a hardware issue with the CH304T that is not completely unpowered and thus has some kind of "data leakage" on pin 36 that generates data that BSB-LAN cannot handle. That's how Dan Koloff could reproduce the problem without even having a BSB-LAN adapter, just by running the software. Again, this is all discussed at great length in the linked issues, so let's not do any further guessing here but focus on the steps that I outlined above. Once these are done and my assumptions are wrong, then we can continue with discussing other potential approaches.

 > For me on the day of the setup all went well and my solely PoE powered Olimex PoE ISO had no restarts and the BSB communication through the UART was rock stable.

If you mean by "communication through the UART" that you connected the Olimex via USB (which is the only way to access the UART, unless you go directly to the RX0/TX0 pins via a USB-TTL-adapter), then you power the Olimex (also) via USB. And again, as described in the linked discussion above, that's when all the problems go away. That is also exactly the workaround solution that I tell people to do if they encounter the problem, but this means that they cannot use PoE anymore – on a regular POE because PoE and USB-power are not allowed at the same time, and on a POE-ISO because it doesn't make sense, of course, to power the device from both sides, even if it is possible to do so.

> So it seems to be only affecting some hardware variants

No, according to Dan Koloff in the linked threads, this affects all PoE-powered Olimex devices because of the hardware design. We cannot verify this anymore on your side since it seems that your Ethernet port has been destroyed. That's why, unfortunately, you won't be able to help us here, because in order to do the abovementioned tests, the Olimex has to be powered via PoE. Powering it via 5V USB will "solve" the problem immediately.

fredlcore, 2025-01-20:
> I'm afraid it's not that easy in my case: I gather room temperature of all rooms with my Timberwolf Server, calculate a mean value and send it via BSB-LAN to the boiler which heats accordingly. This [...]
That's why it's not so easy for me during winter to test this stuff. Sorry.

That's exactly my setup as well. But rest assured, if you set the comfort temperature setpoint to something like 21 degrees and then room influence (parameter 750) to zero, your heater will heat based on the outside temperature and the flow temperature and you won't notice any difference for some time. I frequently have to remove BSB-LAN to do some tests, and with these settings, the overall temperature hardly changes at all during the 30-60 minutes that I usually test - and this is about as long as the tests here will take as well.
Just make sure that you have the original software configuration ready so that you can flash it again once the tests are over. Then the whole tests will be seamless.

uschindler, 2025-01-20:
Hi,

> If you mean by "communication through the UART" that you connected the Olimex via USB (which is the only way to access the UART, unless you go directly to the RX0/TX0 pins via a USB-TTL-adapter), [...]

With UART I was referring to the serial chip that handles the BSB communication on your adapter. As you see in the picture above, initially my device was a OLIMEX PoE ISO with only Ethernet connection, powered with 52 Volts (according to switch) and approxy 2.3 Watts. No USB connector (except for debugging and initial software upload). In the first few weeks after installation, there was no reboots and no failures on the BSB communication. The adaptor worked perfectly fine. So it looks like there are still cases where the hardware works correctly. Maybe the higher voltage of 52 Volts on my Netgear PoE instead of standard 48 Volts lead to exactly 5V side had exactly 5 Volts and not 4.9 Volts like discussed in above threads, so I did not see the issues. This was of course not proofed.

The problems started later (possibly due to the overheating), but there was still no restarts. The only thing that happened for me was random short-time disconnects from Ether net which got more frequent over time till the whole Ethernet socket more or less died / got unuseable.

During the whole time, there was no USB power connected and BSB communication worked well. Of course (as I had no serial connection and the telnet connection was unstable) I cannot 100% say that there may not have been communication failures due to phantome bits on the BSB connection. But the system did not restart while powered via PoE during 4 weeks while I was on vacation in November, it just lost ethernet from time to time.

> And again, as described in the linked discussion above, that's when all the problems go away. That is also exactly the workaround solution that I tell people to do if they encounter the problem, but [...]

So when I buy a new PoE ISO, I will try to add the sleeps into the code. Unfortunately for me it is too late, you're right.

My observation with the damaged PoE due to overheating should maybe also be added as a warning. If you look on Amazon reviews and my linked forum threads, many people complain that the devices break after a few weeks with PoE only due to overheating.

> No, according to Dan Koloff in the linked threads, this affects all PoE-powered Olimex devices because of the hardware design. We cannot verify this anymore on your side since it seems that your [...]

See above. For me I had no obvious instability issues except disconnects on Ethernet. But as you can't see all errors without serial connection, I might have been affected by the issue, too. But I had no restarts. The uptime counter on the "settings" page was several weeks at beginning. Software version was around October 20th.

fredlcore, 2025-01-20:
> Maybe the higher voltage of 52 Volts on my Netgear PoE instead of standard 48 Volts lead to exactly 5 Volts and not 4.9 Volts like discussed in above threads, so I did not see the issues. 

Ok, interesting. Since there are quite a lot of BSB-LAN users with the POE-ISO, I would have also assumed more bug reports. I just thought that they wouldn't have realized the problem, because the reboots would happen more or less transparently or be mistaken with temporary loss of connections. But if you had proof via the uptime counter, then this could be an explanation.

CZvacko, 2025-01-23:
I use Olimex ESP32-PoE-ISO since 2021 and no issues with PoE power (plugged into D-link DGS-1100-08P). When I purchased it, I downloaded this schematic. It is revision G and matches the physical product.

[Schema ESP32-PoE-ISO_Rev_G1.pdf](https://github.com/user-attachments/files/18517889/Schema.ESP32-PoE-ISO_Rev_G1.pdf)

fredlcore, 2025-01-23:
What is the voltage level that your PoE provides? 48V or 52V?

CZvacko, 2025-01-23:
52,2V - measured at D10 diode.

uschindler, 2025-01-23:
> I use Olimex ESP32-PoE-ISO since 2021 and no issues with PoE power (plugged into D-link DGS-1100-08P). When I purchased it, I downloaded this schematic. It is revision G and matches the physical [...]

This one has a different setup of resistors for MPS (R41 and R42, which have 7.8 kOhm in total on two resistors, 0.3 Watts for 48 V) so it may heat up less and distributed over two resistors. It's good that it did not fail for you due to overheating:

[image]

Of course the voltage question is very relevant, because that's what may cause different behaviour. The Netgear switch I use provides voltage and wattage in the port's settings/status through the web interface of switch.

fredlcore, 2025-01-23:
It wouldn't have been the first time that Olimex has "verschlimmbesser" (made a good thing worse) in the design. Newer POE models also have a capacitor(?) right next to the UEXT connector which is taller than the connector itself, forcing you to "bend" any adapter slightly :(...

phlfr, 2025-02-12:
Hello, I am currently facing the same issue:
BSB-Lan installed with Olimex POE mi december, powered by Internet.
Was working fine, but recently I faced several issues: unstable access to the bsb-lan.local page, and very frequent reboot of the Olimex ( something like every 10 seconds...).

After reading the post, I removed the POE from the ethernet connexion and I powered the Olimex via the usb connector.

My system is up since 15 minutes.... great progress!

I will keep you informed.

Bests Regards!

Philippe.

fredlcore, 2025-02-12:
Also, if you know a little bit of programming, could you run the tests that I've mentioned in this post: https://github.com/fredlcore/BSB-LAN/issues/715#issuecomment-2600889668

phlfr, 2025-02-12:
Hello Frederik,  
my switch is a TL-SG105PE, I see nothing inside the documentation about the POE voltage. I will have to measure it ... after finding the relevant pins to check.
I will be very happy to code and run the test... I will do it in fews days.... the power supply of my computer with IDE environnement just went out of order... bad luck. when start to go bad... 

I will keep you informed.

Best regards,

Philippe

fredlcore, 2025-02-12:
Great, thanks a lot!

phlfr, 2025-02-12:
Hello Frederik. 
I did measure the POE voltage. I took the voltage at D10, my value is  51.3Volt.

I also give you a picture of my OLIMEX PCB ( bought 2024/11) around R41...

[image]

I will do other tests as soon as I get my dev computer back !

fredlcore, 2025-02-12:
Oh, that's interesting, I had hoped that the problem of the underpowered IC could be due to the 48V, but apparently, it's a combination of factors. The more important it will be to get the results of the tests above, so thank you for your kind offer to run them once you have your dev computer back!

fredlcore, 2025-02-13:
Thanks for backing this up, @DanKoloff. The problem is that most users in this project use the Olimex to control their heating, that's why some of them are reluctant to do it during the heating season - although my tests will probably just result in half an hour of interruption. But @phlfr has offered his help, and with that, I hope we can come to a conclusion soon...

phlfr, 2025-02-13:
Hello Frederik,

 

I found a work around to supply my dev computer…

 

 

I should be soon able to do the test …

I’ll keep you informed.

Best Regards.

 

Philippe

phlfr, 2025-02-13:
First feedback:
_Could you replace bsb.cpp in the src/BSB folder with the bsb.cpp that Dan Koloff attached to https://github.com/OLIMEX/ESP32-EVB/issues/57#issuecomment-2165214478?:

The Olimex is up since 20 minutes...
The Web interface is not completely  fluid. When switching from parameters( bsb-lan.local/C) to heating parameters ( bsb-lan.local/K) it tooks several minutes to display or need a refresh. only 4 categories are displayed ??) switch back to /C seems morev rapi, but not systematic.

phlfr, 2025-02-13:
Starting following tests....
rapid check of the température of the board, nothing very hot. R41 a little hot but not burning after about 20 minutes of POE supply...

1//OLIMEX with regular software, connected POE, BSB-LAN interface plugged on. web interface a little sticky, not really responsive. All heating parameters displayed. reboot after 4 minutes ( strange !!! few days ago it was rebooting after 10-20seconds...)
few days ago the board has been power through POE since one month... perhaps hot ?

2//OLIMEX with regular software, connected POE, BSB-LAN interface UN plugged. web interface a little sticky, not really responsive sometimes parametres are displayed instantly, sometimes takes minutes. only few parameters displayed.  reboot after 31 minutes -> lets's go to bed ... and continue tomorrow...

phlfr, 2025-02-14:
Hope it will help!

fredlcore, 2025-02-15:
Thanks a lot, @phlfr! May be I wasn't clear enough in my instructions or I didn't understand your test results correctly, but when the BSB-LAN adapter is UNplugged, how can you get parameters displayed? If at all, they should all come up with an error message, because how else would the Olimex be able to talk to the heater if the BSB-LAN adapter is unplugged?

Also, could you do the test that I listed under no. 1, i.e. replacing the bsb.cpp with the one from @DanKoloff in the linked comment? Based on his testings, the reboots should go away, even though, obviously, no connection to the heater would be possible. Again, my adapter needs to be unplugged for this.

The "stickyness" of the responses is, by the way, expected, because if you look at the debug output (serial or telnet), you'll see that BSB-LAN will try each parameter three times with several seconds of timeout, so calling a parameter category will take significantly more time with the adapter disconnected than as if it was connected.

What I also don't understand are the minutes/seconds mentioned after the time, for example, if you write:
> powered on 10h44
navigating
10h47 uptime 4'54''
navigating
10h55 uptime 3'36'' -> reboot

10:44 plus 4 minutes and 54 seconds (that's my reading of 4'54") don't add up to 10:47. Likewise, 10:55 is later than 10:47 plus 3'36". Can you explain more what that means?

I also just realized that I made a mistake in 2.3, it shouldn't be `Serial1.begin(4800, SERIAL_8O1, 1, tx_pin);` but rather `Serial1.begin(4800, SERIAL_8O1, 3, tx_pin);` - because pin 1 is the tx_pin already. Pin 3 would be the RX pin for `Serial`. Could you do this test again, please?

If using all these pins result in a reboot after some time, except for pin 1 (which is `Serial`'s TX pin reporposed as a RX pin for `Serial1` in this example, then there is no pin I can use to modify the hardware of BSB-LAN in order to avoid this bug. 
Once we have confirmaton for this (by checking that pin 3 will work with no reboots), I'll have to stop recommending powering the Olimex boards with via PoE or barrel connector, until Olimex is making the necessary hardware adjustments.

phlfr, 2025-02-15:
> Thanks a lot, [@phlfr](https://github.com/phlfr)! May be I wasn't clear enough in my instructions or I didn't understand your test results correctly, but when the BSB-LAN adapter is UNplugged, how can you get parameters displayed? If at all, they should all come up with an error message, because how else would the Olimex be able to talk to the heater if the BSB-LAN adapter is unplugged?
> 
> Also, could you do the test that I listed under no. 1, i.e. replacing the bsb.cpp with the one from [@DanKoloff](https://github.com/DanKoloff) in the linked comment? Based on his testings, the reboots should go away, even though, obviously, no connection to the heater would be possible. Again, my adapter needs to be unplugged for this.
>  I made this test with BSB Lan connected.. I will redo it. 
> The "stickyness" of the responses is, by the way, expected, because if you look at the debug output (serial or telnet), you'll see that BSB-LAN will try each parameter three times with several seconds of timeout, so calling a parameter category will take significantly more time with the adapter disconnected than as if it was connected.

Phl I would be glad to have a look at the serial output via telnet... how to proceed ? 

> What I also don't understand are the minutes/seconds mentioned after the time, for example, if you write:
>
>
> 10:44 plus 4 minutes and 54 seconds (that's my reading of 4'54") don't add up to 10:47. [...]

Phl: I report what is display on the parameters web page. The time for power up is approximative ( I go downstair in the heater room to plug, tyhen come back upstair to my computer..) I consider that if uptime decrease between 2 update of the parameters page, that means we had a reboot.

> I also just realized that I made a mistake in 2.3, it shouldn't be `Serial1.begin(4800, SERIAL_8O1, 1, tx_pin);` but rather `Serial1.begin(4800, SERIAL_8O1, 3, tx_pin);` - because pin 1 is the tx_pin already. Pin 3 would be the RX pin for `Serial`. Could you do this test again, please?

Phl: I made tests with: Serial1.begin(4800, SERIAL_8O1, Z, tx_pin);   and tried with 1,4,36,16,13,15,2 for Z

> If using all these pins result in a reboot after some time, except for pin 1 (which is `Serial`'s TX pin reporposed as a RX pin for `Serial1` in this example, then there is no pin I can use to [...]

Phl: I most of the cases I have to "stress" the system with clicking on the 2 tabs about ten time every second to obtain the reboot. 
I had case where the system has been up for 2 hours without reboot while I was away and not stressing.
 
> The only thing I don't understand is why changing from `Serial1` to `Serial` in 2.2 is still causing reboots for you. Based on [@DanKoloff](https://github.com/DanKoloff)'s test, `Serial` is not affected by this, so the reboots there could/should have other reasons...

Phl: we have to clarify is I have to stress the system by clicking on the web page, or just let it run and check if it reboots by itself or not after 2 hours for example...

The thing that I do not understand is that fews days ago, with the system powered on POE in his box since 3 weeks, the system was rebooting very often by himself. I have not get the same rate of reboot during all the tests. Is there also a link with the temperature of the board ?

uschindler, 2025-02-15:
> The thing that I do not understand is that fews days ago, with the system powered on POE in his box since 3 weeks, the system was rebooting very often by himself. I have not get the same rate of reboot during all the tests. Is there also a link with the temperature of the board ?

Hi, sure there is, see my comment above: https://github.com/fredlcore/BSB-LAN/issues/715#issuecomment-2600946720

May story is very similar to yours. The device worked fine in the first few weeks (October, November). Mid December I noticed many disconnects in the MQTT connection. Some of them were caused by some buggy code with pubsubclient's keep alive, but some of them were caused by Ethernet disconnects  (no Reboots). On the Switch, you were able to see:
- PoE indicator stays alive
- Ethernet Link indicator went offline for a second and then reconnected.

When looking at serial console there was NO REBOOT (also proven by the Uptime counter). Instead the serial console reported "Ethernet disconnected", followed by "Ethernet Connected" and DHCP discovery (same on server). So it looks from the outside like a reboot (because you see ethernet link flapping and DHCP requests), but actually the uptime counter stayed alive.

To me it looks like @phlfr has a similar problem like I have: When the device overheats due to the resistor in small enclusures it reboots or looses ethernet.

The serial connection issue with BSB bus did not happen on my device, this is why I said, the PoE voltage could be the reason.

So to me and as answer to @DanKoloff: There are major issues with PoE (and people on Amazon or other shops put this in their comments). If you have the device in an enclosure (like heating or small box) and have it powered by PoE it degrades after time and fails in several ways (depends on customer):
- ELKO on other side of board blows up
- unstable Ethernet connection
- completely borken Ethernet connection (that's my case)
- overheating to almost 90 °C
- sudden reboots

So actually the resistor R41 may be needed, but it has several problems:
- it is calculated to draw 10 mA, which is the minimum required by PoE spec. But I think that could be reduced to half - because the Olimex itsself also takes some power. Why to we need to burn ~0.7 Watts wth 52 Volts and overheat everything? The resistor as is would only be needed if the Olimex main chip and regulators takes no current at all! So how about something like 9.4 KOhm instead of 4.7 kOHm?
- Why would somebody would like to put the Olimex into Low Power mode and instead use a resistor to take more power than spared by the lower frequency on the chip. This is - sorry - bullshit. It looks like you have advertisement material for some industial customers and tell them "hey this device is able to do low power and it costs you nothing", but at same time add a useless resistor to draw 0.7 Watts!? This makes no seanse, really. Maybe put into your documentation: If you want to use PoE, don't use low power mode of the ESP32. This can be documented easily. My switch says: with PoE, normal ESP power and BSB-LAN board on to the whole device takes 2.7 Watts! So without the resistor it would still take 2 Watts which is way more than 10 mA.
- Maybe add a dynamic regulation istead of this resistor
- Maybe add a DIP (or similar) switch to disable the resistor. There should be enough space on board to do this. Maybe two pins where you can put a jumper on to would be fine.

@DanKoloff: Is it possible to get a replacement device for free? Mine's Ethernet is now fully broken and does not even work anymore with USB and no PoE! It disconnects every few hours for a second (on USB) and almost minutely when connected via PoE. My Elkos look still fine, but it looks like the device got major damage end December. I have it now connected via WLAN and Smasung USB charger, which works:

[image]

Uwe

fredlcore, 2025-02-15:
@uschindler: We discussed this above already (https://github.com/fredlcore/BSB-LAN/issues/715#issuecomment-2600955730), and I would kindly ask you again not to divert attention from the issue that I want to focus here. Please settle your replacement issues or anything else related to the resistor issue directly with Olimex and open up a ticket there. While the resistor issue may be one reason for the reboots, this is not the point of interest here because it's nothing that I can fix here (as opposed to maybe using different IO pins). Therefore, here, we want to deal with a reproducible issue that has been explained above and in the linked Olimex issue. Thanks.

fredlcore, 2025-02-15:
@phlfr: 
1. You can enable debug output via telnet in the web configuration, however Serial log would be more helpful as it show the reason for the reboot (SW_REBOOT / Stack smashing protect failure). If we get this reboot reason, we know it is caused by the serial bug.
2. Thanks for the clarification about the reboots. This means that there could be even more reboots than just the short uptimes that you mention (for example if the uptime between 10:47 and 10:55 is 3'36", there could be more reboots within this 8 minute timeframe.
3. Yes, these tests were helpful, but I forgot to mention pin 3 for Z. So could you do this one just for pin 3 again, please?
4. The reboots should happen even without activity, because, as mentioned above, all these tests should be done with the BSB-LAN adapter removed. This is because the reasons for the reboot we're investigating here is that due to the not completely unpowered CH340T USB-to-serial IC, irregular data is still received on pin 36 (the current RX pin). This irregular data has the potential to keep BSB-LAN in an endless loop and therefore lead to the crash. What you can do to monitor ist is to have a script calling `/JI` every minute or so, it's a JSON structure that contains the current uptime. 

To avoid mixing up reboots which may be due to the resistor issue and the one due to the irregular serial data issue (which is the only one we focus on here), please have your Olimex run outside the box so that the chances of it overheating are reduced.
The reboots due to the irregular data should nevertheless happen if your board is affected in the same way as @parsley's or @DanKoloff's. If these specific reboots do not occur with you, then your hardware revision is not affected by this bug - which is good for you, but unfortunately then you won't be able to help us narrowing down this issue here.

phlfr, 2025-02-15:
@fredlcore 

If I want to use debug on serial ouput, I need to be connected to my computer thru the USB connection, this is possible as I will not be connected to my heater. Do you prefer me to connect in Ethernet ou ethernet POE ?

Once you tell me your wish, I will do the test with pin 3 for Z. I will also redo the test with Dan Koloff code and the bsb.cpp from "serial = &Serial1;" to "serial = &Serial;"

Which test process would you like ?
 1  just let on line ... and see after 2 hours if a reboot occur
 2 sollicitate the BSB-LAN regularly ? How ? and see if a reboot occurs before 2 hours

fredlcore, 2025-02-15:
Since for these experiments you don't need to be connected to the heater, you can just take the Olimex, power it via PoE and then connect it via USB to your computer (of course provided that it's a POE-ISO, because with the regular POE, you cannot connect it to PoE and USB at the same time). 
The reboots will occur no matter what you do. Because the irregular data will be coming in from the not completely unpowered CH340T and thus lead to reboots after some time, because BSB-LAN will receive this irregular data and wait for a valid telegram which will never come and will then crash. So, bottom line, no stress testing necessary. Just monitor the serial output or query `/JI` regularly to see if there is a reboot...

phlfr, 2025-02-16:
I finally redo all the tests with an other approach.

To be noticed, with OLIMEX board with regular soft, connected to my heater, ethernet not POE and usb power:
It has reboot after a time between 12h00 to 15h00.

The OLIMEX is close to my computer, not in a BOX, with no BSB-LAN adapter card.
No specific stress action, I Just update periodicaly the webpage to have time and uptime.

1// with BSB.CPP from Kon Koloff
      Up during 2h22 without reboot
2// with Serial in place of Serial1
     Up during 3h29 without reboot
3// Rx=1
  Reboot after 3 minutes
4//Rx=4
  Up during 10h34 without reboot
5//Rx=36
  Up during 2h38 without reboot
6// Rx=16
  Reboot after 23 minutes, but stay up during 2h29' afterwards
7// Rx=13
  Reboot after 1h47 minutes
8// Rx=15
In progress...

I will update this post with next results.

fredlcore, 2025-02-17:
Ok, thanks, just to be clear: All these tests are PoE-powered, right?
It's strange that Rx=36 does not reboot because that is the standard value. Can you please do one more test with 
- PoE-powered
- Standard software, no changes
- Removed BSB-LAN adapter

That would be the same setup as @DanKoloff had. The reboots should come in irregular intervals, but noticably frequent. If that is not the case with you, then the causes for your reboots may be different from the issue of the CH340T, especially if the reboots also occur when using USB power...

DanKoloff, 2025-02-17:
@uschindler E-mail me at support@olimex.com with link to this thread so we can discuss further without disturbing the tests, but I don't think your device has hardware damage. Did you ever test it with some simple Ethernet code and not BSB-LAN code? Simplest Ethernet demo code is from the espressif package for Arduino IDE. To check if the Ethernet works, please test board one with the example from File -> Examples -> Ethernet -> ETH_LAN8720 and make sure to select Olimex ESP32-POE-ISO board from "Tools->Board".  Also disconnect everything and test just the ESP32-POE-ISO with PoE Ethernet and USB. See if it still hangs.

phlfr, 2025-02-17:
> Ok, thanks, just to be clear: All these tests are PoE-powered, right?

Yes test are with POE, no BOX, and I decided to unplug USB to be back in the "normal operation config", And I follow reboot thru the status page at BSB-LAN.local/C

 It's strange that Rx=36 does not reboot because that is the standard value. Can you please do one more test with
> 
> * PoE-powered
> * Standard software, no changes
> * Removed BSB-LAN adapter
> * 
I will do.... outside box to limit heat constrainst.

> That would be the same setup as [@DanKoloff](https://github.com/DanKoloff) had. The reboots should come in irregular intervals, but noticably frequent. If that is not the case with you, then the [...]

phlfr, 2025-02-17:
Would it be possible to log into the EEPROM the 10 last startup timestamps, and have them displayed in parameters, this would help tracking reboots...

fredlcore, 2025-02-17:
That would be quite an effort... Can't you just use WiFi an then poll `/JI` periodically?

phlfr, 2025-02-18:
new test:
new test as requested:
PoE-powered, bsb.cpp from DanKoLoff, Removed BSB-LAN adapter, out of box for cooling.
USB disconnected
18h25 powered up
19h33 uptime 1h08
21h01 uptime 12'45'' -> reboot
8h35 uptime 11h35 no reboot during the night...

I will have to stop other tests because I am leaving this house for some days, and will have to reconnect BSB-LAN to the heater in order to be able to manage my heater remotly.

I will connect it without POE, USB powered, out of box for cooling.
We will see how it works, despite I will not be able to monitor the uptime from outside... It could be an idea to add the uptime os the board as an MQQT parameters that can be polled by home assistant time to time.

fredlcore, 2025-02-18:
Thanks a lot again!
I have added some lines of code to protect the stack smashing, so the reboots should no longer occur with the most recent version on GitHub. However, the problem behind that is not solved yet because the irregular data generated by the not completely unpowered CH340T might still disturb the BSB data. Installing the new version would help identify whether the reboots originate from the CH340T problem or from something else.

phlfr, 2025-02-18:
Thanks a lot. I could have the time to update my BSB-LAn today before quitting. Where should I download the last version ?

fredlcore, 2025-02-18:
Just from the master repository at GitHub.

phlfr, 2025-02-18:
Hello frederick, 

I finally built a small config in my Configuration.yaml of Home assistant to monitor the uptime... it will ease the follow up even from outside ...
like that:

rest:
  - resource: http://192.168.1.212/JI
    scan_interval: 600 # update every  10 minutes
    sensor:
      - name: "BSB_UpTime"
        json_attributes_path: "$"
        json_attributes:
          - "name"
          - "version"
          - "hardware"
          - "freeram"
          - "uptime"
        value_template: "{{ value_json.uptime }}"
        device_class: "duration"

phlfr, 2025-02-24:
Hello frederick,

I get remotly conected to my Home assistant and I can see that my Olimex POE-ISO powered with USB and out of his box ( better cooling)  is up since 6 days and 3 hours. This is a breaking change!

I thing that my next test will be to power it through a separate POE splitter.

I will keep you informed !

Best regards,

Philippe.

fredlcore, 2025-02-24:
Thank you for confirming this. This means that at least a significant part of your reboots were due to the CH340T issue. 
That means that invalid data, at least intermittently, led to a buffer overrun and thus to a crash. No longer assuming that there will always be well-formed data (as has been the case for the last 10 years), was the "cure" now.

The important question for me would be now: Does this intermittent invalid data cause (noticeable) malfunction in BSB-LAN, or not? Since BSB-LAN retries twice to send a telegram in case it doesn't get a valid reply, it could be that you wouldn't even notice it except for a 1-2 second delay. In that case, I wouldn't bother with this problem and also wouldn't work on a new hardware revision of the BSB-LAN adapter.

But if "parameter not supported" or other errors show up from time to time when querying a parameter, then this would be an indication that the amount of invalid data is noticeable. In that case, it would be great if we could investigate this further.

I don't know if you are already regularly getting a lot of data from BSB-LAN into Home Assistant, but it would be great to see if there are any anomalies when you are querying maybe 10-15 parameters every minute and see if there are any "holes" in the datasets? If not, then everything is fine. But if there are, I'd like to investigate this further.

phlfr, 2025-02-24:
Hello frederick, I will be happy to continue tests when I will be back in the house where is the BSB. I should be there beginning of march. I will tell you !

phlfr, 2025-03-07:
Hello Frederick, I am back in my house... I have finally connected the BSB Lan with a POE splitter ( few Euros on aliexpress). I did it 8 days ago and BSB is running since without any reboot. This shows that there is really something when running OLIMEX with POE feature powering the board. Tell me if I can help with some extra diagnostics.

fredlcore, 2025-03-07:
Great, thanks a lot again, I'm glad that we could identify the issue. As a last test, could you do some kind of stress test where you poll values from BSB-LAN maybe every 10 seconds or so? It can be the same parameter and it doesn't matter which one it is. I just want to see that the irregular data that the hardware bug is generating is not (significantly) affecting the data access to the heater.
The easiest way would be to do an SD card log of one parameter that always stays the same, for example parameter 750. Then you could download the log file and send it to me, then I can double-check that every poll was successful. Thank you!

phlfr, 2025-03-09:
I have ordered a sd card ( I have plenty in my other house )... should be delivered monday. I imagine that I will have to modify the setup of BSBLAN ... asking to poll parameter 750 every 10 seconds to SD card rather than MQTT. How long should I let it run ?
I imagine I should I format the SD card ? which format ?

fredlcore, 2025-03-09:
You just need to insert a FAT32 formatted micro-SD card and then initialize it using the `/D0` command (this removes all previous logs and initializes a new one in case no log file has been created before). Then go to the web configuration and set the logging mode to SD-card, add, for example, parameter 750 to the list of log parameters and set the logging interval to 10. After saving and waiting for a minute, you can call `/D` and you should see the log file with six entries. If that works, you can leave it running for a few hours and then send me the log file. If there are no significant number of dropouts, we know that the workaround for the hardware issue is safe to use, at least for BSB-LAN.

phlfr, 2025-03-11:
Hello Frederik,

I received my 32 Gb micro SD card. 
Unfortunatly http://bsb-lan.local/D0 is not working, I have an error message: "
, Échec de la création du fichier datalog.txt !"

I tried to reformat the SD card with my computer, same  message...

should I change other parameters before issuing the /D0 command ?

bests Regards,

philippe.

phlfr, 2025-03-12:
Hello Frederik,

here it is !
the file seems fine !
tell us.

have a good day.
Bests regards,
Philippe

[datalog.txt](https://github.com/user-attachments/files/19203542/datalog.txt)

fredlcore, 2025-03-12:
Thanks a lot, this data is really helpful!
It shows that out of the over 3200 records, 110 are between 11 and 13 seconds apart, compared to the expected 10 seconds.
While I cannot rule out other reasons for this delay (such as using the web interface just in the moment when the 10 second interval has passed), it could be that these are the times when the corrupted data on the bus forces BSB-LAN to send the query request again after the one second internal timeout. The good thing is that in the end, the repeat queries all lead to the expected result, so the workaround is working stable both in terms of program stability as well as data validity.

@DanKoloff: If my assumptions are correct, then this bug affects approx. 3% (110 out of 3200) of the data on the bus (at least in phlfr's setup). Since I have now found a doable workaround for BSB-LAN, I can no longer investigate any potential hardware workarounds (such as using different input GPIOs other than GPIO36 for example). I think for data that can be verified (either through CRC or through awaiting an expected response), a software workaround like this is doable, but for other types of incoming data streams without any checksums etc., the additional corrupted data may not be so easy to deal with. So maybe this would be worthwhile for Olimex to investigate further and maybe fix this in a future hardware revision (or otherwise highlight this behavior somewhere where users can be made aware of).

Thank you all again, this was a big stumbling block for me!

[55 more replies]

closed 2025-10-15 by fredlcore