fredlcore/BSB-LAN #716: [BUG] 6700:  000000000000000000000000 - decoding error (Brötje BBK EVO 20/22i)

jstammi, 2025-01-24:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
2.3.4

**Version of the ESP32 or Arduino framework**  
esp32 by Espressif Systems 3.0.7

**BSB-LAN Version**  
4.2.59-20250109021219
Plus some patches (jstammi/hs-20250116) THAT DO NOT AFFECT THE DECODING, as I saw the decoding error already from the very beginning. But I considered it not that relevant as I expected it to be caused by the There-is-NO-Error data receiving. This changed by today having now monitored  the heating problem, that I am after, while BSB-LAN being connected.
It is still not critical to me(!), as the decoded content is well known (unfortuantely).

**Architecture**  
ESP32 NodeMCU

**Bus system**  
BSB

**Describe the bug**  
The log shows
```
...
LAN->HEIZ QUR 6700.0  - Fehlermeldung:
DC C2 00 0B 06 3D 05 00 9A 50 39
HEIZ->LAN ANS 6700.0  - Fehlermeldung:  VT_ENUM len !=2 && len != 3: 00 00 00 00 00 00 00 00 00 00 00 00
DC 80 42 17 07 05 3D 00 9A 00 00 00 00 00 00 00 00 00 00 00 00 B7 65
#6700:  000000000000000000000000 - decoding error
...
```

Causing it is not published to the MQTT (no "Publishing"/"Payload"/"Successfully published" messages, but immediately starting with decoding of next parameter). And finally then not updating in HomeAssistant.

With 6700 being about the latest error status. And I would expect the decodeing should be close to the one of #6803 being the 1st history entry for the errors. It's decoding shows as
```
...
LAN->HEIZ QUR 6803.0  - Fehlercode Historie Eintrag 1:
DC C2 00 0B 06 3D 05 08 14 A9 D6
HEIZ->LAN ANS 6803.0  - Fehlercode Historie Eintrag 1: 133 - 133:Sicherheitszeit für Flammenbild überschritt
DC 80 42 0E 07 05 3D 08 14 00 00 85 CD A6
#6803: 133 - 133:Sicherheitszeit für Flammenbild überschritt
Publishing to topic: XXX
Payload: 133 - 133:Sicherheitszeit für Flammenbild überschritt
Successfully published...
...
```

**To Reproduce**  
Steps to reproduce the behavior:
1. telnet to BSB-LAN
2. wait for decoding of the affected parameter

**Log files - Bug reports without log files will be closed**  
Serial logging is not possible as the esp32 is residing in a remote location. I can access only by vpn network.

**Expected behavior**  
The error status should be correctly decoded to `133 - 133:Sicherheitszeit für Flammenbild überschritt` (95% sure guessing)

**Screenshots**  
[image]

(the `Unbekannt` in the top should show an error message as history parameters below do)

**Desktop (if applicable, please complete the following information):**  
 - OS Windows 11
 - Browser Firefox
 - Browser version 134.0.1

**Smartphone (if applicable, please complete the following information):**  
n/a

**Additional context**  
n/a

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**

fredlcore, 2025-01-24:
There is no way of decoding this telegram because it doesn't fit any known data type. That's why the decoding error comes, but it is a cosmetic one, so no harm done. As far as I know, it also doesn't contain any information that isn't available elsewhere.

jstammi, 2025-01-24:
Einspruch wrt being a cosmetic one: this is the only location showing the actual error. Or which parameter do you suppose to contain the 133 errocode additionally? I did not see any.

Maybe it is pushed later to the history entries. But this is nothing that I would like to rely on(?)

jstammi, 2025-01-24:
I will try to capture the messages while being in error state the next time (this time my mother was too fast on re-starting the heating). Hopefully with some reverse engineering ...

fredlcore, 2025-01-25:
Depending on the model there are several places where this can be found. Some have it in 6705 (current error code), others have it in the first history entry (6800 onwards). I have two controllers, one works like the former, the other like the latter. The only benefit you would get from this parameter is to know that the actual issue has been resolved, but since this most likely does not happen without any kind of active user involvement, that is probably also rather redundant.

Eventually, it is just a cosmetical issue because you already can access this kind of information now via HTML parsing (which I use for almost a decade now in FHEM). I have tried my best to make sure that the structure remains constant so that regex won't have to be adjusted, and so far, they haven't in the nine years that I run this project.
In this case, you can simply call `/10020`and then use
`10020.*?:.*?(\d{4})`
to extract the hexadecimal representation of the error code (provided that this is always the error code).

jstammi, 2025-01-25:
My reason for using the BSB-LAN is the monitoring. As the heating runs to some error from time to time. I need to know the correct code (and maybe any additional code(s) provided).

Parameter 6705 shows some additional internal code (625), but not the (most probable) 133.

The history is updated only after having resetted the heatin and this starting again.

10020 shows the same decoding error.

For now, until understanding the data (structure?), I added a new data type `VT_BINARY` (not the best name, I will change this once again later) showing the raw bytes received hex encoded. And configured the parameters 6700 and 10020 as such.

This way the information is recorded at least (instead of the data getting lost before).

jstammi, 2025-01-27:
This morning I recorded the 12 bytes for the parameter 6700 (Fehlermeldung) as `0x008509002001000000000000`

Obviously the leading 2 bytes `0x0085` refer to the decimal error 133. The remaining ... I cannot identify their meaning (but maybe I get some knowledge on them on wednesday, with the Brötje expert being in-house for (hopefully) identifying the problem's cause).
The additionally known related codes for parameters 6705 and 6706 were 625 = `0x0271` and 20 = `0x0014` - not included there.

For now I will add a 12 bytes decoding only considering the leading 2 as the relevant ones to be used for the VT_ENUM translation.

fredlcore, 2025-01-27:
Just take note that any future update will overwrite your changes, so I guess the HTML-parsing will be the easier way in the long run.
But let me know if you figure out something about the remaining bytes. I created an error by setting two bus masters in my LPB network and got the expectet error 100 (0x64) followed by 030103, but sometimes just 030000. The 03 (or 09 in your case) also appears in another error telegram as part of the payload, so probably at least this one has some additional meaning...

jstammi, 2025-01-27:
I forked your repo. And merge your changes into mine to a dedicated `upstream` branch and from there only to my own `develop` conatining my changes (actually until 4.2.59, the one afterwards pending - I had to "fix" this decoding problem first. And identifying the minimal change to achieve this ... took me some/too much time).

There are some more issues, minor ones, that I have already a workaround for. That I will report, too, with being able to spend more time there.

For the bytes: just noticed that multiple history entries are created for one error. With some additional internal codes providing. But none of them matching the bit pattern.

fredlcore, 2025-01-27:
That's what I meant. Too much time for something that can be achieved otherwise in a less obstructive way. But if it works for you, then that's of course fine. I just wanted to indicate (especially to potential other users who might be interested in this) that there are other ways of acquiring the data without changes in the code.

jstammi, 2025-01-27:
I guess(!) doing web  request polling and parsing the HTML output is in my usecase not the most suitable way to go: I actually monitor >70 parameters. AFAIU the code, each/most(?) are pulled from the controler. To my observation from the update rate this takes for my watch list ~14s already.
Setting for updates was 10s, it is now 15s, and ideally would be <5s - at least for some parameters, as I miss some states on heater ignition, that I would be highly interested in. Not to mention some timing, that I would like to see (burning stopped because of controler closed the valve or vice versa), that would require even higher acquisition rates - but this is not what the BSB protocol is built for.

And asynchronically pulling by REST/parsing HTML outputs ... is for sure less efficient to be handled as this breaks the cyclic loop (without having really checked the code for this).

And as I am a programmer as such it feels much better to me to adopt the code (as long as I understand it's intention) than hacking a workaround ;-)

[image]

[image]

fredlcore, 2025-03-11:
Since a few users have asked me why I talk about parameter 10020 and not 6700: On my system, I don't have parameter 6700, but the same command ID is used for parameter 10020 instead.

In the most recent version of BSB-LAN, I have now added some general definitions for `VT_CUSTOM_BYTE` configurations, so that you can access the error number now by changing the entry in `BSB_LAN_custom_defs.h` from
```
{0x053D009A,  VT_ENUM,            10020, S10020_0, sizeof(E10020_0),   E10020_0,     FL_RONLY,     90, 239},
```
to
```
{0x053D009A,  VT_CUSTOM_BYTE,     10020, S10020_0, sizeof(E_00_02),    E_00_02,      FL_RONLY,     90, 239},
```  

Of course, in your case, with parameter 6700 instead of 10020, it would be from

```
{0x053D009A,  VT_ENUM,             6700, S6700_0,  sizeof(E6700_0),    E6700_0,      FL_RONLY,     90, 239},
```
to
```
{0x053D009A,  VT_CUSTOM_BYTE,      6700, S6700_0,  sizeof(E_00_02),    E_00_02,      FL_RONLY,     90, 239},
```

Should more information from this parameter become verified, one can add additional lines with new parameter numbers and different `VT_CUSTOM_BYTE` definitions to extract this data as well.

fredlcore, 2025-03-11:
It would, of course, nevertheless be helpful if some people would still monitor the data payload as a whole and figure out what the other data segments mean so that these could be made use of as well...