fredlcore/BSB-LAN #719: [FEATURE REQUEST] Sending a setpoint to a connected MAX! wall thermostat (MAXCUL)

erichsch, 2025-01-26:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
Sending a setpoint to a connected MAX! wall thermostat (MAXCUL)

**Describe alternatives you've considered**

**Describe your own contribution**

**Additional context**

Kind regards Erich

fredlcore, 2025-01-27:
Generally, this would be interesting, but I don't know the structure of the MAX! telegrams that are needed to cause such a change in setpoint temperature. Can you run a serial monitor log in BSB-LAN and identify the MAX! telegrams that are caught when changing the setpoint on the thermostat? I'd need 2-3 when changing the temperature on the wall thermostat and 2-3 when changing from a heater's thermostat (I hope you have those, too).
I assume, the telegrams will be (slightly) different, so the difficulty/impossibility may lie in the fact that changing the setpoint needs knowledge about what kind of MAX device there is. Because currently, we are just catching the measured temperatures, and these have the same structure, whether it's wall or heater thermostat. If setting the temperature requires knowledge about what kind of device to address, then this suggestion may not be possible. But I'll be able to figure this out once you have provided me with these telegrams.

erichsch, 2025-01-27:
Hallo Frederik
Danke für deine schnelle Antwort

Ich habe leider keine Heizkörperthermostate nur die Wandthermostate.

Ich habe mal über Serial am MAXCUBE eine kleine Auswertung gefahren.
Vielleicht hilft dir das auch weiter.
Beim Thermostat 094F8B (KEQ0505347)

08:29:38 [ttyACM0] --> Z0B9C0540094F8B12345600732C
Änderung des Sollwertes von 25 Grad auf 25,5 Grad
08:29:44 [ttyACM0] --> Z0F9D0503094F8B12345600191B081D6828
08:29:49 [ttyACM0] --> Z0F9E0503094F8B12345600191B081D6D26
08:29:54 [ttyACM0] --> Z0F9F0503094F8B12345600191B081D7226
08:30:21 [ttyACM0] --> Z0BA00002123456094F8B000000
08:30:21 [ttyACM0] --> Z0BA00540094F8B123456007227
Änderung des Sollwertes von 25,5 auf 25 Grad
08:30:22 [ttyACM0] --> Z0FA10470094F8B0000000019043200E827
08:30:27 [ttyACM0] --> Z0FA20503094F8B12345600191B081E5728
08:30:32 [ttyACM0] --> Z0FA30503094F8B12345600191B081E5C27
08:30:37 [ttyACM0] --> Z0FA40503094F8B12345600191B081E6127

Im BSB-LAN log sehen die Telegramme genauso aus, welche empfangen werden.
BSB-LAN Log geht bei mir örtlich nur über Telnet das läuft natürlich immer
weiter, solange das Fenster offen ist.
Wie kann ich das Telnet Fenster stoppen, wenn die Daten vorbeilaufen? Ich
kann beim laufenden Log die Daten nicht kopieren.
Gibt es da einen Trick? Escape Code?

Liebe Grüße Erich

fredlcore, 2025-01-27:
Prima, danke! Das ist im Grunde das, was ich brauche, aber laut meinen Unterlagen müsste das Telegramm, was an achter Stelle eine 40 hat, die Temperatur senden, also `Z0BA00540094F8B123456007227`, basierend auf dieser Liste aus dem FHEM-Modul:
```
my %msgId2Cmd = (
                 "00" => "PairPing",
                 "01" => "PairPong",
                 "02" => "Ack",
                 "03" => "TimeInformation",

                 "10" => "ConfigWeekProfile",
                 "11" => "ConfigTemperatures", #like eco/comfort etc
                 "12" => "ConfigValve",

                 "20" => "AddLinkPartner",
                 "21" => "RemoveLinkPartner",
                 "22" => "SetGroupId",
                 "23" => "RemoveGroupId",

                 "30" => "ShutterContactState",

                 "40" => "SetTemperature", # to thermostat
                 "42" => "WallThermostatControl", # by WallMountedThermostat
                 # Sending this without payload to thermostat sets desiredTempee
rature to the comfort/eco temperature
                 # We don't use it, we just do SetTemperature
                 "43" => "SetComfortTemperature",
                 "44" => "SetEcoTemperature",

                 "50" => "PushButtonState",

                 "60" => "ThermostatState", # by HeatingThermostat

                 "70" => "WallThermostatState",

                 "82" => "SetDisplayActualTemperature",

                 "F1" => "WakeUp",
                 "F0" => "Reset",
               );
```

Allerdings finde ich in diesem 40er-Telegramm nichts, was sinnvollerweise nach Deinen Temperaturen aussehen könnte. Die Payload in diesem Telegramm ist `007227`, da sehe ich nichts, was nach 25 oder 25,5 aussieht, egal, auch wenn man die Temperatur mit zwei oder 10 multipliziert, was sonst immer eine einfache Art ist, um die Nachkommazahl zu kodieren...

erichsch, 2025-01-27:
Das klingt nicht nach einem Erfolg. Schade.
Ich habe noch etwas gefunden, wie man als Programmierer vielleicht
herausfinden kann, wie daten an den Wandthermostat etc. gesendet werden
können.
Dass kann ich allerdings als Nichtprogrammierer nicht so wirklich
entziffern.

Falls es dich interessiert:
https://github.com/ioBroker/ioBroker.maxcul
main.js
und im lib verzeichnis
culpacket.js
maxcul.js

kannst du aus dem etwas herauslesen?

Liebe Grüße Erich

erichsch, 2025-01-27:
> Ich kann mich jetzt leider nicht durch mehrere tausend Zeilen JavaScript
> code durchforsten, um da dann vielleicht etwas Relevantes zu finden. Ich
> sehe aber auch, dass ich da einen Denkfehler hatte: Beim Ändern der
> Solltemperatur am Wandthermostat kriegen wir ja nicht das Telegramm, was
> dem Wandthermostat eine neue Temperatur übermittelt, sondern höchstens das,
> wo der Wandthermostat dem HK-Thermosat die neue Temperatur übermittelt. Man
> müsste also mit einem bestehenden System (FHEM, ioBroker, etc.) schauen,
> was diese Systeme als Telegramm senden, wenn man dort eine neue
> Soll-Temperatur einstellt. Kannst Du das mal probieren?

Das war auch nicht meine Absicht, dass du hier Tage mit dem Script
verbringst.
Ich dachte, wenn du es überfliegst, siehst den fehlenden Hinweis.

Auf 26 Grad Sollwert

14:00:59 [ttyACM0] --> Z0F320503094F8B12345600191B0E007721
14:01:00 [ttyACM0] --> Z0E010202094F8B123456000119043420
14:01:04 [ttyACM0] --> Z0F330503094F8B12345600191B0E01421F
14:01:05 [ttyACM0] --> Z0E010202094F8B123456000119043420

Auf 25 Grad Sollwert
14:01:09 [ttyACM0] --> Z0F340503094F8B12345600191B0E014721
14:01:10 [ttyACM0] --> Z0E010202094F8B123456000119043422
14:01:15 [ttyACM0] --> Z0C270442094F8B0000000034EF20
14:01:19 [ttyACM0] --> Z0E010202094F8B123456000119043221

Kannst du mit dem was anfangen

Liebe Grüße Erich

erichsch, 2025-01-27:
Finger war beim Senden zu schnell.

Das 2. Thermostat
Sollwert auf 25
14:03:32 [ttyACM0] --> Z0FF70470078A1F000000001904330102FF
14:04:15 [ttyACM0] --> Z0C430442078A1F00000000B30003
14:05:50 [ttyACM0] --> Z0E010202078A1F12345600011904320D

Liebe Grüße Erich

fredlcore, 2025-01-31:
Es tut mir Leid, ich habe zwar in dem ioBroker Code wichtige Hinweise gefunden, aber mir scheint es, als wäre ein "einfaches" Setzen der Soll-Temperatur nicht möglich, da irgendwie immer auch der Betriebsmodus ("Auto", "Manual" etc.) mitgesendet werden muss(?). Den dann immer richtig abzugreifen und mitsenden zu können, wäre ein eher unvertretbarer Aufwand...

erichsch, 2025-01-31:
Hallo Frederik!
Danke für deine Bemühungen.

Ein verstellen des Sollwertes und mitsenden des Modus: "Manuel" (immer bei
Sollwertverstellung seitens BSB-LAN) ist nicht möglich?
Dann würde man sich die Abfrage ersparen.
Vielleicht im BSB-LAN eine fixe manuelle Vorgabe (Auto/Manuell) zum
Auswählen.
Je nachdem was man immer senden will.
Bei mir stehen die Thermostate z.B. immer auf Manuell.

Könnte so etwas einfacher funktionieren?

Liebe Grüße Erich

fredlcore, 2025-02-01:
Naja, nur weil es bei Dir immer auf Manuell steht, muss das ja nicht bei allen der Fall sein. Ich selber habe auch eine Mischung aus Auto und Manuell bei meinen MAX! Geräten. Insofern ist da auch eine Dauerhafte Einstellung nicht wirklich zielführend. Vor allem, wenn die Leute denken, dass es für "ihre" Einstellungen passen müsste und dann die Rückfragen bei mir auflaufen.

Wenn ich es schaffe, einen Telegramm-String zu erzeugen, dann wäre ggf. eine Option das über individuellen Code und die `_custom.h` Dateien zu lösen. Da könnte dann jeder selber im Code eintragen, was er für einen Modus verwendet etc.

erichsch, 2025-02-01:
Vielleicht habe ich mich schlecht ausgedrückt.
Datenfeld Sollwert > Daneben ein Ankreuzen von Auto/Manuell > Senden
Wäre das eine Lösung, oder ist das Programmtechnisch nicht leicht möglich?

fredlcore, 2025-02-01:
Nein, das ganze muss sich ja in die reguläre Prozedur des Setzen eines Wertes einfügen. Da kann man nicht so eben was um die MAX-Parameter drumherum stricken.

erichsch, 2025-02-01:
So stellt sich der kleine Erich eben die Programmierung vor. ;-)
Wenn dir etwas dazu einfällt, wie man das verwirklichen kann, wäre
meinerseits wieder eine größere Kaffeespende fällig.
Ich ahne, was du alleine mit dem Feature Request an Arbeit hattest.
Danke nochmals für deine Bemühungen.

Liebe Grüße Erich

fredlcore, 2025-02-01:
Ich habe gesehen, dass ich bei den Telegrammen einen Fehler gemacht habe, aber irgendwie reagieren meine MAXe gerade nicht auf die FHEM-Befehle.
Könntest Du mal bitte folgende Tests machen, indem Du Dich per Telnet mit dem (MAX-)CUL verbindest und folgende Telegramme per copy/paste absetzt (mit Return absenden):
`Zs0b770040123456094F8B0067` - das müsste den oben genannten Thermostaten (KEQ0505347) auf 19,5 Grad und "Manual" setzen.
`Zs0b770040123456094F8B0028` - auf 20 Grad und "Auto" setzen.

Ich bin gespannt...

erichsch, 2025-02-01:
Hallo frederik,

Ja, probiere ich gerne aus, sobald ich zu Hause bin.
Falls es heute zu spät wird, mache ich es morgen früh.
Ich melde mich.
Liebe Grüße Erich

erichsch, 2025-02-01:
Hab mich gerade erinnert, dass ich noch eine Homematic Anbindung mit einem
cUX-Daemon habe.
Hab mich mit VPN mit dem Handy verbunden und die Codes gesendet.
Lt Rückmeldung im IoBroker wurde genau das getan, was geplant war.
Du bist ein Genie.
Anbei die Sende und Empfangsprotokolle vom cUX

20 Grad Auto
19:21:44 [ttyACM0] <-T Zs0b770040123456094F8B0028
19:21:45 [ttyACM0] --> Z0E770202094F8B123456000118042821
19:21:54 [ttyACM0] --> Z0C680442094F8B0000000028F925

19:22:11 [ttyACM0] --> Z0B8B0002123456094F8B000000
19:22:11 [ttyACM0] --> Z0B8B0540094F8B12345600281D
19:22:17 [ttyACM0] --> Z0F8C0503094F8B12345600190113168F1F
19:22:22 [ttyACM0] --> Z0F8D0503094F8B12345600190113169422
19:22:27 [ttyACM0] --> Z0F8E0503094F8B12345600190113169920

19,5 Grad Manuell
19:22:38 [ttyACM0] <-T Zs0b770040123456094F8B0067
19:22:39 [ttyACM0] --> Z0E770202094F8B123456000119042720
19:23:10 [ttyACM0] --> Z0B8F0002123456094F8B000000
19:23:10 [ttyACM0] --> Z0B8F0540094F8B12345600721C
19:23:16 [ttyACM0] --> Z0F900503094F8B12345600190113178E20
19:23:21 [ttyACM0] --> Z0F910503094F8B12345600190113179321

Liebe Grüße Erich

fredlcore, 2025-02-01:
Super, danke für's schnelle Testen - und das jeweils nachfolgende Telegramm `Z0E770202` mit Typ `02` als viertem Byte ist das Bestätigungstelegramm (was bei mir seltsamerweise nicht kommt, aber das kann an den ausgelaugten Akkus liegen).
Dann überlege ich mir mal was, wie man das sinnvoll umsetzen kann...

erichsch, 2025-02-01:
Gerne, wenn du wieder was zum Testen hast, melde dich.

Liebe Grüße Erich

fredlcore, 2025-02-01:
Nur noch mal für mich zum Verständnis: Was für eine Temperatur setzt man denn, wenn im Beispiel oben die "Auto"-Temperatur gesetzt wird? Bei Auto gibt es doch die Eco und die Komforttemperatur, zwischen denen nach Zeitprogramm gewechselt wird, oder? Hängt es dann davon ab, wann man den Befehl oben setzt, ob dann die Eco- oder Komfort-Temperatur verändert wird?

erichsch, 2025-02-01:
Da ich nie mit Auto gearbeitet habe meine Einschätzung.
Gesendeter Sollwert im Auto Betrieb wird zu den programmierten Zeiten wieder von eco und comfort temp überschrieben.
Alles andere wäre seltsam.
Ist also eine temporäre sollwertänderung.

Klingt das plausibel?

fredlcore, 2025-02-01:
...leider nein, denn für die temporäre Änderung (genauso wie den Boost) gibt es noch mal eigene Werte, die man senden könnte. Hast Du die Möglichkeit, das mal direkt am Gerät zu überprüfen? Muss nicht heute sein...

erichsch, 2025-02-01:
Ja, gerne,

du musst mir nur sagen, welche logik, oder was ich kontrollieren soll.
Ich habe allerdings keine Zeitschaltprogramme aktiv, welche von eco auf
comfort umschalten würden.
Zumindest ist mir nichts bewusst. Habe da nie etwas eingepflegt.

Am Gerät kann ich erst morgen testen.
Ich sende die Befehle nochmals und sehe morgen am Gerät nach.
Ist es das, was ich zuerst testen soll?

fredlcore, 2025-02-02:
Danke, ich glaube, ich habe eine Lösung gefunden, wie das mit dem Auto/Manual etc. über einen kleinen Umweg doch im Webinterface realisierbar ist, so dass dieser Test erstmal nicht mehr nötig ist.
Was dagegen noch mal wichtig wäre zu testen, ist, ob die "Absenderadresse" vom Wandthermostat geprüft wird.
Könntest Du von daher noch mal diese Telegramme per Telnet absetzen?
`Zs0b770040000000094F8B0067`
`Zs0b770040000000094F8B0028`
`Zs0b770040111111094F8B0067`
`Zs0b770040111111094F8B0028`
Die sollten jeweils den Thermostaten auf 19,5 Grad und "Manual" bzw. 20 Grad und "Auto" stellen.
Wenn alles funktioniert, prima, wenn nur jeweils eine Variante davon, auch gut, nur wenn beides nicht geht, muss ich eine eigene Einstellung für die Adresse des CUL vorsehen.

erichsch, 2025-02-02:
guten Morgen Frederik!

die codes im letzten Mail bewirken am CUL (Seriell) nichts. Keine Quittung
:-(
Ergo wird die "Absenderadresse" geprüft, so wie ich das verstehe.
Ich habe den code vom vorigen Mail nochmals zu hause probiert (serial CUx).
der funktioniert. ablesbar am wandthermostat. OK.

Liebe Grüße Erich

Anbei der Log vom CUL:

07:01:51 [ttyACM0] <-T Zs0b770040000000094F8B0067 (keine Quittung)
07:03:17 [ttyACM0] <-T Zs0b770040000000094F8B0028 (keine Quittung)
07:03:32 [ttyACM0] <-T Zs0b770040111111094F8B0067 (keine Quittung)
07:04:08 [ttyACM0] <-T Zs0b770040111111094F8B0028 (keine Quittung)
07:05:45 [ttyACM0] --> Z0BB00002123456094F8B000000
07:05:45 [ttyACM0] --> Z0BB00540094F8B12345600721F
07:05:51 [ttyACM0] --> Z0FB10503094F8B1234560019020705B123
07:05:56 [ttyACM0] --> Z0FB20503094F8B1234560019020705B61F
07:06:01 [ttyACM0] --> Z0FB30503094F8B1234560019020705BB22
07:06:32 [ttyACM0] --> Z0C7A0442078A1F0000000032E70B
07:06:54 [ttyACM0] --> Z0C5E0442094F8B0000000032E321
07:07:04 [ttyACM0] <-T Zs0b770040123456094F8B0067 (Code vom vorigen Mail)
07:07:05 [ttyACM0] --> Z0E770202094F8B123456000119042722 (Quittung vom Cul)
07:07:48 [ttyACM0] --> Z0BB50002123456094F8B000000
07:07:48 [ttyACM0] --> Z0BB50540094F8B123456007220
07:07:54 [ttyACM0] --> Z0FB60503094F8B1234560019020707B420
07:07:59 [ttyACM0] --> Z0FB70503094F8B1234560019020707B920
07:08:04 [ttyACM0] --> Z0FB80503094F8B12345600190207088221
07:09:39 [ttyACM0] --> Z0C5F0442094F8B0000000032E422

fredlcore, 2025-02-02:
Prima, danke! Das ist auch kein großes Problem, nur dass ich halt die Adresse des CUL (auch wenn die in der Standardeinstellung 123456 ist) auch irgendwo hinterlegen muss.
Falls Du Dich irgendwie erinnerst, wo/wie man die damals beim Einrichten des CUL/culfw hinterlegt hat, dann kann ich da noch mal genauer nachlesen, sonst klemme ich mich selbst noch mal hinter Google...

erichsch, 2025-02-02:
Ich könnte mich nicht erinnern, dass ich die Adresse angeben hätte können.
Die muss in der BIN von der Firmware enthalten sein.
Hab auch die GIT (source) von der aculfw durchsucht, hätte nichts gefunden.

Ich denke, jeder der Flasht bekommt die 123456. Wenn ich allerdings 2 Cubes
hätte wäre das vielleicht spannend.

Liebe Grüße Erich

fredlcore, 2025-02-02:
Ja, ich finde die 123456 auch weder im SourceCode der culfw noch in dem ioBroker-Modul, was Du oben verlinkt hast. Ich habe damals aber nicht die fertige HEX-Datei genommen, sondern wegen ein paar Anpassungen ;) selber kompiliert. Aber wie gesagt, ich habe da nichts gefunden. 
Auf jeden Fall scheint es nicht zwingend 123456 zu sein. Denn auch auf [dieser Seite](https://github.com/Bouni/max-cube-protocol/blob/master/S-Message.md) in der das Protokoll aufgeschlüsselt wird, ist ein SetTemperature-Beispiel genannt, wo der Absender auf 000000 steht.

Ich habe damals die 123456 beim Anlegen des MAX_CUL Devices in FHEM angegeben, aber auch nur, weil das in der Migrationsanleitung so angegeben ist. In der CommandRef steht dazu aber nur:
> Define

    define <name> CUL_MAX <addr>

    Defines an CUL_MAX device of type <type> and rf address <addr>. The rf address must not be in use by any other MAX device. 

Ich hätte vermutlich also auch 654321 nehmen können. 000000 ist anscheinend die Broadcast-Adresse, so dass die vielleicht als Absender nicht funktioniert, aber wie gesagt, beim Link oben ist so ein Telegramm enthalten. Wenn es aber weder mit 000000 noch mit 111111 klappt, ist es natürlich schwierig. 

Kannst Du Dich noch erinnern, wie Du Deine Thermostate mit dem MAXCUL gepaired hast? Bei FHEM muss ich das Pairing in FHEM aktivieren. Da kann es natürlich sein, dass die 123456, die ich in FHEM angegeben habe, dann an die Thermostate beim Pairing übermittelt wird. Wenn Du das bei Dir in Deiner Software ähnlich gemacht hast, wäre das vielleicht eine Erklärung...

erichsch, 2025-02-02:
Ich habe das pairing damals mit iobroker gemacht.
Hab gerade nachgesehen. In der main.js ist folgendes codeschnipsel für das
Pairing enthalten:

if (adapter.config.serialport && adapter.config.serialport !== 'DEBUG') {
        max.connect();
    } else if (adapter.config.serialport === 'DEBUG') {
        setTimeout(() => {
            max.emit('PairDevice', {
                src: '160bd0',
                type: 1,
                raw: 'Z17000400160BD0123456001001A04E455130363731393837'
            });
        }, 100);

Auch hier ist wieder die 123456 enthalten.

fredlcore, 2025-02-02:
Das ist interessant, main.js hatte ich mir bisher noch nicht angeschaut. Die 123456 kommt hier als Ziel vor, die Absenderadresse ist 160BD0, was ebenso fest eincodiert zu sein scheint. In dem Pairing-String enthalten ist die Seriennummer `NEQ0671987`. Keine Ahnung, warum, und ob die echt ist (bisher kenne ich nur JEQ und KEQ als Kürzel vor der Seriennummer).

Was ich noch gefunden habe, ist, dass man bei ioBroker wohl in der Admin-Einstellungs-Seite eine "BaseAddress" angeben kann, kannst Du mal schauen, was da bei Dir angegeben ist?
https://github.com/ioBroker/ioBroker.maxcul/blob/d176791802e3234309e2698f93f92e20e295adcd/admin/index.html#L155

Aber wenn ich mir den Code hier so zum Pairing ansehe:
https://github.com/ioBroker/ioBroker.maxcul/blob/d176791802e3234309e2698f93f92e20e295adcd/lib/maxcul.js#L689
Dann scheint es so zu sein, dass ein zu pairendes Gerät nur bei einem Reset als Zieladresse die Broadcast-Adresse 000000 verwendet. Im Falle eines Re-Pairings (z.B. nach Batteriewechsel) wird das Pairing an die zuvor gepairte Adresse gesendet, von daher wäre es nachvollziehbar, wenn so ein Gerät dann auch nur Telegramme von dem gepairten Gerät annimmt, was ja schon alleine aus Sicherheitsgründen Sinn macht.

Ich würde von daher vermuten, dass die 123456 bei Dir in den ioBroker-Einstellungen hinterlegt sind (auch wenn mich dann die fest codierten Einträge wundern würden. So oder so muss ich dann davon ausgehen, dass diese auch abweichen kann und sie von daher beim Senden von Telegrammen berücksichtigen.

erichsch, 2025-02-02:
Ja, du hast recht. Die default Basisadresse im Iobroker-Adapter ist 123456.
Ist mir noch nie aufgefallen.
Kann aber auch sein, weil sie in der Firmware immer enthalten ist.
Ich habe die Basisadresse im Adapter natürlich nicht geändert. Möglich
gewesen wäre es allerdings schon.
NEQ könnte ein Fensterkontakt sein. Die gab es ja auch.

fredlcore, 2025-02-02:
Stimmt, NEQ könnte ein Fensterkontakt sein. Aber gut zu wissen, dass die 123456 im ioBroker-Adapter enthalten sind. Vielleicht hat irgendjemand mal mit 123456 angefangen und dann haben alle anderen das in ihre Integrationen übernommen. Aber laut Code ist es so, dass der ioBroker-Adapter auf eine Broadcast-Anfrage (also die nach dem Werksreset) eines Gerätes antwortet und da dann die Basisadresse als Antwort hinschickt, so dass dann das Gerät weiß, von wem es Anfragen annehmen muss.

Damit ist es klar, dass ich diesen Wert als Einstellung berücksichtigen muss, dann kann ich mich jetzt an die Arbeit machen...

erichsch, 2025-02-03:
schaut fast danach aus, dass die 123456 blind übernommen wurden.
d.h. du hast die Lösung?
Das wäre toll, wenn das funktionieren würde.

fredlcore, 2025-02-03:
So, bitte jetzt einmal die aktuellste Version von BSB-LAN installieren und testen. Wenn einer der MAX-Soll-Temperatur-Parameter (205XX.2) gesetzt wird, wird auch ein entsprechendes Telegramm an den CUL geschickt. Ich hoffe, das klappt dann so, wie oben, bitte gerne noch mal mitloggen, was da am CUL passiert.

Die Konfiguration ist jetzt erst einmal fest einprogrammiert:
```
// MAX variables (can be overwritten in BSB_LAN_custom_global.h if necessary)
uint32_t max_cul_rf_addr = 0x123456;  // RF address of MAX_CUL to which devices are paired to
uint8_t max_group_id = 0x00;         // Group ID ("room number") - set to 0x00 when addressing individual devices
uint8_t max_flag = 0x00;             // Flag - set to 0x04 when addressing a group ("room"), set to 0x00 for addressing individual devices. 
uint8_t max_temp_mode = 0x01;        // Temperature mode: 0x00 - auto, 0x01 - manual, 0x02 = vacation (not supported here), 0x03 = boost (not supported here)

```
Man kann diese Werte in der `BSB_LAN_custom_global.h` überschreiben, wenn man z.B. den Auto-Modus anstatt Manual-Modus verwenden möchte (dann einfach ein `max_temp_mode = 0` dort hinzufügen. Das gilt dann aber in der jetzigen Form für alle MAX-Devices gleichermaßen. Ein Setzen des einen Gerätes im Auto- und des anderen im Manual-Modus ist also nicht möglich.

erichsch, 2025-02-03:
> Die Konfiguration ist jetzt erst einmal fest in der BSB_LAN.ino

Hallo Frederik!

Ich habe die neue version kompiliert, aber ich bekomme folgende
fehlermeldungen:
Störe diese, oder kann ich da etwas machen?
Liebe Grüße Erich

In file included from U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_defs.h:1144,
                 from U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino:168:
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:16: error: stray
'\' in program
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |                ^
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:26: error: stray
'\' in program
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |                          ^
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:55: error: stray
'\' in program
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |                                                       ^
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:70: error: stray
'\' in program
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |
 ^
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:78: error: stray
'\' in program
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |
         ^
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:15: error: found
':' in nested-name-specifier, expected '::'
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |               ^
      |               ::
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:12: error: 'S8u'
has not been declared
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |            ^~~
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:4955:27: error:
expected initializer before 'BSB'
 4955 | const char S8u:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino008_0[] = S8008_0_TEXT;  //
Status Feststoffkessel
      |                           ^~~
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:5970:42: error:
'S8008_0' was not declared in this scope; did you mean 'S8108_0'?
 5970 | {0x053D17E0,  VT_ENUM,            8008,  S8008_0,  sizeof(E8008_0),
   E8008_0,      FL_OEM+FL_RONLY, 211, 127}, // Status Feststoffkessel
      |                                          ^~~~~~~
      |                                          S8108_0
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:5134:18: error:
'S8008_0' was not declared in this scope; did you mean 'S8108_0'?
 5134 | #define S10032_0 S8008_0                       // Status
Feststoffkessel
      |                  ^~~~~~~
U:\Downloads\BSB-LAN-master MAX_
Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN_custom_defs.h:6186:42: note: in
expansion of macro 'S10032_0'
 6186 | {0x053D17E1,  VT_ENUM,            10032, S10032_0,
sizeof(E10032_0),   E10032_0,     FL_RONLY,     211, 127}, // Status
Feststoffkessel
      |                                          ^~~~~~~~
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
  Benutzt:
C:\Users\erich\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WiFi
  Nicht benutzt:
C:\Users\erich\OneDrive\Dokumente\Arduino\libraries\WiFiEspAT
exit status 1

Compilation error: stray '\' in program

fredlcore, 2025-02-03:
Das sind Fehlermeldungen, die aus Deiner BSB_LAN_custom_defs.h Datei kommen, schaut so aus, als hättest Du da in Zeile 4955 diesen Dateipfad reinkopiert: 
`u:\Downloads\BSB-LAN-master MAX_Sollwert\BSB-LAN-master\BSB_LAN\BSB_LAN.ino`

Und noch eine Bitte: Wenn Du Logs etc. hier reinkopierst, bitte mit dem "<>" Symbol oben als Code markieren, dann ist das sehr viel besser lesbar...

erichsch, 2025-02-03:
> Ich werde es am Abend ausprobieren und melde mich.
> *Gesendet:* Monday, February 3, 2025 11:45:19 AM

Fehlermeldungen sind weg. War ein copy paste fehler. Danke für den hinweis.

Was mir aufgefallen ist:
1.) wie und wo verstelle ich die sollwerte?
2.) die durchschnittstemperatur (Mittelwert) der beiden MAX thermostate ist
0,0 Grad

Liebe Grüße Erich

fredlcore, 2025-02-03:
Na, direkt bei den angezeigten Parametern, da sollte jetzt neben dem .2'er Parameter ein "Set"-Button sein.
Durchschnittstemperaturen werden nach Neustart immer neu berechnet. Aber jetzt bitte erst mal bei einer Sache bleiben.

erichsch, 2025-02-03:
Durch das neueinspielen der ino habe ich noch keine schreibberechtigung
gesetzt gehabt.
senden funktioniert perfekt.

allerdings die Mittelwerte noch nicht.
Raumtemperatur-Sollwert gemittelt (8741) Anzeige 25 Grad (falsch) 1.
thermostat 25.0 und am 2. thermostat ist 26.0 eingestellt
Raumtemperatur-Istwert gemittelt (8740) 0.0 Grad (falsch)
Raumtemperatur-Istwert gemittelt(10028) 23,2 (falsch)

fredlcore, 2025-02-03:
Also nur zur Klärung: Die 8740/8741/10028 haben nichts mit den MAX-Parametern zu tun, das ist Dir bewusst, oder? Das sind die Parameter, die sich auf das Raumgerät Deiner Heizung beziehen. Und wenn Du keine SD-Karte hast, werden die Mittelwerte auch nicht über einen Reset hin "gerettet", d.h., die werden dann erst nach einiger Zeit wieder aufgebaut.

erichsch, 2025-02-03:
SD Karte war zwar drinnen, aber hab ich sie auch aktiviert? weiß ich heute
nicht mehr. anscheinend nicht.

Alles klar, dann sollten werte irgendwann kommen.
derzeit ca. eine stunde aktiv und noch immer 0,0 oder 25
Der mittelwert des sollwertes (8741) sollte sich doch mit meinen
Raumthermostaten per "sofort" verstellen.
Oder bin ich am falschen datenpunkt?

Das senden funktioniert perfekt. Die thermostate übernehmen den sollwert
und zeigen diesen auch an. perfekt.

fredlcore, 2025-02-04:
Ja, die müssten nach ein paar Minuten kommen, wenn die zu überwachenden Werte größer als 0 sind. Wenn das vorher funktioniert hat, dann mach' bitte ein neues Issue dafür auf. Ansonsten schau' erst einmal, ob Du alles richtig konfiguriert hast. An der Funktion war ich nämlich nicht dran, so dass es eigentlich keine Probleme geben sollte...

closed 2025-02-04 by fredlcore

fredlcore, 2025-02-04:
P.S.: Ich vermute, Du hast in den Einstellungen unter Logging nicht die 24h-Durchschnittswerte angehakt...