fredlcore/BSB-LAN #721: [Bug Report] SD Card Logging Issue on Olimex ESP32-EVB ("Free space in file system: 0 Bytes")

fredlcore, 2025-01-27:
### Discussed in https://github.com/fredlcore/BSB-LAN/discussions/720

<div type='discussions-op-text'>

<sup>Originally posted by **aa-fr-75** January 26, 2025</sup>
Hello,
I’ve successfully deployed BSB-LAN on my Olimex ESP32-EVB with the help of @fredlcore !
Everything works great except for SD card logging. I’m encountering an issue where the system doesn’t recognize the free space on the SD card. Below are the details:

**Device Details:**
- Hardware: Olimex ESP32-EVB with v4.1 from the github project
- SD Card: 32GB, formatted to FAT32 (using GParted on Linux)

**Configuration:**
- "Write to SD Card" enabled in logging mode
- "Calculate 24h Averages" also enabled

**Observed Behavior:**
On the settings page, it shows: Free space in file system: 0 Bytes.
The SD card partition is completely empty and has no other data.
The SD card appears properly inserted and functional when tested on Linux.

**Steps Taken:**
Formatted the SD card to FAT32 using GParted.
Verified the partition and functionality on Linux.
Ensured the SD card is inserted properly into the ESP32-EVB.

**Question:**
Are there additional steps required to configure SD card logging on the ESP32-EVB?
If relevant, could this be related to a specific partition requirements?

Any advice or troubleshooting steps would be greatly appreciated!
Thank you for your time and support.

Best regards</div>

aa-fr-75, 2025-01-31:
Hi,
Sorry for the delay.
Find attached my BSB_LAN_config.h
[BSB_LAN_config.h.txt](https://github.com/user-attachments/files/18624018/BSB_LAN_config.h.txt)
I'll try to add some log latter.
Best regards

fredlcore, 2025-02-01:
The log file is most relevant, starting from booting the ESP32 until the setup procedure is completed.

aa-fr-75, 2025-02-02:
I finally had time to investigate this issue!
Yes, @fredlcore you're right! log is mandatory and has been included:
You can find the associated log [bsblan-1.log](https://github.com/user-attachments/files/18632520/bsblan-1.log)

To "solve" the issue I made the following change in my BSB_LAN_config.h
-//byte LoggingMode = 0; //CF_LOGMODE_SD_CARD | CF_LOGMODE_24AVG | CF_LOGMODE_MQTT | CF_LOGMODE_UDP
+byte LoggingMode = 3;

After flashing the EEPROM, the SD Card is now detected, as shown in log [bsblan-2.log](https://github.com/user-attachments/files/18632531/bsblan-2.log)

**It seems that SD Card logging needs to be enabled in config.h to allow SD Card storage to initialize properly at boot.**

The preceding steps enable the usage of the SD Card as expected.

fredlcore, 2025-02-02:
Yes, you are right, logging needs to be enabled, maybe I'll clarify that in the manual.

closed 2025-02-02 by fredlcore