fredlcore/BSB-LAN #723: [BUG]

jstammi, 2025-02-01:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
2.3.4

**Version of the ESP32 or Arduino framework**  
esp32 by Espressif Systems 3.0.7

**BSB-LAN Version**  
4.2.59-20250109021219
Plus some patches (jstammi/hs-20250116) THAT DO NOT AFFECT THIS ISSUE

**Architecture**  
ESP32 NodeMCU

**Bus system**  
BSB

**Describe the bug**  
All parameters related to the Ionisationsstrom and with unit micro-ampere are showing 1/10 of the value shown on the heating's display (and that the heating expert was expecting to see while doing some troubleshooting last week).

This affects for me:
* 2700
* 2741.1
* 2742.2
* 2743.1
* 2747.1 (? not verfied, yet)
* 2748.1 (? not verfied, yet)
* 8329

**To Reproduce**  
* Open `http://bsb-lan/K20!0`
* check value for e.g. `2700 Sitherm Pro - Ionisationsstrom gefiltert`

**Log files - Bug reports without log files will be closed**  
Serial logging is not possible as the esp32 is residing in a remote location. I can access only by vpn network.

Telnet logging example:
```
LAN->HEIZ QUR 8329.0  - Ionisationsstrom Istwert:
DC C2 00 0B 06 3D 09 0E 16 56 53
HEIZ->LAN ANS 8329.0  - Ionisationsstrom Istwert: 3.16 µA
DC 80 42 0E 07 09 3D 0E 16 00 0C 53 51 F6
#8329: 3.16 µA
Publishing to topic: bsb-lan-heider-see/0/38/8329/status
Payload: 3.16
```

**Expected behavior**  
In the log example above the value should show as `31,55 µA`

**Screenshots**  
[image]

Not sure about the limit values. But best guess same for them.

**Desktop (if applicable, please complete the following information):**  
* OS Windows 11
* Browser Firefox
* Browser version 134.0.1

**Smartphone (if applicable, please complete the following information):**  
n/a

**Additional context**  
n/a

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/user-attachments/files/18629406/BSB_LAN_config.txt)

jstammi, 2025-02-01:
I guess the type used in `BSB_LAN_custom_defs.h` used for those should be `VT_CURRENT` instead of `VT_CURRENT1000`

Patched for me by:

[bsblan-723.patch](https://github.com/user-attachments/files/18629418/bsblan-723.patch)

fredlcore, 2025-02-01:
Thanks, I now compared other parameter lists, and it seems that there is no parameter that uses a divisor of 100 (which VT_CURRENT was supposed to take care of), but rather that all µA parameters use a divisor of 1000. This is now fixed in the most recent version. 
P.S.: I have removed your patch, by the way, as it wouldn't have worked on (almost) any other setup because there are hundreds of different possible variants of device-specific parameter lists out there.

jstammi, 2025-02-01:
Sorry, but your change `41fbabd` is wrong. Now both types devide by 1000. But here division by 100 is needed only.

From the telnet log shown above the decimal results in `3155`. The corrct value to be shown is `31,55`. As before your change by `VT_CURRENT`.

fredlcore, 2025-02-01:
Hm, strange... I'll check with the other users who reported a 1000 divisor instead of a 100, to check with the values that are displayed on their heater. Both your dump as well as those of the others use the same divisor id for this parameter, so I can only go by that. If there is a difference, then there's probably an error in the way the controller was programmed to send the dump. But let me check...

jstammi, 2025-02-01:
I had the heating expert for troubleshooting at my mother's home. And the valuey `<10uA` irritated him. Until he noticed, that the values were shown in the heating's display 10 times higher. And that was within the range, that he expected.

The range being used in the screenshot below should by x10. What is the range that your heating is showing while operating? Single uA or tenth of uA?

[image]

fredlcore, 2025-02-01:
No, your heating configuration is in the dump you sent to me. It may or may not correlate with any of the parameters above. It doesn't matter because none of the above parameters are contained in the dump. The dump provides information for each parameter, and that is all that is needed, as the same parameter can be the same across different models, but it can also differ within the same kind of model, potentially based on the software version of that model. See the pinned discussion for more background on this.

I don't have access to my heater right now, so I cannot check this, that's why I have asked some other users to check on that. What I can tell you is that my BSB-LAN instance states a value of 14.27 µA for parameter 8329, and that one is using a divisor of 1000. If a divisor of 100 were correct in this case, the value would be 142.7 µA – which sounds quite a lot to me.

fredlcore, 2025-02-01:
What is relevant is what is displayed on the heater. That has to be the same as in BSB-LAN. And that is determined by a divisor that is part of the dump. If you tell me that the display is 10 times higher than what BSB-LAN displays, then this is all I need for the comparison. As I said, I cannot check mine right now, but it's currently running at over 30% power and BSB-LAN shows the aforementioned 14.27 µA. But it's a LMU7, not a LMS14, so this may or may not be comparable.

fredlcore, 2025-02-01:
Ok, the first user has already responded and confirmed your observation (heating system shows 9.3 µA and BSB-LAN shows 0.93 µA).

uschindler, 2025-02-02:
Hi, I just have seen this issue.

I have LMU7 (LMU 74) on Brötje WBS 22 C and I can confirm the issue. The values on display are 10 times higher than in BSB-LAN; the screenshot was done at approximately same time than the photo:

[image]

[image]

My custom_defs.h file has `VT_CURRENT1000`:

```
{0x153D2FF0,  VT_CURRENT1000,     8329,  S8329_0,  0,                  NULL,         FL_RONLY,     97, 100}, // Ionisationsstrom Istwert
```

Uwe

fredlcore, 2025-02-02:
**TLDR: Problem solved, both divisors exist, existing users without an OCI420 can just change any VT_CURRENT1000 entry with VT_CURRENT and it will work.**

**Background**
Thank you. That confirms one more time that even the way a heater is connected to the bus can make a difference in the way the data is encoded (again, see the pinned discussion).
Since I also have a LMU74, and my BSB-LAN instance is showing more or less the same value as Uwe's, but with a divisor of 1000, we can first of all establish that both divisor exist, and that I took the VT_CURRENT1000 based on the when I checked the values on my LMU7.

However, my LMU7 is connected via the OCI420's LPB to the bus, not via BSB (where I would probably get the same result as you two), because my heater is controlled by a RVS63. The OCI420 is a weird kind of device as it not just takes anything that is available via BSB and puts it out on the LPB bus, but it seems to have its own complete set of parameters. Which makes sense because it was originally introduced for the LMU64 which did not have BSB or LPB, but OpenTherm (for the room unit).
That's the reason why the OCI420 lacks many parameters that my LMU74 offers via BSB, but on the other hand even adds a few that were not available before. 

If all that wasn't confusing enough, the OCI420 uses different divisors as well, compared to the ones my LMU74 uses for BSB. Again, this makes sense somehow because the newer LMU74 brought higher resolution for some parameters (0.1 slopes instead of 0.5 for some temperatures, for example), and I have checked this for most (but apparently not all) parameters when I created a special list for the OCI420-connected devices (LMU64 and LMU74).

Having reconstructed the changes I made with the e-mails I wrote regarding that matter, I can see now that I used VT_CURRENT initially for both "regular" as well as OCI420-connected devices. Then, an LMU7 user with an OCI420 contacted me and said that his ionistaion values were too high by a factor of 10, so I assumed (and probably checked myself on my heater back then) that this was the case for everyone and that nobody had complained because this is a rarely used parameter where someone without the relevant knowledge wouldn't even notice the difference (I, at least, can't, otherwise, I'd be able to tell whether my value of 14.72 µA is way too low, so that a factor 10 would have to be applied). 

So, thanks again for bringing this up, I will use a divisor of 100 in the future when generating the device-specific parameter lists for "regular devices" and a divisor of 1000 for OCI420-connected devices. The former users can just replace any occurrence of VT_CURRENT1000 with VT_CURRENT, and the problem is solved.

uschindler, 2025-02-02:
Thanks. That was my plan how to do it: Search, replace, fine. I was about to suggest the same.

closed 2025-02-03 by fredlcore