fredlcore/BSB-LAN #724: [FEATURE REQUEST]

apocaliss92, 2025-02-02:
**Is your feature request related to a problem? Please describe.**
More a quality of life thingy, too many entities are discovered into home assistant 

**Describe the solution you'd like**
Would be great if only the entities related to the logging parameters requested (besides custom ones, I guess?) would be discovered on mqtt

**Describe alternatives you've considered**
Disabling the not needed entities from homeassistant side

**Describe your own contribution**
Can try to help if addressed to the piece of code dedicated.

fredlcore, 2025-02-02:
I'll look into that.

jstammi, 2025-02-02:
Please consider, that values having not chosen for logging, are sent currently by mqtt though, if you browse parameters in the web ui. 

I do not know if this is a behaviour by intention or by accident. But it is very helpfull for "static" parameters, e.g. settings to the heating, that you do not expect to change and therefore are not contained in the list. But in case of a change (or maybe even regularly automated once a day) one can easily update the values shown in the HomeAssistant, but just browsing single parameters or categories or maybe even all(?) by just calling the related urls.

I would prefer the list getting filtered to those, that the heating ever provided any value for. If this is not discoverable upfront, then maybe lazily, on 1st time having received a value(?).

fredlcore, 2025-02-02:
It's by design, as explained in the docs. And it has nothing to do with what the OP wants to achieve which has to do with HA's MQTT auto-detection feature where currently all parameters are advertised for discovery. That aside, your suggestion wouldn't be doable, as BSB-LAN would have to keep a list of up to over 1000 parameters that it may have discovered, and since a simple query of all parameters (e.g. by doing `/0-9999`) would have `discovered` all existing parameters in one go, the whole idea of limiting the advertised parameters would be in vain.

Generally speaking, implementing this feature is something for rather lazy people which is why I'm still hesitant to do so. Because it just takes me about 5 minutes to first hide all discovered parameters and then add the 20-odd parameters that I want to be visible in HA, as I have shown in the HA setup video in my YT channel. Since this is a one-time effort, I'm not sure if it's worth to add extra code for this, but since several users have asked for it, I agreed to look into it.

fredlcore, 2025-02-03:
Ok, I have now tied the auto-discovery to the configuration setting `CF_LOGMODE_MQTT_ONLY_LOG_PARAMS`, so if you activate the option to only send log parameters to the MQTT broker, this will also affect auto-discovery.
Please install the latest version and check for yourself.

closed 2025-02-03 by fredlcore

apocaliss92, 2025-02-04:
Thank you very much! Will try around today!