fredlcore/BSB-LAN #725: [BUG] HomeAssistant shows many auto-detected parameters being in fact numerical ones as states instead

jstammi, 2025-02-02:
**Version of the Arduino IDE**  
2.3.4

**Version of the ESP32 or Arduino framework**  
esp32 by Espressif Systems 3.1.1

**BSB-LAN Version**  
4.2.71-20250126152957
Plus some patches, one preliminary fixing this for those parameters affected for me

**Architecture**  
ESP32 NodeMCU

**Bus system**  
BSB

**Describe the bug**  
Some numerical parameters, that one would expect a graph plotted for in HomeAssistant, are presented as a "states" timeline bar instead. E.g. 2702 (Sitherm Schrittmotor Position):
shows up in this style (for illustration; from different parameter, as this is preliminary patched for me already):
[image]

This seems to affect all numerical parameters without an unit.

**To Reproduce**  
* start the HA autodiscovery preocess (assuming that HA is configured for this already)
* wait for data for HA entity created for e.g. 2702
* click to that entity for showing the time series graph

**Log files - Bug reports without log files will be closed**  
MQTT auto-discovery message generated for this parameter:
```
{"~":"bsb-lan/0/20/2702","unique_id":"2702-163-16-202380010","state_topic":"~/status","icon":"mdi:numeric","name":"00-20 Sitherm Pro - 2702 - Position Schrittmotor","device":{"name":"bsb-lan","identifiers":"bsb-lan-D0EF76489194","manufacturer":"bsb-lan.de","model":"4.2.71"}}
```

**Expected behavior**  
The generated HA entity should present it's data as a x/y-graph

**Screenshots**  
Is: see above
Expected:
[image]

**Desktop (if applicable, please complete the following information):**  
* OS Windows 11
* Browser Firefox
* Browser version 134.0.1

**Smartphone (if applicable, please complete the following information):**  
n/a

**Additional context**  
n/a

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/user-attachments/files/18632153/BSB_LAN_config.txt)

jstammi, 2025-02-02:
Comparing the HA entity configuration for affected parameters, I found the defintion `"state_class": "measurement"` missing.

With that I identified the `if` construct in `BSB_LAN/include/mqtt_handler.h` lines 467ff that seems responsible.

My preliminary patch changes this for those parameter types of the parameters affected for me. As I am not sure about side-effects for other custom defs.

See https://github.com/fredlcore/BSB-LAN/compare/master...jstammi:BSB-LAN:pr/ha-numbers-as-states

fredlcore, 2025-02-02:
Since I'm not a HA user and I'm also not familiar with what counts as a "measurement" in HA, maybe you can help here by checking which one of the data types listed in `const units optbl[]={` in BSB_LAN_defs.h that has the `U_NONE` setting would require the `state_class: "measurement"` to be added and which ones not. Otherwise, we'll end up adding dozens of data types here that do not have a unit but could potentially be a measurement or not.

fredlcore, 2025-02-03:
Should be fixed in the most current version. Please check.

maitmandel, 2025-03-26:
@jstammi - can you advise:
after the improved firmware has been flashed to bsb-lan, do I need to re-install the integration in HA?

jstammi, 2025-03-27:
On HA side nothing needs to be down afais

It is in bsb-lan that you have to re-run sending the configs again to the
mqtt auto-detection topics that HA is listening on

Afterwards the items in HA should change automagically.

If not: there is also a link in bsb-lan to remove them from HA. Try this
then prior another auto-detection run from bsb-lan