fredlcore/BSB-LAN #728: [BUG] Rich JSON id value is not encapsulated with " "

omuskyo, 2025-03-12:
**Version of the Arduino IDE**  : _arduino IDE 2.3.4_
**Version of the ESP32 or Arduino framework**  : Arduino SAM Boards (32-bits ARM Cortex-... 1.6.12
**BSB-LAN Version**  : _Version: 4.1.11-20241107004949_
**Architecture**  :  _Hardware: Due_
**Bus system**  : BSB

**Describe the bug**  

When using Rich JSON, "id" value is not encapsulated with " ", so it's like a numeric value with operators:
id: 0/37/8773

It's difficult to process such values in node red

A workaround (or fix ?) is to modify include/mqtt_handler.h on line 97
from :
`appendStringBuffer(&sb_payload, "\":{\"id\":%d/%d/%g", (param.`...
to
`appendStringBuffer(&sb_payload, "\":{\"id\":\"%d/%d/%g\"", (param.`...

**To Reproduce**  
use mqtt Rich JSON

**Log files - Bug reports without log files will be closed**  
I don't think it is useful

**Expected behavior**  
get id value inside "  " so it's known as string value

**Screenshots**  

**Desktop (if applicable, please complete the following information):**  

**Smartphone (if applicable, please complete the following information):**  

**Additional context**  
Add any other context about the problem here.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress

fredlcore, 2025-03-12:
You are using an older version, and the line you mention is no longer part of the code. So please install the latest version (as also stated in the bug report) and see if your problem still persists.

closed 2025-03-12 by fredlcore

fredlcore, 2025-03-12:
The most recent one is directly from the master repository, using the green download button.