fredlcore/BSB-LAN #732: [BUG] Home Assistant unable to parse value '---' from MQTT messages

Koli0842, 2025-04-08:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
2.3.5

**Version of the ESP32 or Arduino framework**  
3.2.0 esp32 by Espressif Systems

**BSB-LAN Version**  
BSB-LAN, Version 5.0.6-20250404021554

**Architecture**  
ESP32 Olimex-POE-ISO

**Bus system**  
BSB

**Describe the bug**  
 Home assistant logs failures to parse '---' values for some sensors. The expected measurement unit would be °C, and --- is not a valid numeric value. I am not sure how to communicate an unknown value to HA, if it's possible even

**To Reproduce**  
Steps to reproduce the behavior:
1. Set up MQTT
2. Create auto-discovery messages
3. Enable Home Assistant BSB-LAN integration with MQTT
4. Observe error logs

**Log files - Bug reports without log files will be closed**  
```
2025-04-08 23:30:23.462 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating st                                                                             ate of sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8951_oszto_eloremeno_homerseklet_alapjel, topic: 'BSB-LAN/0/35/8951/status' with payload: b'---'
ValueError: Sensor sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8951_oszto_eloremeno_homerseklet_alapjel has device                                                                              class 'None', state class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value:                                                                              '---' (<class 'str'>)
2025-04-08 23:31:19.947 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating st                                                                             ate of sensor.bsb_lan_00_34_hoterm_diagnosztika_8411_hoszivattyu_alapjel, topic: 'BSB-LAN/0/34/8411/status' with payload: b'---'
 ValueError: Sensor sensor.bsb_lan_00_34_hoterm_diagnosztika_8411_hoszivattyu_alapjel has device class 'None', state                                                                              class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '---' (<class 'str'>
2025-04-08 23:35:19.692 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating st                                                                             ate of sensor.bsb_lan_00_33_kaszkad_diagnosztika_8138_kasazkad_eloremeno_homerseklet_aktualis_ertek, topic: 'BSB-LAN/0/33/8138/status' with payload: b'---'
ValueError: Sensor sensor.bsb_lan_00_33_kaszkad_diagnosztika_8138_kasazkad_eloremeno_homerseklet_aktualis_ertek has                                                                              device class 'None', state class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric                                                                              value: '---' (<class 'str'>)
2025-04-08 23:35:22.863 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating st                                                                             ate of sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8950_oszto_aktualis_homerseklet_ertek, topic: 'BSB-LAN/0/35/8950/status' with payload: b'---'
ValueError: Sensor sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8950_oszto_aktualis_homerseklet_ertek has device cl                                                                             ass 'None', state class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '-                                                                             --' (<class 'str'>)
```

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**  
N/A

**Smartphone (if applicable, please complete the following information):** 
N/A

**Additional context**  
I used to have some issues with HA integration unable to parse values, most of them have been resolved by updating to 5.0.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
[BSB_LAN_config.txt](https://github.com/user-attachments/files/19658205/BSB_LAN_config.txt)

fredlcore, 2025-04-09:
It's not a bug; it's a feature ;). The thing is that `---` has a different meaning than the value `0`, as the former means that the parameter is *disabled* and the latter is an actual temperature. Think, for example, of the comfort setpoint temperature for heating circuit 1 and you have a heat pump that can be used for both heating and cooling: If you want to disable that heating circuit, you set it to `---`. That means that the heating system will neither cool or heat for that circuit. But if you set the value to `0`, it would try to bring the room temperature down to zero (of course, there are other safety mechanisms in place to prevent that, but just to illustrate the difference). The same goes for other parameters, such as sensors: If, for example, the outside temperature sensor is not present, it sends `---`. That tells you that the value is not applicable. Sending zero would give a wrong impression, because it might not be freezing outside.

There are two ways to solve this: Either you somehow deal with this issue inside Home Assistant, for example by converting the value to zero even though this "wrong". Or I change the auto discovery to not send parameters that can have the value `---` as numerical values, but as text. But then you couldn't have those operations on those parameters anymore that require the parameter to be defined as numerical. Both ways have their pros and cons, but changing BSB-LAN's behaviour would require very good reasoning before I can do so as it will affect every user immediately...

Koli0842, 2025-04-09:
Hey, thanks for the quick reply! I also thought that 0 might not be the ideal solution, but it sounds more complex than I initially thought. Probably the only good way to deal with this would be if HA supported a special message for setting a given sensor back to "unknown". I know it does revert to unknown by some criteria, for example when it doesn't get values when expected, but not sure what exactly makes it tick. It could be a nice solution, as unknown values are not plotted on the history charts, but if other systems already depend on '---' being sent, this change could break that functionality. I might ask around on the HA discord, whether something like this already exists

Is there a sort of "less official" way to bring up questions like this, than a bug report? I've heard a forum being mentioned time to time but haven't really it. I've been really happy with BSB-LAN the past 2 months as I slowly started to set up HA and other tools for the first time, thank you for your work again!

fredlcore, 2025-04-09:
There is also the "Discussions" section here on GitHub, but in this case, it's fine here as well. And yes, let us know here once you have found out how HA is addressing this issue in other contexts.

jstammi, 2025-04-09:
I guess this could be solved by adding "rules" to the mqtt discovery for **numerical items** that tell HA that a non-numerical value (here: '---') means, that there is no value is available:

```
...
    "availability_topic ": "~/status",
    "availability_template": "{% if value | is_number %}'online'{% else %}'offline'{% endif %}",
    "value_template": "{% if value | is_number %}{{ value }}{% else %}{% endif %}",
...
```

(described here: https://www.home-assistant.io/integrations/sensor.mqtt/)

Unfortunately it is not sufficient to just specify the availability rules as the `value_template` is evaluated before them. This is why there is additionally the `value_template` returning an empty value for non-numerical values which leads to them being ignored - without issuing error messages to the logs.

With the change above the values displayed remain same in HA. The non-numeric '---' were not shown before, too, but spam the logs.

fredlcore, 2025-04-09:
Thanks! Since I don't use HA, could someone add this functionality and test it and if it works submit a pull request?

This is the part in `mqtt_handler.h` where any additional configuration lines could be added to:
```
          } else {
            appendStringBuffer(&sb_topic, "sensor/");
            if (decodedTelegram.unit[0]) {

```
(currently after line 481)

jstammi, 2025-04-09:
Not sure if I will manage already today. But will test this short term.

For the @Koli0842 concern: maybe we should handle values consisting only of minus chars (+ whitespaces) this way if expected to be numerical. To keep awareness on other non-numeric values that BSB-LAN might submit and whose semantic could be something different (any ideas for such?).

fredlcore, 2025-04-09:
Keep in mind that this only affects HA, and only if auto discovery is used.
What I haven't really understood is whether the "online/offline" in your example is relevant here or if this is just an example. 
If HA can deal with `---` if `status` is set to `offline`, then I can just take those lines above 1:1 and add them to the discovery payload, right? That I can do myself and do it quickly. I thought that these lines need to be adjusted to the `---`, but the more I look at it, it seems that `if value | is_number` will do this check. 

The question for me would be whether the `---` would still be visible in HA. Because it's not really true that the sensor is offline. It's that it is disabled. That is a significant difference because I would want to know if the parameter cannot be reached (for example because there is a problem with the query and it returns an empty value which would also fall through the `if value | is_number` test. Or if it is disabled. In that regard, it's important to keep in mind that some parameters can legitimately change from `---`, `0` and greater than zero values (room temperature influence, parameter 750, for example), so I would want to have that information available whether the parameter is marked `offline` or not...

Koli0842, 2025-04-09:
With my basic HA knowledge I would say the only status for a sensor besides it's value is "unknown", which means HA cannot get a value for whatever reason. I think the status of online / offline is for each entry of the device, and not the device in global. HA has a concept of disabling the sensor, but it's for it's own usage, HA stops fetching it's data.
Currently `---` is not visible in HA at all, besides the logs as an exception, and history graphs keep plotting it as the last known value (which is a bug / weird feature on their end to be fair). If my assumption is correct, by applying the change above, given a `---` message, HA would change the value of the sensor to "unknown", and after a numerical value it would go back as normal.
I don't think there is a way in HA to differentiate the reason a sensor has no value.

jstammi, 2025-04-09:
1) the `---` is not shown in the HA UI. Not witout nor with this change.
But it is no longer registered as error. Not in the log. Nor in the logs viewer that does some analysis on the errors and sorts them according to their number of occurrences and hints the user problems to solve with the configuration. So the change helps there, too, as one no longer needs to bother with that not-an-error-but-shown-as-such

2) HA supports in principle an `unknown` status. Not sure, if with different (in HA "speaking") integrations this is possible to set. With the MQTT one it is not AFAIK.

3) the semantics for BSB-LAN `---` as "disabled" is IMHO comparable to HA semantics of "not available". And that status addresses the value only, not the sensor in total. So this should fit.
BUT ... HA seems not to do anything with that information, at least with my running a little bit old version and with the data arriving by MQTT. Honestly I would expect, that no value is shown then for that time. But HA impl keeps the value before AFAIS - but this should not affect the way defining the sensor I guess.

4) `online` / `offline` are just the default literals for the availability status, With 2 more lines one can define different ones.
And honestly, if the BSB-LAN meaning/wording is `enabled` / `disabled`, then we should name them like this, no problem ;-)

5) I would stay with my proposal to update the logic to match only for whitespace/minus-only character combinations. I guess their semantic is used consistently for BSB-LAN sensors (with possibly the number of minusses differing from sensor to sensor).
But if for some reason there are used different characters (additionally) resulting in values that cannot be parsed as numbers, then this should not be completely hidden by this catch-all approach as given before.
I will update the rule to consider this. But ... as my jinja2 knowledge is still only some basic one, I need to develop this tomorrow, with live-testing immediately in parallel with my HA. For today it is too late.

GN

fredlcore, 2025-04-10:
Hm, I'm not sure that we should proceed this way if HA simply keeps the old value because this would be a wrong information. There are some sensor (I'd have to look it up in more detail) where this information is relevant to know, similarly to room temperature influence (parameter 750). With 750, it may not make a big difference because `---` is eventually the same as 0. But if the previous value wasn't zero, then it will stay at a wrong value. 
I think the more accurate way would be to send these parameters not as numerical values, but as text values. Then the `---` would be logged correctly as well and the error logs would also disappear. You may not be able to do all the aggregate functions or whatever HA offers, but in doubt, I'd rather go for a clean than a convenient solution...

jstammi, 2025-04-10:
This would break the display of all numerical data, no option at all

And... all data IS sent as text, including '---'. The sensor definition
would be changed.

And... it is not the problem of the sender, if the receiver handles the
data received semantically incorrect. All Info is provided. I guess I
should raise this at HA as an issue (and search prior, as I doubt this was
not noticed before).

(and again: with change as proposed, the display of the data does not
change atm afais)

fredlcore, 2025-04-10:
Don't make such absolute statements if you don't know the details. 
First of all, not every parameter, and not even every numerical parameter, can have the disabled status, and for each parameter type, it is known in BSB-LAN whether or not it can have this status. So, any change could be done in a way affecting only those that can have the disabled status as a possibility. Therefore, it wouldn't "break the display of all numerical data." Strictly speaking, a parameter that can contain non-numerical content is not a numerical parameter after all.

Secondly, not all data is sent as `text`. Not even all numerical data. So far, only writable parameters are sent as `text` in order to account for the possibility of disabling them. Other numerical data is sent, among others, as `sensor`, and a number of them even with `state_class` `measurement`. But by adding such a proposed distinction, writeable parameters that cannot be disabled could be transmitted as `sensor`, and not as `text`, as well.

What I don't understand is: When I open a parameter in HA that is transmitted during AD as `sensor`, such as 8951 (current flow temperature) - what is shown in HA if the value is `---`? Is it "unknown"? "offline"? "disabled"? empty? remaining at the last value? What I don't want to do is mix it up with something that seems to be the closest approximation but has a different meaning after all. First and foremost, I don't want to get a lot of support requests because people don't understand their heating system well enough and then say "Oh, my BSB-LAN/my parameter is offline/disabled/unknown/a different/older value compared to the one shown on the heater." What most people are always capable of is comparing what is in BSB-LAN/HA and what is shown on the heater. So, the closer both displays are, the lesser the risk of confusion and resulting support queries.

After all, the auto-discovery feature is one that prioritizes convenience over functionality, but not over accuracy. Users always have the alternative to configure everything via YAML, and most power users apparently do so. Auto-discovery should focus on bringing in all parameters and allow access to the parameter values as directly as possible. Any lack of feature functions such as log plotting etc. takes a back seat. If I can fix the errors coming up on HA's side, I'm happy to do so, and maybe changing the entity type from `sensor` to `text` might do the trick. But concealing the exact value for the sake of fixing log displays etc. is not an option.

Koli0842, 2025-04-10:
When the value sent is '---' such as 8951 during tonight on my system, the value in HA is the last parseable one. I think as an end result in HA, this is the most confusing probability. I have attached a screenshot of the history view, there is no textual activity log for sensors with numerical values it seems.
(On this note, 8703 is transmitted as text instead of a numerical value, despite seemingly a celsius value, correct? I do have a textual activity log for that one, and it doesn't show up on plots.)

[image]

For reference, the exceptions are aggregated on the UI and tell some info about when they occurred:

```
Logger: homeassistant.components.mqtt.models
Source: components/mqtt/models.py:366
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: April 9, 2025 at 6:19:24 PM (12 occurrences)
Last logged: April 9, 2025 at 10:23:28 PM

    Exception raised while updating state of sensor.bsb_lan_00_34_hoterm_diagnosztika_8411_hoszivattyu_alapjel, topic: 'BSB-LAN/0/34/8411/status' with payload: b'---'
    Exception raised while updating state of sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8951_oszto_eloremeno_homerseklet_alapjel, topic: 'BSB-LAN/0/35/8951/status' with payload: b'---'
    Exception raised while updating state of sensor.bsb_lan_00_33_kaszkad_diagnosztika_8138_kasazkad_eloremeno_homerseklet_aktualis_ertek, topic: 'BSB-LAN/0/33/8138/status' with payload: b'---'
    Exception raised while updating state of sensor.bsb_lan_00_35_hofogyaszto_diagnosztika_8950_oszto_aktualis_homerseklet_ertek, topic: 'BSB-LAN/0/35/8950/status' with payload: b'---'
```

I have seen "unavailable" as a status when HA itself was down (due to a power outage), and "unknown" before being able to grab an initial parameter. After that, it seems to behave sticky, with the current implementation

Sidenote 2: please excuse my sensor names all in hungarian.. I didn't think that the language the installers set the machine to would translate into the config files when I sent in the parameter dump, and didn't want to bother you with regenerating it, so for now I'm stuck with this nightmare 😅

fredlcore, 2025-04-10:
Thanks a lot for that - that is exactly what I mean: If we use `sensor` as the type of entity, then this is what will happen because `sensor` expects a numerical value always. If, on the other hand, it is defined as `text`, such as 8703, for example, then it will be accurate, but not display as a graph. That's the problem that needs to be addressed, because in the graph form now, at least the visual representation is simply wrong and misleading. 

As for the Hungarian version: I can create any language supported by the room unit that the user asks me for, but if they don't state anything, I'll use their country's language, assuming that this is what they are most comfortable with. While it takes me some time to create a new file (see the pinned post in the discussion section here), I can create one in English for you, if you prefer and it makes fixing this matter easier.

Koli0842, 2025-04-10:
On the HA discord, a developer suggested I could wrap the value in question in a "proxy-sensor" to wrap the original value, and apply basically the same rule @jstammi suggested here. I think the end result would be the same if someone wants to use the value as a sensor (I would, it makes sense to me personally), but with more hoops.

How I see this question, it comes down to what's the primary goal of the MQTT auto-discovery feature; it's seemingly made just for HA, based on the topic structure. Is it more important to be convenient while being as accurate as HA allows, or to be 100% true to BSB states, and let the users deal with building their own integration on the HA end for general usability. I personally would probably create a manual MQTT configuration in yaml at that point, instead of wrapping the sensor. (Which I have already been considering, the default sensor names picked up in HA are too long for the UI, changing the list of submitted topics requires regenerating the auto-discovery messages, which resets any updated made in HA anyways. I think it's intended, and probably no way to solve it.)

What was the reason 8703 ended up being a text field anyways, but 8700/8704 not, if you don't mind me asking? I guess when there is no outside temperature sensor connected, they would all be `---`. I found a lengthy Siemens documentation online that I use as reference for these values, but as the language suggests it's aimed at installers, I haven't had the chance to actually understand all the correlations..

As for the translation: My bad in that case! I didn't even occur to me there'd be a hungarian translation. I've always found my native language more confusing than helpful in IT gear, lots of weird translations and very long and uncommon words; I personally use all  English everywhere. In BSB I see german, english, and hungarian mixed (missing translations I guess), and as a bonus hungarian has a fair bit of typos (swapped letters). If the source for that is out in the open and not dependent on hardware provided IDs, I am happy to contribute by fixing it, just point me in a direction. I don't think it's more relevant to fixing this issue than the ease of understanding of my screensots and logs between us, I see it as an inconvinience both here and personally. I was planning to just deal with it as-is until we can generate the file ourselves to not take up more of your time than necessary. If that's still long ways off and not a big deal, I wouldn't mind having it in English, maybe the next time you set up your environment for others or something like that.

jstammi, 2025-04-10:
Please give me some seconds. I guess I found a **proper** solution

fredlcore, 2025-04-10:
@Koli0842: My setup is set up every day, but mostly I work on the translatons on the weekend.
As for which sensor is defined as a `text` or as a `sensor`, this depends on a number of factors:
All writeable parameters that do not use a dropdown or a switch are sent as `text` because that is the only writeable type that really works. `numbers` would have been an option, but it requires min and max values which we don't know beforehand, and the defaults are not suitable for all types of parameters. So, could it be that 8703 is writeable in your configuration?
All read-only parameters are sent as either `binary_sensor` or `sensor`, depending on the scope. All other types of entities do not make sense or have other complications when applying them to BSB-LAN's parameters.

jstammi, 2025-04-10:
How do you send a value not as text through MQTT?

The interface between BSB-LAN and HA are text messages:
BSB-LAN encodes the values to string and writes it to the topic.
HA parses the string then again. The way of customizing this is by specifying this in the auto-discovery.

And yes, I stay with the absolute statement. As with **treating** as text sensor in HA, the graphs no longer work, are rendered useless (expect to look like 2nd screenshot in @Koli0842 message abobe, one color per number then).

I understand, that showing wrong data is a bad idea (that no one got aware of, yet, btw). But rendering everything else useless and mis-using the system (here: HA), to overcome this, is a hack but no solution.

Back to the topic:

After checking the docs twice again, I checked the sources now. And there one finds the (AFAIS: un-documented) solution: instead of submitting an empty string as new value (resulting in staying with the last good one before) when receiving the '---', one has to transmit `None` (see https://github.com/home-assistant/core/blob/ea50bbeb11543df18a7e974fee6826bddbddfa00/homeassistant/components/mqtt/sensor.py#L325). In that case the value changes to `Unbekannt` then (with HA configured for german language). The availability checking must be removed then again, as this leads to those `None` values not being considered any longer afterwards.

**Only** this needs to be added:
```
    "value_template": "{% if value | regex_match('^[-]+$') %}None{% else %}{{ value }}{% endif %}",
```

The catch-all `is_number` is replaced now by matching values consisting of any number of minus characters only. If different characters are used, the error is shown in the log but it is ignored as it is without adding the below. Which seems legimite to me, as the semantics of such is not known then and one need to investigate on such (again).

Results in looking like this "live" then:

[image]

And when plotted (gaps represent times being `---` here:

[image]

(PS: the small increase shown short after 12:41 is a glitch in the plotting, whose value is not in the MQTT history and that appears there only with the 1st value plotted after the gap, retrospectively - I will raise an issue against HA for this, if this was not found and/or even fixed already within the last months)

fredlcore, 2025-04-10:
You're confusing sending the data as text (as opposed to sending it in binary) with the MQTT component `text` (as opposed to `sensor`, `binary_sensor` etc.) that is specified in the auto-discovery process and which determines the representation of the data received. We're obviously only talking about the latter and how `text` represents data im comparison to `sensor`. One does not have an issue with `---`, the other one does.

You're also wrong about no one having had an issue with that, it's been coming up every once in a while for years, both here on GitHub (see here, for example: https://github.com/fredlcore/BSB-LAN/issues/469) and in direct mails to me, and it is not limited to HA, but is an issue with other systems as well. The only difference compared to other systems is that the auto-discovery feature is the only one where BSB-LAN is offering something geared towards one specific home automation system (although auto-discovery is technically open to other systems as well)

And again, neither are all parameters affected, but only those that actually can be disabled (maybe 50%); nor is sending a - strictly speaking - non-numerical parameter as text a "mis-use of the system". Ask any programmer in a language that distinguishes between numerical and string values which one this would fall into.

Anyway, thanks for making an effort to find a solution, I really appreciate it, and it looks like an easy implementation. But having a status of "Unknown" is simply not a correct representation. And the gaps in the graph make it easy to be mistaken for an interruption or broken transmission, because that's when usually my alarm bells are starting to ring when I see these gaps in the graphs.
When you click on the parameter at a time it is disabled, what does the parameter show then? "None" or `---`? If it's the former (or zero or empty), then this solution is off the table. If it's the latter, then I would consider it.

Of course it would be easier if we'd know if `---` is occuring in a context where it means "no sensor available" or "function not activated" or "unable to retrieve data". But the heater only gives us `---`, and that's all. That's why this context-dependent information can only be interpreted correctly in the context of the parameter's meaning, and therefore needs to be distinguishable from other values. To reduce all these different meanings to `None` and treat it as if no data was transmitted just to make it plot the graph somehow, is not a solution. 

For situations like this, I don't think that auto-discovery is the right tool. For people who just want to retrieve the current status of their heating system and have a log-file-like history of values, this is not an issue in the first place. And whoever wants to reduce the depth of information in order to have a nicely looking graph should do so willingly and knowingly, for example by creating a YAML configuration for that parameter.

Koli0842, 2025-04-10:
Thanks both of you for your efforts! I think I will go with manually constructed MQTT definitions in HA based on @jstammi's solution. There is seemingly no way to handle any special value for "normally-numeric" parameters in HA basides this, and since I already send the status parameters aswell, I have the full context in my history to tell if the value is expected to be missing or not. I prefer to keep the graph functionality, I'll probably have to do it for 8703 too, which is indeed writable on my system for some reason. Data as-is feels a bit needlessly complicated for my case

fredlcore, 2025-04-11:
What do you mean with "since I already send the status parameters aswell, I have the full context in my history"? You mean the log history in HA or elsewhere? What happens if you go to a parameter (not the plot) currently while the value is at `---`? Does it show the `---` or something else?
To me, the preservation of `---` or, if that's not possible, at least an unambiguous entry (and not a misleading "None", "Unavailable" or "Unknown") would be the goal of any change.

Koli0842, 2025-04-12:
I've just meant that I already send 8000, 8003, 8006 to HA too, so viewing data from BSB there, I as a user can connect that if 8006 shows there is no heat need, 8951 is not a problem if it's missing to me personally. You cannot really view sensor values in a non-plotted way as far as I've seen, I'm new to HA too so I might be mistaken; this is how the entity modal looks like in settings:

[image]

Note that "5 hours ago" means the last change of value HA was able to parse, in this case when it switched to 52°C, not related to when it went '---'. According to 8006 history, 21:22 it turned off, while 8951 reports the last change of value as 21:00. Other sensors that have not changed since my last reboot show 3 days ago for example, despite the fact that messages are continously reported, so it's pretty safe conclusion to say HA just tracks actual value changes. I don't think it's possible to achieve your goal with HA currently, and I don't really see a way forward everyone's happy with, so I plan on just defining my topics myself as such.

If you feel like providing these values as text might be the most correct way, it would solve at least the error logs so that might be one way to close this issue. To me viewing temperature values in HA that change by .1°C on my heating system as text would harm usability, so I'll personally just stick to using it as a sensor and have it as None for ---. At least until I get better using HA and find a better way.

fredlcore, 2025-04-12:
Thank you for these observations - what I mean by viewing the sensor values in a non-plotted way is the display of the value as shown on top of the graph (52.0°). If you or @jstammi make the changes as suggested by him, could you send me a screenshot of how this value is displayed when the status switches to `---`? Does it say `None` then? Or some other textual representation? Or does it just stay empty?

I know that there is not an ideal situation as long as we either have to tweak the `---` value to `None` or leave things as they are and get even more inaccurate values or get accurate values that are not plottable when sending the values as `text`.

I guess if you build your devices using YAML you could represent the `---` accordingly, for example, by visually indicating that the sensor is not active. According to the manual, it should also be possible to adjust the the entity's device class:
https://www.home-assistant.io/docs/configuration/customizing-devices/
But after checking, this somehow only seems to apply for manually added entities, not for those via auto-discovery.

Another downer is that the heater only transmits the information about whether parameters come with and without `---` if these parameters are writeable. Since we're only plotting read-only parameters, my initial plan to send only those numerical parameters that can have the status `---` as `text` thus would, therefore, affect far more parameters than actually affected by this problem.

Since, as mentioned above, other home automation systems are also affected by this issue, I'm thinking of making this configurable. Rather than dealing with this (only) in the MQTT auto-detection, I could add a configuration option that, if active, sends the configured value instead of `---`. For HA, this could then be `None`, for others either some other string or as worst-case-fallback simply zero. In any case, it would require the user to make that decision, thereby reducing any (legitimate) support inquiries. 

Could you therefore try and see what happens if you send `None` to any of those parameters via MQTT Explorer or a similar tool? That would be great!

fredlcore, 2025-04-13:
I have now taken the more holistic appraoch and added the config variable `replaceDisabled` which, in the case of Home Assistant, can be filled with `None`.  This would then be used instead of `---` in all forms of communication with external connections (JSON, MQTT, URL-parsing). For writeable parameters, the value in the text input field will remain `---` to make sure that the correct value to disable the parameter is being sent. I tried it with MQTT Explorer, where `None` is received instead of `---`, so I assume it would work in HA as well.

closed 2025-04-13 by fredlcore

Koli0842, 2025-04-14:
Sorry it took me so long, my "thermostat" input is 3 BSK Zephyrs averaged, and they forgot to renew their ssl cert causing a sunday outage and my heater to continue to run. (man I love random closed down iot devices) Just for completeness' sake, the workaround above, and from my understanding your replaceDisabled setting with None as well, shows up in HA as "unknown". So far however, only 8411 and 8951 went ---. When I opened my ticket, 8138 and 8950  were also ---, I guess I have some homework left in understanding what does what in the system.

Thank you again!

fredlcore, 2025-04-14:
Ok, "unknown" is not ideal, but now with the `replaceDisabled` variable, it is up to the user to make their own decision. Thanks again for testing!