fredlcore/BSB-LAN #737: [BUG]

jstammi, 2025-05-09:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
2.3.6

**Version of the ESP32 or Arduino framework**  
3.2.0

**BSB-LAN Version**  
5.0.9

**Architecture**  
ESP32 NodeMCU

**Bus system**  
BSB

**Describe the bug**  
Web UI shows `2503 Kessel - Wartezeit Schalter: s 0102 - decoding error` instead of the decoded value

**To Reproduce**  
1. open BSB-LAN Web UI
2. click `Heizungsfunktionen`
3. click `Kessel 2203 - 2670` (url path `/K19!0`)
4. check parameter `2503`

**Log files - Bug reports without log files will be closed**  
Log from telnet connection (no physical access to BSB-LAN at the moment):
```
LAN->HEIZ QUR 2503.0 - Wartezeit Schalter:
DC C2 00 0B 06 3D 05 1B 92 0E B8
HEIZ->LAN ANS 2503.0 - Wartezeit Schalter: FIXPOINT len !=3: 01 02
DC 80 42 0D 07 05 3D 1B 92 01 02 37 26
#2503:  s 0102 - decoding error
```

**Expected behavior**  
Show the decoded value instead of an error message

**Screenshots**  
[image]

**Desktop (if applicable, please complete the following information):**  
 - OS: Windows 11
 - Browser: Firefox
 - Browser version: 137.01

**Smartphone (if applicable, please complete the following information):**  
n/a

**Additional context**  
n/a

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**
```
/************************************************************************************/
/**********************************************<**************************************/
/* Settings -   BEGIN                                                               */
/************************************************************************************/
/************************************************************************************/

// Upon first installation, rename this file from BSB_lan_config.h.default to BSB_lan_config.h and adjust settings accordingly

// Optionally, if one intends to commit the settings to git, too, move all customizations containing data considered to
// remain private to a new file named "BSB_LAN_config_private.h" and un-comment the line below
#include "BSB_LAN_config_private.h"

/* Select language; so far German is the most complete, with English following.
[... 319 lines cut ...]
// Enable ESP32 over-the-air updates.
// Pro: You don't need to physically connect your PC to the ESP32 in order to update BSB-LAN.
// Contra: Someone who has access to your network and finds out about BSB-LAN's username and password can install their own software on the ESP32.
bool enable_ota_update = true;

// Reduce clock speed of ESP32 from 240 MHz to 80MHz, saving ca. 20% energy.
// Works well when connecting via LAN, but since it reduces WiFi range and log file display times when using WiFi, it is disabled by default.
bool esp32_save_energy = false;

// "External" web server. Read files from SD-card / flash memory. Only static content: html, js, css, jpg, etc.
bool webserver = false;

// Debug options
byte debug_mode = 1;  // Debug mode: 0 - disabled, 1 - send debug messages to serial interface, 2 - send debug messages to telnet client
byte verbose = 1;     // If set to 1, all messages on the bus are printed to debug interface. Set to 2 only if you are asked by the developers as the extensive output will significantly slow down the microcontroller.
bool show_unknown = true; // true - show all parameters, false - hide unknown parameters from web display (parameters will still be queried and therefore take time!)

#define CONFIG_VERSION 39

/************************************************************************************/
/************************************************************************************/
/* Settings -   END                                                                 */
/************************************************************************************/
/************************************************************************************/
```

jstammi, 2025-05-09:
Best guess: is it the wrong type being used in `BSB_LAN_custom_defs.h`? The next paramater, 2504, shows a seconds value, too, is decoded correctly, but is defined as `VT_SECONDS_SHORT5` (2 bytes) instead of for 2503 being defined as `VT_SECONDS_WORD5_N` (3 bytes - which is AFAIS wrong) ... ?

fredlcore, 2025-05-09:
Thanks for pointing that out, there was indeed an error in the conversion from your dump to the parameter file. You can simply fix it by changing it to `VT_SECONDS_SHORT2_N`.