fredlcore/BSB-LAN #738: [BUG] Bus pins stored in EEPROM prevents future changes

TCWORLD, 2025-05-12:
**Before submitting a bug report, please make sure that you have checked the [FAQ](https://docs.bsb-lan.de/faq.html) and [troubleshooting section](https://docs.bsb-lan.de/troubleshooting.html) of the manual**. Most problems that are reported to us can be solved by going through the solutions provided there.

**Do not remove any of these questions! Bug-Reports that do not contain all required information will not be accepted. If you cannot provide any required information, explain why this is the case.**

**Version of the Arduino IDE**  
Please note: Versions 1.8.19 and below are not supported by this project anymore.

2.3.6

**Version of the ESP32 or Arduino framework**  
See ***Tools/Board/Board Manager*** for exact version you are using

3.2.0

**BSB-LAN Version**  
As displayed in the web-configuration or copy from `bsb-version.h` file. Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly. We don't fix bugs for discontinued platforms (such as Arduino Mega).

5.0.9-20250414104155

**Architecture**  
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.)

Olimex ESP32-POE-ISO

**Bus system**  
Which bus BSB-LAN is running on (BSB, LPB, PPS)?

BSB

**Describe the bug**  
A clear and concise description of what the bug is.

Currently the `bsb_pins` value is being stored in the EEPROM config data. This means if the value is then changed in the config header (e.g. discovering that the default {0,0} auto-discover isn't applicable), the new value is ignored.

However this parameter doesn't appear to be settable via the web interface, so in order to ever change it, the EEPROM is first wiped.

**To Reproduce**  
Steps to reproduce the behavior:
1. Compiled with the default settings `{0,0}` 
2. Uploaded to the board, which means `{0,0}` will be stored in the EEPROM.
3. Change the parameter in the BSB_LAN_config.h and rebuild/re-upload
4. New value set in the config header being completely ignored, as the old value is restored from the EEPROM.

In `BSB_LAN_config.h`:
```
// Setting bus pins
byte bus_pins[2] = {36,4}; //First value - RX pin, second value - TX pin
```

After recompile/reupload. Note I also added two extra debug logs, one just before the EEPROM `registerConfigVariable` lines, and a second just before the pin auto-discovery code that prints out the value of `bus_pins` just to make sure the compilation got the changes.

This is the serial monitor output. Note that the `bus_pins` value from the config is ignored and overwritten with the EEPROM values.

```
**Config.h RX/TX pins 36, 4**
Reading EEPROM
...
... omitted for brevity
...
Reading done.
EEPROM schema v.14 Program schema v.14
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 3
Address EEPROM option 3: 7
Address EEPROM option 4: 227
Address EEPROM option 5: 307
Address EEPROM option 6: 491
Address EEPROM option 7: 0
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 496
Address EEPROM option 13: 816
Address EEPROM option 14: 820
Address EEPROM option 15: 1140
Address EEPROM option 16: 1146
Address EEPROM option 17: 1147
Address EEPROM option 18: 1151
Address EEPROM option 19: 1155
Address EEPROM option 20: 1159
Address EEPROM option 21: 1163
Address EEPROM option 22: 1165
Address EEPROM option 23: 1169
Address EEPROM option 24: 1173
Address EEPROM option 25: 1237
Address EEPROM option 26: 1301
Address EEPROM option 27: 1302
Address EEPROM option 28: 1303
Address EEPROM option 29: 1313
Address EEPROM option 30: 1314
Address EEPROM option 31: 1634
Address EEPROM option 32: 1635
Address EEPROM option 33: 1639
Address EEPROM option 34: 1640
Address EEPROM option 35: 1641
Address EEPROM option 36: 1642
Address EEPROM option 37: 1675
Address EEPROM option 38: 1740
Address EEPROM option 39: 1805
Address EEPROM option 40: 1870
Address EEPROM option 41: 1902
Address EEPROM option 42: 1904
Address EEPROM option 43: 1905
Address EEPROM option 44: 1906
Address EEPROM option 45: 1938
Address EEPROM option 46: 2002
Address EEPROM option 47: 2003
Address EEPROM option 48: 2004
Address EEPROM option 49: 2036
Address EEPROM option 50: 2037
Address EEPROM option 51: 2038
Address EEPROM option 52: 2039
Address EEPROM option 53: 2040
Address EEPROM option 54: 2041
Address EEPROM option 55: 2042
Address EEPROM option 56: 2043
Address EEPROM option 57: 2044
Size of cmdtbl: 2912
free RAM: 248492
BSB-LAN version: 5.0.9-20250414104155
Microcontroller: ESP32_POE_ISO
**Selected RX/TX pins 0, 0**
Using RX/TX pins 36, 5
Bus type defined in config: BSB
ESP Arduino version: 3.2.0
```

**Expected behavior**  
A clear and concise description of what you expected to happen.

The software to respect the value in the BSB_LAN_config.h file. Or at the very least a message to indicate there is a mismatch.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**  
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**  
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. Safari, FireFox]
 - Browser version [e.g. 22]

**Additional context**  
Add any other context about the problem here.

Note this is a custom hardware implementation based on the schematics in this repository. Pin 4 is used as Tx in the repo comitted schematic, but it seems this was changed to Pin 5 at some point without updating the available schematics due to a boot config pin issue.

In my setup I use an additional pin, GPIO 32, to control the BSB-LAN transmit and keep it disabled until the end of setup as a way of preventing short circuit detections by the BSB controller.

**Attach your BSB_LAN_config.h file to your bug report (remove any confidential information if necessary). Do not ZIP or otherwise compress it. Bug reports without this file attached will be closed immediately.**

Note have to change file extension...

[image]

[BSB_LAN_config.h.txt](https://github.com/user-attachments/files/20165247/BSB_LAN_config.h.txt)

fredlcore, 2025-05-12:
Do you have `UseEEPROM = 1` in your config? In that case, the configuration from BSB_LAN_config.h does not overwrite the EEPROM settings because otherwise every update would force you to either reconfigure the web configuration or to always keep track of any changes in the web interface also in your configuration file. If you want to (only) use the configuration in BSB_LAN_config.h, then just set UseEEPROM to 0 or change the web interface settings to not use the web configuration.

TCWORLD, 2025-05-12:
Ah, I see, I only changed the bare minimum of config settings from their defaults, then discovered that I needed to change the pins after uploading.

So I set the UseEEPROM in the config to 0, will it still write everything in the config file to the EEPROM?
I.e. if I then turn "Read config from EEPROM" in the web interface to "On", will it still recall the old settings, or use the new settings.

fredlcore, 2025-05-12:
No. You just change the values in the web configuration, if you want to use that one. Or you set UseEEPROM to 0 and do all the configuration via the config file. Up to you. You only need to wipe the EEPROM if you misconfigured it in such a way that you cannot use BSB-LAN anymore (see Troubleshooting).

closed 2025-05-12 by fredlcore